esphome/issues #6921: Bug With platform: i2s_audio it keeps repeating audio like its stuck in a buffer

comet424, 2025-04-06:
### The problem

so i dont know if anyone has reported this issue or not if so  sorry..

so i use nodered to export audio to little esphome esp32 devices   in the last couple fixes prior it would work  it would play audio and stop after 20 seconds..  now  it doesnt stop it randomly plays the audio message 
if you shut off nodered... if you remove  esp devices from home assistant.. it still keeps playing the audio file.. nothing stops it except a reboot of the esp32..  so i use it to anouncing an incoming phone call.. but after 20 seconds it should stop playing and it does that...  but then randomly every 20 seconds or what not it will repeat the audio..  like the audio is saved in the esp32 device  and it keeps replaying it   like a stuck in ram or stuck buffering or whatever you call it..

### Which version of ESPHome has the issue?

2025.3.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.1

### What platform are you using?

ESP32

### YAML Config

```yaml
here is my code

##############
### Bedroom ###
##############
esphome:
  name: speaker-1
  friendly_name: "Speaker #1"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "NojvwCefQii8YN3rIz/vBGtcth0xMBB4KFvt8/nfaM4="

ota:
  platform: esphome
  password: "00b8433143fa1e2b21ddaf8da034f8a4"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Speaker-1 Fallback Hotspot"
    password: "BHlK4SVkN5Cn"

captive_portal:
web_server:

i2s_audio:
  i2s_lrclk_pin: GPIO16
  i2s_bclk_pin: GPIO21

media_player:
  - platform: i2s_audio
    name: ESPHome I2S Media Player
    dac_type: external
    
    i2s_dout_pin: GPIO17
    mode: mono

switch:
#####################
### Reboot Switch ###
#####################
  - platform: restart
    name: "Restart"

binary_sensor:
  - platform: status
    name: "Status"

text_sensor:
######################################################    
### Uptime Sesnor Template To Show In Hour Min Sec ###
######################################################
  - platform: template
    name: "Uptime"
    id: uptime_human
    icon: mdi:clock-start

sensor:
  - platform: uptime
    name: Uptime Sensor
    id: uptime_sensor_speaker_1
    update_interval: 60s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor_speaker_1).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? to_string(days) + "d " : "") +
                (hours ? to_string(hours) + "h " : "") +
                (minutes ? to_string(minutes) + "m " : "") +
                (to_string(seconds) + "s")
              ).c_str();
```

### Anything in the logs that might be useful for us?

```txt

```

MichaelMKKelly, 2025-04-06:
This has cropped up a few times on home assistant discord.
It appears that whilst using the arduino framework the player is not reporting that its finished playing so gets stuck in "playing" and therefore ends up repeating.

the solution for most people seems to be switching to esp-idf and using the components:
i2s_audio - speaker - media_player - voice_assistant

this seems to be more stable.

I am not sure the exact cause of the problem, perhaps its upstream in the framework?

switching to esp-idf may not be possible for everyone but does seem to be doable for people building custom voice satellites and the such

comet424, 2025-04-06:
i dont think i have a esp-idf??  i have  esp32-wroom  devices  that i use..

and i dont use the voice-assistant  i tried that stuff when it first was coming out to make my own voice assistants but it wasnt working right so i just gave up...  how i do is 
i use nodered to use tts.speak 
and then for that node i use
```
{           "message": msg.payload, 
            "media_player_entity_id": [
            "media_player.speaker_1_esphome_i2s_media_player",
            "media_player.speaker_2_esphome_i2s_media_player",
            "media_player.speaker_3_esphome_i2s_media_player"
            ]
}
```

so far  i not doing voice satellite...  this just to hook up a bluetooth speaker  i use the aux in  and i just able to wifi to a speaker..  no mic  i havent had luck  and maybe just home assistant just too early to play with it yet ..  i was waiting for the bugs to be worked out  as i was having 0 luck at the time  last year

MichaelMKKelly, 2025-04-06:
I think you should be fine to switch esp-idf framework and use 
i2s_audio - speaker - media_player
without the voice assistant component and may solve the issue

comet424, 2025-04-06:
so do i just change
```
esp32:
  board: esp32dev
  framework:
    type: arduino
```
to
```
esp32:
  board: esp-idf
  framework:
    type: arduino
```

and this is how i do it in the node red  as it repeats every 3 seconds  but after a delay of 20 seconds it gets send the stop code
id done a screen shot  but aparently i cant add it  grrr

but here is the nodered code
```
[{"id":"8a8b7a9b0777e4c5","type":"comment","z":"dbde2ad1b4af7ed2","name":"Caller ID Audio To Speakers","info":"","x":120,"y":580,"wires":[]},{"id":"e6b7b86b19ef7e79","type":"link in","z":"dbde2ad1b4af7ed2","name":"Caller ID Audio In","links":["53e42447df6b3486"],"x":15,"y":620,"wires":[["0d81f49ea1c2a940"]]},{"id":"b2c586bf01754d32","type":"trigger","z":"dbde2ad1b4af7ed2","name":"Repeat Calling Message","op1":"","op2":"0","op1type":"pay","op2type":"str","duration":"-3","extend":false,"overrideDelay":false,"units":"s","reset":"stop","bytopic":"all","topic":"topic","outputs":1,"x":390,"y":620,"wires":[["5464265c26e3e88c","cf52d032b140505f"]]},{"id":"0d81f49ea1c2a940","type":"function","z":"dbde2ad1b4af7ed2","name":"Is Calling","func":"msg.payload = msg.payload.Name + \" Is Calling\"\nreturn msg;","outputs":1,"timeout":0,"noerr":0,"initialize":"","finalize":"","libs":[],"x":140,"y":620,"wires":[["b2c586bf01754d32","cf52d032b140505f"]]},{"id":"d8a655169ef0cdc5","type":"change","z":"dbde2ad1b4af7ed2","name":"Send Stop","rules":[{"t":"set","p":"payload","pt":"msg","to":"stop","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":150,"y":680,"wires":[["b2c586bf01754d32","bbec5f67e127845d"]]},{"id":"ef669585a0fc7126","type":"link in","z":"dbde2ad1b4af7ed2","name":"Caller ID Audio Stop In","links":["e17b0690b082e16f"],"x":15,"y":680,"wires":[["d8a655169ef0cdc5"]]},{"id":"5464265c26e3e88c","type":"api-call-service","z":"dbde2ad1b4af7ed2","name":"Text-To-Speech To Devices","server":"915baa48.9b1898","version":7,"debugenabled":false,"action":"tts.speak","floorId":[],"areaId":[],"deviceId":[],"entityId":["tts.piper"],"labelId":[],"data":"{           \"message\": msg.payload, \t            \"media_player_entity_id\": [\t            \"media_player.speaker_1_esphome_i2s_media_player\",\t            \"media_player.speaker_2_esphome_i2s_media_player\",\t            \"media_player.speaker_3_esphome_i2s_media_player\"\t            ]\t}\t","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","blockInputOverrides":false,"domain":"tts","service":"speak","output_location":"","output_location_type":"none","x":700,"y":700,"wires":[["d735bfc525933aee"]]},{"id":"bbec5f67e127845d","type":"debug","z":"dbde2ad1b4af7ed2","name":"debug 23","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":400,"y":680,"wires":[]},{"id":"cf52d032b140505f","type":"debug","z":"dbde2ad1b4af7ed2","name":"debug 25","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":400,"y":560,"wires":[]},{"id":"d735bfc525933aee","type":"debug","z":"dbde2ad1b4af7ed2","name":"debug 26","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":820,"y":620,"wires":[]},{"id":"915baa48.9b1898","type":"server","name":"Home Assistant","version":5,"addon":true,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":true,"cacheJson":true,"heartbeat":false,"heartbeatInterval":"30","areaSelector":"friendlyName","deviceSelector":"friendlyName","entitySelector":"friendlyName","statusSeparator":"at: ","statusYear":"hidden","statusMonth":"short","statusDay":"numeric","statusHourCycle":"h23","statusTimeFormat":"h:m","enableGlobalContextStore":true}]
```

but ill try that esp-idf  its only recently come up from one of these esphome updates or home assistant updates

MichaelMKKelly, 2025-04-06:
@comet424 can try this:

```
##############
### Bedroom ###
##############
esphome:
  name: speaker-1
  friendly_name: "Speaker #1"

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "NojvwCefQii8YN3rIz/vBGtcth0xMBB4KFvt8/nfaM4="

ota:
  platform: esphome
  password: "00b8433143fa1e2b21ddaf8da034f8a4"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Speaker-1 Fallback Hotspot"
    password: "BHlK4SVkN5Cn"

captive_portal:
web_server:

i2s_audio:
  i2s_lrclk_pin: GPIO16
  i2s_bclk_pin: GPIO21

speaker:
  - platform: i2s_audio
    id: output_speaker
    i2s_dout_pin: GPIO17
    dac_type: external    
    bits_per_sample: 32bit
    channel: right
    buffer_duration: 60ms

media_player:
  - platform: speaker
    name: "ESPHome I2S Media Player"
    id: media_out
    announcement_pipeline:
      speaker: output_speaker
      format: WAV
    codec_support_enabled: false
    buffer_size: 6000
    on_pause:
      - logger.log: "Playback paused!"
    on_idle:
      - logger.log: "Playback finished!"

switch:
#####################
### Reboot Switch ###
#####################
  - platform: restart
    name: "Restart"

binary_sensor:
  - platform: status
    name: "Status"

text_sensor:
######################################################    
### Uptime Sesnor Template To Show In Hour Min Sec ###
######################################################
  - platform: template
    name: "Uptime"
    id: uptime_human
    icon: mdi:clock-start

sensor:
  - platform: uptime
    name: Uptime Sensor
    id: uptime_sensor_speaker_1
    update_interval: 60s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor_speaker_1).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? to_string(days) + "d " : "") +
                (hours ? to_string(hours) + "h " : "") +
                (minutes ? to_string(minutes) + "m " : "") +
                (to_string(seconds) + "s")
              ).c_str();
```

I don't use node red.

the entity name in home assistant might change so you may need to update automations

comet424, 2025-04-06:
ok ill try it and let you know.. i appreciate it.. frustrating i tell ya  when it doesnt stop  and a reboot fixes it lol  but ill give it a try see how it works

comet424, 2025-04-06:
apparently it didnt like that code change

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf

```

```
INFO ESPHome 2025.3.3
INFO Reading configuration /config/esphome/speaker-1.yaml...
Failed config

media_player.i2s_audio: [source /config/esphome/speaker-1.yaml:42]
  
  This feature is only available with frameworks ['arduino'].
  platform: i2s_audio
  name: ESPHome I2S Media Player
  dac_type: external
  i2s_dout_pin: GPIO17
  mode: mono
```

comet424, 2025-04-06:
nvm  i copy and pasted your entire code its compiling now..  i only had changed that little bit.. of code i showed thats where i got the error

MichaelMKKelly, 2025-04-06:
they are just warnings dont worry about them

comet424, 2025-04-06:
so that work around didnt work.. its still repeating the message at a random time..  so its still doing it to my 3 speakers  i have   to stop it i have to reboot the esp32 devices

MichaelMKKelly, 2025-04-06:
have you restarted home assistant since deploying new firmware version?

comet424, 2025-04-06:
ya ive rebooted HA multiple times also if i remove the esp device from the devices settings  it still keeps going only fix is reboot of the esp device... might have to add a temp reboot after 20 sec delay  reboot the module

comet424, 2025-04-06:
so my work around is after the loop is over like 20 second delay and it sends a stop for repeating the message.  it then reboots all 3 speaker esp modules...   as a temp fix  till a fix can be found i just dont know if its a esp issue  or if HA isnt sending a cease and dismiss of the message   like you mentioned its not getting told  to stop repeating or release the audio from memory and then stuck in a glitch loop

FredyReady, 2025-04-11:
+1 Here, since last update from 2025.2.2 to 2025.3.3 my ESP speaker with I2S component is repeating TTS every minute. Rolling back to 2025.2.2. did not help.

avbfr, 2025-04-11:
+1 here. Since update to 2025.3.3

comet424, 2025-04-11:
@FredyReady @avbfr my temp fix is in my automation i did in node red since i have a restart button in the every esp device..  i added after its done speaking it reboots the esp  its the only work around i was able to do till there is a fix

vogon1, 2025-04-13:
+1 here as well, keeps repeating last mesage.

BobHome29, 2025-04-17:
Hello, I have the same problem, the ESP32 media_player keep repeating messages, you have to add a reboot of the ESP to each automation so as not to have the problem, it's not a solution. 
Do you think you can solve the problem quickly or should you go back to an older version of ESPHome builder for now? Thank you for your DEV ESPHome. 
Sincerely, Bruno

comet424, 2025-04-18:
@BobHome29 i not a programmer to python etc.  old school programer.. i was only offering the work around i was only able to do after each message is complete was to reboot the esp32 device.. so it just toggles the restart switch..  there is a new esphome update  but it doesnt update esp-idf  mode  but esp32dev   i had to create a new bug in esphome section and i not sure if the esp32dev  mode with the new firmware 4  will solve it as they do have a fix for speaker pause  but i do not know if it solves the issue as i still finding a bug and havent had a chance to test

cizar-szymon, 2025-05-01:
I confirm that such cyclic playback also occurs for me. Turning off the power turns off the playback loop until the first audio playback. The system worked correctly for about a year before the update.

[19:07:02][C][api:143]:   Using noise encryption: YES
[19:07:02][C][audio:225]: Audio:
[19:07:02][C][audio:247]:   External DAC channels: 1
[19:07:02][C][audio:248]:   I2S DOUT Pin: 9
[19:07:34][W][component:239]: Component i2s_audio.media_player took a long time for an operation (55 ms).
[19:07:34][W][component:240]: Components should block for at most 30 ms.
[19:08:14][W][component:239]: Component i2s_audio.media_player took a long time for an operation (55 ms).
[19:08:14][W][component:240]: Components should block for at most 30 ms.
[19:08:54][W][component:239]: Component i2s_audio.media_player took a long time for an operation (55 ms).
[19:08:54][W][component:240]: Components should block for at most 30 ms.
[19:09:35][W][component:239]: Component i2s_audio.media_player took a long time for an operation (55 ms).
[19:09:35][W][component:240]: Components should block for at most 30 ms.

comet424, 2025-05-01:
so with the latesty esphome  still cant update the speakers  in idf mode  to 2025.4.1
still bugged  was hoping a new fix

olimexsmart, 2025-05-04:
Same problem here. 

This guy says it found a workaround but it's clear to me: [](https://community.home-assistant.io/t/music-assistant-media-media-player-playing-on-repeat-non-stop/849443/3)

buteomont, 2025-05-04:
Same problem here. Started a few updates ago.  Only way to stop it from repeating is to cycle the power on the speaker.

anabolyc, 2025-05-05:
Same problem. Started with all TTS commands gets repeated every minute or so. But recently noticed that wav playback also gets repeated after 120 seconds or so. Device log shows nothing, it just playback restarts and usual warning about component block time. 

Here is the ESPHome log (22:44:40 - legit playback start, 22:42:37 it restarts itself, 22:44:33 once again)

[image]

Wierdly, in the HA player time keep counting beyond track time and never really stops itself

[image]

p-jean, 2025-05-06:
Please see here:
https://github.com/home-assistant/core/issues/142022

p-jean, 2025-06-18:
Fixed in esphome 2025.6.0, thanks to everyone