esphome/issues #6925: TTS message to media player repeats

cmeldas, 2025-04-08:
### The problem

After the update to ESPHOME 2025.3.X when I issue TTS message with cloud to the esphome media_player, the message repeats from the speaker indifferently with a period of around the 30s. 

It is doing this on my all esphome speakers. Both are running platform arduino, one is ESP32 the other is ESP32S3.

The current fix is to call pause for the media player after some delay. 

### Which version of ESPHome has the issue?

2025.3.X

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.X

### What platform are you using?

ESP32

### Board

ESP32 and ESP32S3

### Component causing the issue

media_player

### YAML Config

```yaml
First one:
esphome:
  name: paster-display
  platformio_options:
    build_flags: “-DBOARD_HAS_PSRAM”

esp32:
  board: lolin_s3
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ***********

ota:
  - platform: esphome
    password: ********************

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Paster-Old Fallback Hotspot"
    password: **************

captive_portal:
   

spi:
  clk_pin: GPIO40
  mosi_pin: GPIO41
  miso_pin: GPIO39

color: !include ./shared/my_colors.yaml
font: !include ./shared/my_fonts.yaml

display:
  - platform: ili9xxx
    model: ST7796
    data_rate: 40MHz
    rotation: 90
    invert_colors: false
    dc_pin: GPIO42
    reset_pin: GPIO2
    update_interval: 1s
    lambda: |-
      it.fill(COLOR_BLACK);
      it.print(0, 0, id(helvetica_24), id(my_red), TextAlign::TOP_LEFT, "Hello World!");

      static int xx = 10;
      xx +=10;
      if (xx > it.get_width())
        xx = 0;
      it.print(0, 50, id(helvetica_48), id(my_green), TextAlign::TOP_LEFT, "Hello World!");
      it.print(xx, it.get_height(), id(helvetica_48), id(my_blue), TextAlign::BOTTOM_RIGHT, "Hello World!");

      auto touch = id(my_touchscreen)->get_touch();
      if (touch) // or touch.has_value()
        it.filled_circle(touch.value().x, touch.value().y, 10, id(my_red));

touchscreen:
  platform: xpt2046
  id: my_touchscreen
  cs_pin: GPIO38
  interrupt_pin: GPIO48
  update_interval: 50ms
  threshold: 400
  calibration:
    x_min: 280
    x_max: 3860
    y_min: 340
    y_max: 3860  
  on_touch:
    - lambda: |-
        ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
            touch.x,
            touch.y,
            touch.x_raw,
            touch.y_raw
            );

output:
  - platform: ledc
    pin: GPIO45
    id: my_backlight_gpio

light:
  - platform: monochromatic
    output: my_backlight_gpio
    #name: "Display Backlight"
    id: my_backlight
    restore_mode: ALWAYS_ON

i2s_audio:
  i2s_lrclk_pin: GPIO47
  i2s_bclk_pin: GPIO21

media_player:
  - platform: i2s_audio
    name: Paster player
    dac_type: external
    i2s_dout_pin: GPIO46
    mode: mono

Second:

esphome:
  name: obchudek-zvonek
  friendly_name: obchudek-zvonek

esp32:
  board: lolin32_lite
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: *********

ota:
  platform: esphome
  password: ****************

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Obchudek-Zvonek Fallback Hotspot"
    password: ****************

sensor:
  - platform: wifi_signal
    name: "RSSI"
    id: sensor_rssi
    update_interval: 90s
    entity_category: "diagnostic"

  - platform: uptime
    name: "Uptime"
    id: sensor_uptime
    update_interval: 300s
    entity_category: "diagnostic"    

status_led:
  pin: GPIO22       

media_player:
  - platform: i2s_audio
    name: "Repracek"
    dac_type: external
    i2s_dout_pin: GPIO4
    mode: mono

i2s_audio:
  i2s_lrclk_pin: GPIO17
  i2s_bclk_pin: GPIO16
  

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO18
      inverted: True
      mode: INPUT_PULLUP
    name: "Zvonek"
    filters:
      - delayed_on_off: 10ms

light:
  - platform: neopixelbus
    type: GRB
    variant: WS2812
    pin: GPIO14
    num_leds: 1
    name: "Indikator"
    id: my_light
    effects:
      - pulse:
          name: "Slow Pulse"
          # transition_length: 1s      # defaults to 1s
          update_interval: 2s      

   

i2c:
  scl: GPIO26
  sda: GPIO25

pn532_i2c:
  update_interval: 1s      
  on_tag:
    then:
      - if:
          condition:
            lambda: !lambda |-
                if (!tag.has_ndef_message()) {
                  return 0;
                }
                auto message = tag.get_ndef_message();
                auto records = message->get_records();
                for (auto &record : records) {
                  std::string payload = record->get_payload();
                  size_t pos = payload.find("https://www.home-assistant.io/tag/");
                  if (pos != std::string::npos) {
                    return 1;
                  }
                }
                return 0;
          then:
            - homeassistant.tag_scanned: !lambda |-
                  if (!tag.has_ndef_message()) {
                    return x;
                  }
                  auto message = tag.get_ndef_message();
                  auto records = message->get_records();
                  for (auto &record : records) {
                    std::string payload = record->get_payload();
                    size_t pos = payload.find("https://www.home-assistant.io/tag/");
                    if (pos != std::string::npos) {
                      return payload.substr(pos + 34);
                    }
                  }
                  return x;
          else:
            - logger.log: "NFC tag unsupported"
```

### Anything in the logs that might be useful for us?

```txt

```

avbfr, 2025-04-11:
Same issue here.

avbfr, 2025-04-11:
Same issue here.

p-jean, 2025-05-06:
Please see here:
https://github.com/home-assistant/core/issues/142022

closed 2025-11-27 by github-actions[bot]