esphome/issues #6926: Manual Download not work

bdvboing, 2025-04-08:
### The problem

Build project, select manual download. Receive: (see picture

[image]

)

### Which version of ESPHome has the issue?

2025.3.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-04-08:
Can you use the browser network log to see what it's trying to download?
Is it compiling successfully first?

bdvboing, 2025-04-08:
Compiling good. If I select install "Wirelessly" all good

[logs_jk-bms-u_compile.txt](https://github.com/user-attachments/files/19656497/logs_jk-bms-u_compile.txt)

ssieb, 2025-04-08:
What do you see in the browser network log when you try to download?

bdvboing, 2025-04-08:
So, I can't even start the download.
At this point, there should be an option to choose the format: Factory format or OTA format, but instead I get an error right away, without the option to choose.

ssieb, 2025-04-08:
So what do you see in the network log?  Something is getting a 404.

bdvboing, 2025-04-08:
[image]

fedottt, 2025-04-09:
same here

have three HASS (1 - https valid domain with certs, 2,3 - http on IP)
on **HTTP** get 404 after compiling
on **HTTPS** file download **good**

same versions esphome, same yaml

some screens

[image]

[image]

[image]

my yaml

```javascript
substitutions:
  name: jk-bms-den
  bms0: "${name} bms0"
  bms1: "${name} bms1"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  bms0_mac_address: C8:47:80:E1:4F:2B
  bms1_mac_address: C8:47:80:21:20:31
  # Please use "JK02_24S" if you own a old JK-BMS < hardware version 11.0 (hardware version >= 6.0 and < 11.0)
  # Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
  # Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  bms0_protocol_version: JK02_32S
  bms1_protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.1.0

esp32:
  board: esp32dev  #esp32-s3-devkitc-1   #esp32dev
  framework:
    type: esp-idf
    #sdkconfig_options:
    #  CONFIG_COMPILER_OPTIMIZATION_SIZE: y
    #  CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
    #  CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
    #  CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: smart
  password: "password"
  fast_connect: true

ota:
  platform: esphome
  on_begin:
    then:
      - switch.turn_off: ble_client_switch0
      - switch.turn_off: ble_client_switch1
      - logger.log: "BLE connection suspended for OTA update"

web_server:
  port: 80
[... 161 lines cut ...]

  - platform: ble_client
    ble_client_id: client0
    id: ble_client_switch0
    name: "${bms0} enable bluetooth connection"
  - platform: ble_client
    ble_client_id: client1
    id: ble_client_switch1
    name: "${bms1} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    errors:
      name: "${bms0} errors"
    total_runtime_formatted:
      name: "${bms0} total runtime formatted"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    errors:
      name: "${bms1} errors"
    total_runtime_formatted:
      name: "${bms1} total runtime formatted"
```

fedottt, 2025-04-09:
Ahha..
This is after update Hass to latest release 2025.4.1
Before update to it worked well

m2teusz, 2025-04-13:
I have exactly the same problem. The problem is with ESP32 mini D1, with other boards ESP32dev or ESP mini the problem does not occur

[image]

odis68, 2025-04-29:
It is noted that another project does not give errors!!!
Configuration board :
esp32:
  board: esp32dev
  framework:
    type: arduino

[image]