esphome/issues #6928: ld2420.cpp references undeclared operating_selector_ — compile fails in ESPHome 2025.2.0

brianseaders, 2025-04-11:
### The problem

The LD2420 component fails to compile in ESPHome 2025.2.0 due to references to `operating_selector_` in ld2420.cpp that are not declared in the LD2420Component class. This causes a hard compiler failure even with minimal valid YAML configuration.

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A — not applicable

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

ld2420

### YAML Config

```yaml
esphome:
  name: gpt-test
  friendly_name: GPT TEST

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:
  encryption:
    key: "xwZAt/o8OnkbM/DQMMRNrtJr3NpmEQzB6nUPe16kRjI="

ota:
  - platform: esphome
    password: "ef96dc7a8d381405b21220f7f0bf8635"

wifi:
  ssid: "test"
  password: "12345678"

  ap:
    ssid: "Gpt-Test Fallback Hotspot"
    password: "3ewPru113TbA"

captive_portal:

uart:
  id: uart_ld2420
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 256000

ld2420:
  uart_id: uart_ld2420
  id: radar

sensor:
  - platform: uptime
    name: "Uptime Sensor"
    update_interval: 60s

  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s

  - platform: ld2420
    ld2420_id: radar
    moving_distance:
      name: "LD2420 Moving Distance"

binary_sensor:
  - platform: status
    name: "Device Online"

  - platform: ld2420
    ld2420_id: radar
    has_target:
      name: "LD2420 Has Target"
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs\gpt-test\src\esphome\components\sensor\filter.cpp.o
src/esphome/components/ld2420/ld2420.cpp: In member function 'virtual void esphome::ld2420::LD2420Component::dump_config()':
src/esphome/components/ld2420/ld2420.cpp:88:45: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
   LOG_SELECT(TAG, "  Operating Mode", this->operating_selector_);
                                             ^~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:88:3: error: 'LOG_SELECT' was not declared in this scope
   LOG_SELECT(TAG, "  Operating Mode", this->operating_selector_);
   ^~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:88:3: note: suggested alternative: 'LOG_SECRET'
   LOG_SELECT(TAG, "  Operating Mode", this->operating_selector_);
   ^~~~~~~~~~
   LOG_SECRET
src/esphome/components/ld2420/ld2420.cpp: In member function 'virtual void esphome::ld2420::LD2420Component::setup()':
src/esphome/components/ld2420/ld2420.cpp:140:11: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
     this->operating_selector_->publish_state(OP_SIMPLE_MODE_STRING);
           ^~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:145:11: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
     this->operating_selector_->publish_state(OP_NORMAL_MODE_STRING);
           ^~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp: In member function 'void esphome::ld2420::LD2420Component::set_operating_mode(const string&)':
src/esphome/components/ld2420/ld2420.cpp:296:11: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
     this->operating_selector_->publish_state(state);
           ^~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:315:11: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
     this->operating_selector_->publish_state(OP_SIMPLE_MODE_STRING);
           ^~~~~~~~~~~~~~~~~~~
*** [.pioenvs\gpt-test\src\esphome\components\ld2420\ld2420.cpp.o] Error 1
============================================= [FAILED] Took 146.58 seconds =============================================
PS C:\esphome-test>
```

swoboda1337, 2025-04-11:
If you add the select to the yaml it will build: https://esphome.io/components/sensor/ld2420.html#select
```
select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode
```

ssieb, 2025-04-12:
It would probably work if you just have `select:` in the config, but try the linked PR.

descipher, 2025-04-12:
I would think the right path would be to make it required, I can't see any reason to omit the mode select. The only time it can be omitted is with an ld2420 firmware that extreme old.

descipher, 2025-04-12:
In order to calibrate the ld2420 you must have select mode available.

ssieb, 2025-04-12:
Ok, then make it required.  I'm not quite sure how to do that though.

descipher, 2025-04-12:
Code gen does it but what's odd here is it already is required! So there should be a message stating it when the YAML is missing the requirement. So my question becomes did we ignore the message or are we asking that it can be optional?

```CONFIG_SCHEMA = {
    cv.GenerateID(CONF_LD2420_ID): cv.use_id(LD2420Component),
    cv.Required(CONF_OPERATING_MODE): select.select_schema(
        LD2420Select,
        entity_category=ENTITY_CATEGORY_CONFIG,
    ),
}

descipher, 2025-04-12:
@jesserockz  There may be another issue that's not obvious any thoughts on this one?

swoboda1337, 2025-04-12:
> Code gen does it but what's odd here is it already is required! So there should be a message stating it when the YAML is missing the requirement. So my question becomes did we ignore the message or are we asking that it can be optional?

That check is inside the select schema. Since there is no select in the yaml that file isn't used.

swoboda1337, 2025-04-12:
You can add a dependency on select: 
```
DEPENDENCIES = ["select"]
```
However there is no guarantee that its the select in ld2420. Too back something like this wouldnt work:
```
DEPENDENCIES = ["ld2420.select"]
```

swoboda1337, 2025-04-12:
Actually I had that backwards you can do this in ld2420 init.py, it works:
```
DEPENDENCIES = ["select.ld2420"]
```

descipher, 2025-04-12:
@swoboda1337  Right!, thanks for that bit of info, @ssieb if  we add that to the PR and leave the null check (good practice anyway) we should be good.

swoboda1337, 2025-04-12:
Would remove the need for ugly looking ifdefs

ssieb, 2025-04-12:
> You can do this in ld2420 `__init__.py`, I tested it and it works:
> 
> ```
> DEPENDENCIES = ["uart", "select.ld2420"]
> ```

That's not quite correct, but it's close enough and with the nullptr check, it will compile and won't crash.  It's unlikely that someone will have two of these sensors on the same device.

ssieb, 2025-04-12:
@descipher if you make the PR, then I can review it.  Otherwise, I can edit mine if you want.

swoboda1337, 2025-04-12:
> That's not quite correct, but it's close enough and with the nullptr check, it will compile and won't crash.  It's unlikely that someone will have two of these sensors on the same device.

Right

descipher, 2025-04-13:
@ssieb I can do one later tonight, you can't put two of these sensors that close together anyway. They will interfere with each other.

descipher, 2025-04-13:
@ssieb  PR done https://github.com/esphome/esphome/pull/8555
I did not check for a null since it will always be set now.