esphome/issues #6933: qspi_amoled Missing __init__.py

andrewNiziolek, 2025-04-13:
### The problem

Hey, hoping I'm putting this in the right place but I'm trying to make use of the [qspi_amoled component](https://github.com/clydebarrow/esphome/tree/0d0673ec2f894c42d22876ff711ce63cff6824ad/esphome/components/qspi_amoled) and receive this error when trying to load it:

```
Failed config

external_components: [source /config/esphome/esphome-web-224800.yaml:11]
  - source: github://clydebarrow/esphome@i8080
    components: 
      - i80
      - io_bus
      
      Could not find __init__.py file for component qspi_amoled. Please check the component is defined by this source (search path: /data/external_components/fbb92d5f/esphome/components/qspi_amoled/__init__.py).
      - qspi_amoled
```

Have I missed something here? I'm not seeing any commits in later versions of ESPHome to suggest that it's a WIP.

### Which version of ESPHome has the issue?

2025.3.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

LilyGO T-Display S3 AMOLED Plus

### Component causing the issue

qspi_amoled

### YAML Config

```yaml
esphome:
  name: esphome-web-224800
  friendly_name: LilyGO-S3Plus-AMOLED
  min_version: 2024.11.0
  name_add_mac_suffix: false
  platformio_options:
    build_unflags: -Werror=all
    board_build.flash_mode: dio

external_components:
  - source: github://clydebarrow/esphome@i8080
    components: [ i80, io_bus, qspi_amoled ]

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf

logger:
  level: DEBUG
  logs:
    lvgl: WARN

api:

ota:
  platform: esphome

wifi:
  ssid: "secret"
  password: "secret"
  ap:
    ssid: "S3Plus Display"

captive_portal:

esp32_improv:
  authorizer: none

web_server:
  port: 80

i2c:
  sda: 3
  scl: 2

font:
  - file: "gfonts://Roboto"
    id: font1
    size: 20

i80:
  id: i80_display
  data_pins: [15, 14, 13, 12, 11, 10, 9, 8]  # OL_D0 to OL_D7
  write_pin: 47                              # OL_SCK (WRX)
  dc_pin: 46                                 # OL_DC
  cs_pin: 6                                  # OL_CS
  reset_pin: 17                              # OL_RESET
  clockless: true

display:
  - platform: qspi_amoled
    model: RM67162
    id: my_display
    update_interval: 1s
    dimensions:
      width: 536
      height: 240
    color_order: rgb
    io_bus: i80_display
    brightness: 255
    lambda: |-
      it.fill(COLOR_BLACK);
      it.print(0, 0, id(font1), COLOR_WHITE, "Plus model test!");
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.3.3
INFO Reading configuration /config/esphome/esphome-web-224800.yaml...
Failed config

external_components: [source /config/esphome/esphome-web-224800.yaml:11]
  - source: github://clydebarrow/esphome@i8080
    components: 
      - i80
      - io_bus
      
      Could not find __init__.py file for component qspi_amoled. Please check the component is defined by this source (search path: /data/external_components/fbb92d5f/esphome/components/qspi_amoled/__init__.py).
      - qspi_amoled
```

ssieb, 2025-04-13:
That component was renamed.  The `__init__.py` file should still be there though because without it, you get the error instead of the message about the rename.
https://github.com/esphome/esphome/pull/7594

andrewNiziolek, 2025-04-13:
Thanks for pointing this out @ssieb. My mistake.

closed 2025-04-13 by andrewNiziolek