esphome/issues #6935: Component name not showing in web v2, does is v3

loucks1, 2025-04-14:
### The problem

When using a component based on, [](https://github.com/loucks1/Directolor-ESPHome) the name is not shown when using the default (v2) webserver.  Shows correctly when using v3.

It shows correctly either way in Home Assistant.

### Which version of ESPHome has the issue?

2025.3.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.2

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

directolor_cover (derives from cover)

### YAML Config

```yaml
esphome:
  name: "esp32mp"
  friendly_name: ESPHome Media Player
  min_version: 2024.11.0
  name_add_mac_suffix: false

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

web_server:
 port: 80
# version: 3

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .lan

external_components:
  source: github://loucks1/Directolor-ESPHome
  refresh: 0s

nrf24l01_base:
  id: nrf24l01
  led_pin: 17
  ce_pin: 22
  cs_pin: 21

# Example configuration entry
cover:
  - platform: directolor_cover
    base: nrf24l01
    id: my_cover1
    name: compliant_name1
    radio_code:
      - 0x0B
      - 0x60
      - 0xB0
      - 0x05
    channel: 1
```

### Anything in the logs that might be useful for us?

```txt
no error messages or anything in the logs that seems related to this
```

### Additional information

[image]

[image]

wojciechkrol, 2025-05-05:
I am experiencing the same issue with version 2025.4.1.

luusl, 2025-05-10:
Same here. However, only the cover component seems to be affected. I also have a GPIO button in my configuration and its name appears correctly.