esphome/issues #6939: Component PN7150 NFC requires pin not available on device board

titchy, 2025-04-15:
### The problem

I have a Bi2C Easy NFC card that offers a PN7150 via a low profile i2c bus and a single additional wire for Interrupt.
The PN7150 NFC component requires a ven_pin be defined, but there is no such pin.
The ven_pin definition needs to be optional so that the device can be used.

### Which version of ESPHome has the issue?

2025.3.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.2

### What platform are you using?

ESP32

### Board

LILYGO T-Camera

### Component causing the issue

PN7150 nfc

### YAML Config

```yaml
pn7150_i2c:
  irq_pin: 45
  i2c_id: nfcI2C
  emulation_message: https://www.home-assistant.io/tag/pulse_ce
  tag_ttl: 1000ms
```

### Anything in the logs that might be useful for us?

```txt

```

randybb, 2025-04-15:
This one? https://ilabs.se/product/bi2c-easy-nfc/ "The module also need to have the interrupt signal connected which is a separate lead."

titchy, 2025-04-16:
Yes, that is the device I have.