esphome/issues #6940: Template Event - WDT reset in 2025.3.0+

Cossid, 2025-04-15:
### The problem

Updating from 2025.2.2 to 2025.3.0-2025.3.3, my switches which have a template event start WDT resetting within the first minute after boot, and continually WDT reset.

The issue did not exist on 2025.2.2 or before, and goes away if removing the `event` section on 2025.3.0+

It doesn't seem to matter if the events are ever triggered or not.

### Which version of ESPHome has the issue?

2025.3.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2025.4.2 (I don't think HA has anything to do with this though)

### What platform are you using?

BK72XX

### Board

WB3S

### Component causing the issue

template event

### YAML Config

```yaml
event:
  - platform: template
    id: event_scene_on
    name: Scene On
    event_types:
      - "scene_on"
  - platform: template
    id: event_scene_off
    name: Scene Off
    event_types:
      - "scene_off"
```

### Anything in the logs that might be useful for us?

```txt
Nothing seems to be emitted to the log, though the reset reason is always `WDT Reset`.  I have not had a chance to serial log or increase levels, as these devices are currently in-wall.
```

qwertz3, 2025-04-23:
Are you sure that it makes no difference whether the event has already been triggered?

I think there is a problem in the `web_server` component. If you call the web page without an event already triggered, this seems to lead to the dereferencing of a NULL pointer (`*(((event::Event *) (source))->last_event_type)`) in the function `event_all_json_generator`. Is this a problem? Unfortunately I am not a C++ professional.

In my case, the initial event stream gets stuck here. However, if I trigger an event after booting and then call the web page, everything works.
Perhaps you can test whether triggering an event after booting bypasses the error:

```
esphome:
  #...
  on_boot:
    - priority: -100
      then:
        - event.trigger:
            id: "event_scene_on"
            event_type: "scene_on"
```

Cossid, 2025-04-23:
Maybe I should have been more clear.  The crash happens within a minute of boot when the event is not triggered.  I was simply stating that using the event wasn't causing the problem, but rather it was the default case.  My events are set up for physical button presses, and I was not triggering them when the crashes were happening.  It's certainly possible that actually triggering them within a certain amount of time may avoid the crash.

nkinnan, 2025-05-07:
> I think there is a problem in the `web_server` component. If you call the web page without an event already triggered, this seems to lead to the dereferencing of a NULL pointer (`*(((event::Event *) [...]

How did you manage to discover this?  Did you decode the stack dump?

If that is indeed the actual line that is failing, then yes, it appears to be an inadvertent regression due to my change https://github.com/esphome/esphome/pull/7538  Thanks to the three of you for bringing it to my attention.  I will try to take a look as soon as I can, but it may not be immediate.  

Does this repro purely with the yaml that was shared, plus adding web_server at the root level?  Just hit the local webserver endpoint and it'll crash?

```yaml
web_server:
event:
  - platform: template
    id: event_scene_on
    name: Scene On
    event_types:
      - "scene_on"
  - platform: template
    id: event_scene_off
    name: Scene Off
    event_types:
      - "scene_off"
```

The quickest way to resolve it is if I can reproduce it easily, otherwise I'm sort of guessing at what's going on to trigger the problem.

Thanks!

AzonInc, 2025-05-07:
> Does this repro purely with the yaml that was shared? Just hit the local webserver endpoint and it'll crash?

yes that should reproduce it.

nkinnan, 2025-05-08:
Yup, that's almost certainly my change, I'll see what I can do.

Cossid, 2025-07-16:
@nkinnan - Any update on this?

AzonInc, 2025-07-17:
Give this a try:
```
external_components:
  - source: github://pr#9618
    components: [web_server]
```

Cossid, 2025-07-17:
@AzonInc fix looks likely correct, but I can't currently test due to being based on dev and incompatible with 2025.6.x, and there is an unrelated open bug for 2025.7.0 that is preventing compile for the devices I would be testing on.  I suspect it will be fixed when 2025.7.1 rolls around, and I'll test then.

Cossid, 2025-07-17:
Found the workaround for my other issue, and testing looks good, no startup crash, events are triggering.

AzonInc, 2025-07-17:
Perfect, so we can finally remove workarounds which were causing unwanted state changes