esphome/issues #6941: media_player plays for 1-2 seconds, then stops

efschu, 2025-04-16:
### The problem

latest esphome on esp32-box-s3
```
[15:17:26][D][media_player:074]: 'Media Player' - Setting
[15:17:26][D][media_player:081]:   Media URL: http://192.168.0.1:8123/api/esphome/ffmpeg_proxy/25ff07ffacb2452be5095a99ff4a8b0b/QGEsr2xffTo2NqOFjZFFww.flac
[15:17:26][D][micro_wake_word:194]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[15:17:26][D][speaker_media_player:426]: State changed to PLAYING
[15:17:26][D][micro_wake_word:129]: Stopping Microphone
[15:17:26][D][micro_wake_word:194]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[15:17:26][D][micro_wake_word:194]: State changed from STOPPING_MICROPHONE to IDLE
[15:17:26][D][ring_buffer:034][med_read]: Created ring buffer with size 1000000
[15:17:26][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[15:17:27][D][binary_sensor:036]: 'Presence detect': Sending state ON
[15:17:28][D][speaker_media_player.pipeline:124]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[15:17:29][D][ring_buffer:034]: Created ring buffer with size 19200
[15:17:29][D][speaker_mixer:310]: Starting speaker mixer
[15:17:29][D][speaker_mixer:318]: Started speaker mixer
[15:17:29][D][ring_buffer:034][speaker_task]: Created ring buffer with size 48000
[15:17:29][D][i2s_audio.speaker:111]: Starting Speaker
[15:17:29][D][i2s_audio.speaker:116]: Started Speaker
[15:17:29][D][binary_sensor:036]: 'Presence detect': Sending state OFF
[15:17:29][E][speaker_media_player:330]: The media pipeline's audio decoder encountered an error.
[15:17:30][D][speaker_mixer:323]: Stopping speaker mixer
```

### Which version of ESPHome has the issue?

from 2025.2.0 to 2025.4.0b3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.2

### What platform are you using?

ESP32

### Board

esp32-box-s3

### Component causing the issue

media_player

### YAML Config

```yaml
default config, only ip of homeassistant is configured
```

### Anything in the logs that might be useful for us?

```txt
[15:17:26][D][media_player:074]: 'Media Player' - Setting
[15:17:26][D][media_player:081]:   Media URL: http://192.168.0.1:8123/api/esphome/ffmpeg_proxy/25ff07ffacb2452be5095a99ff4a8b0b/QGEsr2xffTo2NqOFjZFFww.flac
[15:17:26][D][micro_wake_word:194]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[15:17:26][D][speaker_media_player:426]: State changed to PLAYING
[15:17:26][D][micro_wake_word:129]: Stopping Microphone
[15:17:26][D][micro_wake_word:194]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[15:17:26][D][micro_wake_word:194]: State changed from STOPPING_MICROPHONE to IDLE
[15:17:26][D][ring_buffer:034][med_read]: Created ring buffer with size 1000000
[15:17:26][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[15:17:27][D][binary_sensor:036]: 'Presence detect': Sending state ON
[15:17:28][D][speaker_media_player.pipeline:124]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[15:17:29][D][ring_buffer:034]: Created ring buffer with size 19200
[15:17:29][D][speaker_mixer:310]: Starting speaker mixer
[15:17:29][D][speaker_mixer:318]: Started speaker mixer
[15:17:29][D][ring_buffer:034][speaker_task]: Created ring buffer with size 48000
[15:17:29][D][i2s_audio.speaker:111]: Starting Speaker
[15:17:29][D][i2s_audio.speaker:116]: Started Speaker
[15:17:29][D][binary_sensor:036]: 'Presence detect': Sending state OFF
[15:17:29][E][speaker_media_player:330]: The media pipeline's audio decoder encountered an error.
[15:17:30][D][speaker_mixer:323]: Stopping speaker mixer
```

efschu, 2025-05-02:
Adding
`    buffer_size: 4000000`
`    task_stack_in_psram: true`
to media_player and changing
`      format: MP3`
`      sample_rate: 44100`
in announcement_pipeline made it work.

efschu, 2025-05-15:
and after a short while the error is back again

bitters1453, 2025-05-16:
I get a similar error occasionally. It might be related to wifi signal strength. I haven't done enough tests to prove that to myself, but I've only seen this happen when the box is where I want it. It never happens when I put the box in the same room as the access point.
ESP32-S3-Box-3
ESPHome 2025.4.1 is the current installed version, but it's been happening since I first setup the device in February
YAML copied from here
https://github.com/esphome/wake-word-voice-assistants/blob/main/esp32-s3-box-3/esp32-s3-box-3.yaml

[13:39:16][D][media_player:074]: 'Box-3' - Setting
[13:39:16][D][media_player:081]:   Media URL: http://192.168.0.250:8123/api/tts_proxy/ubXvLc5AzySdgODH_80UJg.flac
[13:39:16][D][media_player:087]:  Announcement: yes
[13:39:16][D][voice_assistant:505]: State changed from IDLE to STREAMING_RESPONSE
[13:39:16][D][voice_assistant:512]: Desired state set to STREAMING_RESPONSE
[13:39:16][D][micro_wake_word:194]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[13:39:16][D][speaker_media_player:426]: State changed to ANNOUNCING
[13:39:16][D][micro_wake_word:129]: Stopping Microphone
[13:39:16][D][micro_wake_word:194]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[13:39:16][D][micro_wake_word:194]: State changed from STOPPING_MICROPHONE to IDLE
[13:39:16][D][ring_buffer:034][ann_read]: Created ring buffer with size 1000000
[13:39:16][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[13:39:19][D][speaker_media_player.pipeline:124]: Decoded audio has 1 channels, 48000 Hz sample rate, and 16 bits per sample
[13:39:22][D][ring_buffer:034][speaker_task]: Created ring buffer with size 9600
[13:39:22][D][i2s_audio.speaker:111]: Starting Speaker
[13:39:22][D][i2s_audio.speaker:116]: Started Speaker
[13:39:22][E][speaker_media_player:341]: The announcement pipeline's audio decoder encountered an error.
[13:39:22][D][i2s_audio.speaker:123]: Stopping Speaker
[13:39:22][D][i2s_audio.speaker:129]: Stopped Speaker
[13:39:22][D][micro_wake_word:399]: Resetting buffers and probabilities
[13:39:22][D][micro_wake_word:194]: State changed from IDLE to START_MICROPHONE
[13:39:23][D][speaker_media_player:426]: State changed to IDLE

closed 2025-10-07 by github-actions[bot]