esphome/issues #6943: Psram error on atom5 echo with 2025.4

Anto79-ops, 2025-04-17:
### The problem

Hello,

Upon updating to 2025.4 from 2025.3, I received this error on failed compile:

```
Compiling .pioenvs/m5stack-atom-echo-a61210/src/esphome/components/psram/psram.o
src/esphome/components/psram/psram.cpp:5:10: fatal error: esp_psram.h: No such file or directory

*******************************************************************
* Looking for esp_psram.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_psram.h"
* Web  > https://registry.platformio.org/search?q=header:esp_psram.h
*
*******************************************************************

 #include <esp_psram.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/m5stack-atom-echo-a61210/src/esphome/components/psram/psram.o] Error 1
Compiling .pioenvs/m5stack-atom-echo-a61210/src/esphome/components/switch/switch.o
========================== [FAILED] Took 2.20 seconds ==========================
```

I'm using stock config for the Atom5 Echo

### Which version of ESPHome has the issue?

2025.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.2

### What platform are you using?

ESP32-IDF

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

Taomyn, 2025-04-17:
I confirm the same error after update ESPHome on my HAOS:

```
Compiling .pioenvs/m5ae-31b338/src/esphome/components/psram/psram.o
src/esphome/components/psram/psram.cpp:5:10: fatal error: esp_psram.h: No such file or directory

*******************************************************************
* Looking for esp_psram.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_psram.h"
* Web  > https://registry.platformio.org/search?q=header:esp_psram.h
*
*******************************************************************

 #include <esp_psram.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/m5ae-31b338/src/esphome/components/psram/psram.o] Error 1
========================= [FAILED] Took 27.16 seconds =========================
```

```
substitutions:
  name: "m5ae-31b338"
  friendly_name: M5Stack Atom Echo 31b338
  micro_wake_word_model: hey_jarvis # okay_nabu, hey_jarvis and hey_mycroft are supported
packages:
  # m5stack.atom-echo-voice-assistant: github://esphome/firmware/wake-word-voice-assistant/m5stack-atom-echo.adopted.yaml@main
  m5stack.atom-echo-voice-assistant: github://esphome/wake-word-voice-assistants/m5stack-atom-echo/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: !secret m5ae_key
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.39
    subnet: 255.255.255.0
    gateway: 192.168.1.1
    dns1: 192.168.1.11
    dns2: 192.168.1.71
  domain: !secret dns_suffix
  on_connect:
    - delay: 5s  # Gives time for improv results to be transmitted
http_request:
  verify_ssl: false
```

plplaaa2, 2025-04-17:
me too, ESP32 S3 devkitc
```
Compiling .pioenvs/voice-assistant-r1/src/esphome/components/ota/ota_backend_esp_idf.o
Compiling .pioenvs/voice-assistant-r1/src/esphome/components/psram/psram.o
src/esphome/components/psram/psram.cpp:5:10: fatal error: esp_psram.h: No such file or directory

*******************************************************************
* Looking for esp_psram.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_psram.h"
* Web  > https://registry.platformio.org/search?q=header:esp_psram.h
*
*******************************************************************

 #include <esp_psram.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/voice-assistant-r1/src/esphome/components/psram/psram.o] Error 1
========================= [FAILED] Took 82.84 seconds =========================
```

darth-hp, 2025-04-17:
Also see https://github.com/esphome/wake-word-voice-assistants/issues/98

RoGueMS, 2025-04-17:
I also report this error both on existing devices that try to update and on new devices that flash for the first time. Couldn't find any way around it, not even with preparing the first install on the device via ESPHome web and then trying to update via ESPHome Device Builder

`
Updating /config/esphome/m5stack-atom-echo-23a424.yaml
------------------------------------------------------------

INFO ESPHome 2025.4.0
INFO Reading configuration /config/esphome/m5stack-atom-echo-23a424.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing m5stack-atom-echo-23a424 (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
|-- ESPMicroSpeechFeatures @ 1.1.0
|-- esp-audio-libs @ 1.1.3
Compiling .pioenvs/m5stack-atom-echo-23a424/src/esphome/components/psram/psram.o
Compiling .pioenvs/m5stack-atom-echo-23a424/src/esphome/components/safe_mode/safe_mode.o
Compiling .pioenvs/m5stack-atom-echo-23a424/src/esphome/components/script/script.o
Compiling .pioenvs/m5stack-atom-echo-23a424/src/esphome/components/select/select.o
src/esphome/components/psram/psram.cpp:5:10: fatal error: esp_psram.h: No such file or directory

*******************************************************************
* Looking for esp_psram.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_psram.h"
* Web  > https://registry.platformio.org/search?q=header:esp_psram.h
*
*******************************************************************

 #include <esp_psram.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/m5stack-atom-echo-23a424/src/esphome/components/psram/psram.o] Error 1
========================= [FAILED] Took 18.44 seconds =========================
== [ERROR] /config/esphome/m5stack-atom-echo-23a424.yaml ==

======================== [SUMMARY] ========================
  - /config/esphome/m5stack-atom-echo-23a424.yaml: FAILED`

RoGueMS, 2025-04-22:
> Same still happening.
> 
> ```
> Compiling .pioenvs/m5stack-atom-echo-3049c0/src/esphome/components/psram/psram.o
> src/esphome/components/psram/psram.cpp:5:10: fatal error: esp_psram.h: No such file or directory
> 
> *******************************************************************
> * Looking for esp_psram.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:esp_psram.h"
> * Web  > https://registry.platformio.org/search?q=header:esp_psram.h
> *
> *******************************************************************
> 
>  #include <esp_psram.h>
>           ^~~~~~~~~~~~~
> compilation terminated.
> *** [.pioenvs/m5stack-atom-echo-3049c0/src/esphome/components/psram/psram.o] Error 1
> ========================= [FAILED] Took 247.08 seconds =========================
> ```

Try this:

```
substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo

packages: default - (i dont remember what was here from the top of my head)
  

**external_components:
  - source: github://esphome/esphome@2025.3.3
    components: [psram]**

esphome:
  name: ${name}
  name_add_mac_suffix: true
  friendly_name: ${friendly_name}
  min_version: 2025.2.0
```

Nathanator, 2025-04-23:
> external_components:
>   - source: github://esphome/esphome@2025.3.3
>     components: [psram]

That's the bit that just fixed it for me.  Thanks to @RoGueMS. I'll keep it in there until the issue is resolved.

davidlav, 2025-04-23:
Yep, that did it! Thanks so much for the quick reply, appreciate!

Clasper101, 2025-04-24:
Just remembered that the last update didn't work for me, tried again, still not working so tried the @RoGueMS suggestion and it has worked for me. Thank you :-)

closed 2025-04-29 by Anto79-ops

RoGueMS, 2025-04-29:
> 2025.4.2 has solved this issue. Thank you.

You mean 2025.4.1?

`2025.4.1
Fix psram below idf 5 [esphome#8584](https://github.com/esphome/esphome/pull/8584) by [@jesserockz](https://github.com/jesserockz)`

How did you manage to correct it now? It just Update ESPHome Builder and then remove the line from the YAML and save? Or did you need to recompile the whole thing?

Anto79-ops, 2025-04-29:
Corrected my post. It's was 2025.4.1 I just updated and it works