esphome/issues #6944: PSRAM Compatibility Issue In the last version

Code394, 2025-04-17:
### The problem

I am using an ESP32-POE-WROVER board which includes a WROVER module.
(In platformio, it is defined as esp32-poe).
However, in the latest ESPHome update, it was determined that only ESP32-S3 supports PSRAM mode, whereas in the past this worked on my esp32-poe board as well.

Essentially, there is a change in compatibility settings that prevents me from using the PSRAM configuration as I have used until now, despite the fact that my hardware supported it in previous versions.

### Which version of ESPHome has the issue?

2025.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.2

### What platform are you using?

ESP32

### Board

ESP32-POE-WROVER Olimex

### Component causing the issue

PSRAM - WROVER

### YAML Config

```yaml
esp32:
  board: 'esp32-poe'
  framework:
    type: arduino
  
# Example configuration entry
psram:
  mode: octal
  speed: 80MHz

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_OUT
  phy_addr: 0
  power_pin:
      number: GPIO12
      ignore_strapping_warning: true
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.4.0
INFO Reading configuration /config/esphome/screen-2.yaml...
Failed config

psram: [source /config/esphome/screen-2.yaml:26]
  
  Octal PSRAM is only supported on ESP32-S3, not ESP32.
  mode: octal
  speed: 80000000.0
  enable_ecc: False
```

Pmatsol58, 2025-04-17:
Having the same issue with ESP32-S2 module which does have psram.

**YAML**
```
esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino

psram:
  mode: octal
  speed: 80MHz

```
**LOG**
```
INFO ESPHome 2025.4.0
INFO Reading configuration /config/esphome/sensorbox.yaml...
Failed config

psram: [source /config/esphome/packages/display.yaml:6]
  
  Octal PSRAM is only supported on ESP32-S3, not ESP32S2.
  mode: octal
  speed: 80000000.0
  enable_ecc: False
```

Code394, 2025-04-18:
Hi @clydebarrow
Would you mind examining this issue when you have a moment?
I've observed that you recently merged a PSRAM-related modification [esphome#8519](https://github.com/esphome/esphome/pull/8519) in the latest stable release, which might be connected to the problem I'm experiencing.

clydebarrow, 2025-04-18:
The message seems pretty clear to me - octal PSRAM is only supported on ESP32-S3, neither of these chips is one of those, so if they do have PSRAM it  will be quad, not octal.

Just remove the mode line.

Pmatsol58, 2025-04-18:
Thanks for the update, I can confirm that either changing mode line to quad or removing it (since quad is default) works fine and no need for the external components section.

closed 2025-04-18 by clydebarrow