esphome/issues #6946: [lvgl] ESPHome 2025.4.0 id() for line widget return esphome::lvgl::LvLineType*

tdvtdv, 2025-04-17:
### The problem

Maybe it's not a bug, but just something I misunderstood.
Before ESPHome 2025.4.0 the id(somelinewidget) request returned a pointer to _lv_obj_t
which you could work with normally, but now what to do with 'esphome::lvgl::LvLineType* and how to get _lv_obj_t* from it?

### Which version of ESPHome has the issue?

ESPHome 2025.4.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

swoboda1337, 2025-04-17:
`id(whatever)->obj()`

tdvtdv, 2025-04-18:
Yes, this will work, but just **id(whatever)->obj**. obj is not a function, but this change doesn't make it any clearer. Other LVGL objects either return a pointer to _lv_obj_t or are automatically cast to the required type, and only 'line' behaves strangely in my case.

clydebarrow, 2025-05-02:
Yes, this changed due to added functionality for lines. Various other widgets have types that are not `lv_` derived types, and similar changes may occur in the future, but like this one, won't affect any yaml config.

Why do you need the id? What can't you do in YAML?

closed 2025-05-02 by clydebarrow