esphome/issues #6950: Shelly 2PM Gen3 with ESP-Shelly-C38F boot loop

florix121, 2025-04-18:
### The problem

Hello,

I hope I've come to the right place with my problem.

I have tried to install esphome on the Shelly 2PM Gen3. According to the manufacturer, an ESP-Shelly-C38F is installed in it. Therefore I used the configuration of the Shelly 1PM Gen3 with the same chip (https://devices.esphome.io/devices/Shelly-1PM-Gen3).

After the installation, however, the esp was boot-looping and restarted every time after `Enabling STA.`. 
The same happened with `esp-idf` framework instead of `arduino`, as shown in the log.

### Which version of ESPHome has the issue?

2025.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.2

### What platform are you using?

ESP32

### Board

Shelly 2PM Gen3 with ESP-Shelly-C38F

### Component causing the issue

wifi

### YAML Config

```yaml
esphome:
  name: shelly2pm-test
  platformio_options:
    board_build.mcu: esp32c3
    board_build.variant: esp32c3
    board_build.f_cpu: 160000000L
    board_build.f_flash: 80000000L
    board_build.flash_mode: dio
    board_build.flash_size: 8MB

esp32:
  variant: esp32c3
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

logger:
  level: VERY_VERBOSE
  hardware_uart: UART0

wifi:
  ssid: !secret ssid
  password: !secret password
```

### Anything in the logs that might be useful for us?

```txt
[16:35:29]I (25) boot: ESP-IDF 5.1.5 2nd stage bootloader
[16:35:29]I (26) boot: compile time Apr 18 2025 16:33:50
[16:35:29]I (26) boot: chip revision: v0.4
[16:35:29]I (28) boot.esp32c3: SPI Speed      : 80MHz
[16:35:29]I (33) boot.esp32c3: SPI Mode       : DIO
[16:35:29]I (38) boot.esp32c3: SPI Flash Size : 4MB
[16:35:29]I (43) boot: Enabling RNG early entropy source...
[16:35:29]I (48) boot: Partition Table:
[16:35:29]I (52) boot: ## Label            Usage          Type ST Offset   Length
[16:35:29]I (59) boot:  0 otadata          OTA data         01 00 00009000 00002000
[16:35:29]I (66) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[16:35:29]I (74) boot:  2 app0             OTA app          00 10 00010000 001c0000
[16:35:29]I (81) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[16:35:29]I (89) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[16:35:29]I (96) boot: End of partition table
[16:35:29]I (101) boot: No factory image, trying OTA 0
[16:35:29]I (105) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=20cb0h (134320) map
[16:35:29]I (136) esp_image: segment 1: paddr=00030cd8 vaddr=3fc91600 size=029c0h ( 10688) load
[16:35:29]I (138) esp_image: segment 2: paddr=000336a0 vaddr=40380000 size=0c978h ( 51576) load
[16:35:29]I (151) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=840c0h (540864) map
[16:35:30]I (240) esp_image: segment 4: paddr=000c40e8 vaddr=4038c978 size=04c2ch ( 19500) load
[16:35:30]I (249) boot: Loaded app from partition at offset 0x10000
[16:35:30]I (273) boot: Set actual ota_seq=1 in otadata[0]
[16:35:30]I (273) boot: Disabling RNG early entropy source...
[16:35:30]I (284) cpu_start: Unicore app
[16:35:30]I (284) cpu_start: Pro cpu up.
[16:35:30]I (292) cpu_start: Pro cpu start user code
[16:35:30]I (292) cpu_start: cpu freq: 160000000 Hz
[16:35:30]I (292) cpu_start: Application information:
[16:35:30]I (295) cpu_start: Project name:     shelly2pm-test
[16:35:30]I (301) cpu_start: App version:      2025.4.0
[16:35:30]I (306) cpu_start: Compile time:     Apr 18 2025 16:32:27
[16:35:30]I (312) cpu_start: ELF file SHA256:  761498b04dc888c2...
[16:35:30]I (318) cpu_start: ESP-IDF:          5.1.5
[16:35:30]I (323) cpu_start: Min chip rev:     v0.3
[16:35:30]I (328) cpu_start: Max chip rev:     v1.99 
[16:35:30]I (332) cpu_start: Chip rev:         v0.4
[16:35:30]I (337) heap_init: Initializing. RAM available for dynamic allocation:
[16:35:30]I (344) heap_init: At 3FC98420 len 00027BE0 (158 KiB): DRAM
[16:35:30]I (351) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
[16:35:30]I (358) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
[16:35:30]I (365) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
[16:35:30]I (372) spi_flash: detected chip: gd
[16:35:30]I (376) spi_flash: flash io: dio
[16:35:30]W (380) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
[16:35:30]I (393) sleep: Configure to isolate all GPIO pins in sleep state
[16:35:30]I (400) sleep: Enable automatic switching of GPIO sleep configuration
[16:35:30]I (407) app_start: Starting scheduler on CPU0
[16:35:30]I (412) main_task: Started on CPU0
[16:35:30]I (416) main_task: Calling app_main()
[16:35:30][I][logger:171]: Log initialized
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[VV][esp-idf:000][main]: I (489) main_task: Returned from app_main()

[C][wifi:048]: Setting up WiFi...
[VV][esp-idf:000]: I (493) pp: pp rom version: 9387209

[VV][esp-idf:000]: I (493) net80211: net80211 rom version: 9387209

[VV][esp-idf:000]: I (504) wifi:
[VV][esp-idf:000]: wifi driver task: 3fca4f6c, prio:23, stack:6656, core=0
[VV][esp-idf:000]: 

[VV][esp-idf:000][wifi]: I (514) wifi:
[VV][esp-idf:000][wifi]: wifi firmware version: bbd58fc31
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000][wifi]: I (533) wifi:
[VV][esp-idf:000][wifi]: wifi certification version: v7.0
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000][wifi]: I (553) wifi:
[VV][esp-idf:000][wifi]: config NVS flash: enabled
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000][wifi]: I (574) wifi:
[VV][esp-idf:000][wifi]: config nano formating: disabled
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000][wifi]: I (584) wifi:
[VV][esp-idf:000][wifi]: Init data frame dynamic rx buffer num: 32
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000][wifi]: I (605) wifi:
[VV][esp-idf:000][wifi]: Init static rx mgmt buffer num: 5
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000][wifi]: I (625) wifi:
[VV][esp-idf:000][wifi]: Init management short buffer num: 32
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000][wifi]: I (635) wifi:
[VV][esp-idf:000][wifi]: Init dynamic tx buffer num: 32
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000][wifi]: I (656) wifi:
[VV][esp-idf:000][wifi]: Init static tx FG buffer num: 2
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000][wifi]: I (676) wifi:
[VV][esp-idf:000][wifi]: Init static rx buffer size: 1600
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000][wifi]: I (686) wifi:
[VV][esp-idf:000][wifi]: Init static rx buffer num: 10
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000][wifi]: I (707) wifi:
[VV][esp-idf:000][wifi]: Init dynamic rx buffer num: 32
[VV][esp-idf:000][wifi]: 

[VV][esp-idf:000]: I (728) wifi_init: rx ba win: 6

[VV][esp-idf:000]: I (728) wifi_init: tcpip mbox: 32

[VV][esp-idf:000]: I (738) wifi_init: udp mbox: 6

[VV][esp-idf:000]: I (738) wifi_init: tcp mbox: 6

[VV][esp-idf:000]: I (748) wifi_init: tcp tx win: 5760

[VV][esp-idf:000]: I (758) wifi_init: tcp rx win: 5760

[VV][esp-idf:000]: I (768) wifi_init: tcp mss: 1440

[VV][esp-idf:000]: I (768) wifi_init: WiFi IRAM OP enabled

[VV][esp-idf:000]: I (778) wifi_init: WiFi RX IRAM OP enabled

[C][wifi:061]: Starting WiFi...
[C][wifi:062]:   Local MAC: XX:XX:XX:XX:XX:XX
[V][esp32.preferences:059]: nvs_get_blob('2120106733'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][wifi_esp32:222]: Enabling STA.
[VV][esp-idf:000]0;32mI (26) boot: ESP-IDF 5.1.5 2nd stage bootloader
```

Jason2866, 2025-04-19:
The setting of `8MB` flash is not working. The Bootlog shows `[16:35:30]W (380) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.` which is wrong and does end in havoc

florix121, 2025-04-21:
Thank you for the quick answers.

I didn't manage to use the `board.json` file from [#6455 (comment)](https://github.com/esphome/issues/issues/6455#issuecomment-2533682358) in my Esphome and therefore rebuilt it as best I could in the Esphome `.yaml`:

```
esphome:
  name: shelly2pm-test
  platformio_options:
    board_build.partitions: "/config/esphome/shelly2pm.csv"
    board_build.core: esp32
    board_build.f_cpu: 160000000L
    board_build.f_flash: 40000000L
    board_build.flash_mode: dio
    board_build.extra_flags: "-DARDUINO_ESP32C3_DEV"
    board_build.mcu: esp32c3
    board_build.variant: esp32c3

esp32:
  variant: esp32c3
  board: esp32-c3-devkitm-1
  flash_size: 8MB
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_PERF: y
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: n
      CONFIG_BT_GATTC_CACHE_NVS_FLASH: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"
      CONFIG_WIFI_PROV_STA_FAST_SCAN: y
```

This then fixed the flash-size error, but unfortunately could not fix the boot-loop:

```
[19:53:26]I (30) boot: ESP-IDF 5.3.1 2nd stage bootloader
[19:53:26]I (30) boot: compile time Apr 21 2025 19:51:10
[19:53:26]I (30) boot: chip revision: v0.4
[19:53:26]I (33) boot.esp32c3: SPI Speed      : 40MHz
[19:53:26]I (38) boot.esp32c3: SPI Mode       : DIO
[19:53:26]I (43) boot.esp32c3: SPI Flash Size : 8MB
[19:53:26]I (47) boot: Enabling RNG early entropy source...
[19:53:26]I (53) boot: Partition Table:
[19:53:26]I (56) boot: ## Label            Usage          Type ST Offset   Length
[19:53:26]I (64) boot:  0 otadata          OTA data         01 00 00009000 00002000
[19:53:26]I (71) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[19:53:26]I (79) boot:  2 app0             OTA app          00 10 00010000 00200000
[19:53:26]I (86) boot:  3 app1             OTA app          00 11 00210000 00200000
[19:53:26]I (94) boot:  4 nvs              WiFi data        01 02 00410000 00010000
[19:53:26]I (101) boot:  5 nvs2             WiFi data        01 02 00420000 003e0000
[19:53:26]I (109) boot: End of partition table
[19:53:26]I (113) boot: No factory image, trying OTA 0
[19:53:26]I (118) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=20f24h (134948) map
[19:53:26]I (155) esp_image: segment 1: paddr=00030f4c vaddr=3fc92000 size=02d10h ( 11536) load
[19:53:26]I (159) esp_image: segment 2: paddr=00033c64 vaddr=40380000 size=0c3b4h ( 50100) load
[19:53:26]I (173) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=855d8h (546264) map
[19:53:26]I (291) esp_image: segment 4: paddr=000c5600 vaddr=4038c3b4 size=05b3ch ( 23356) load
[19:53:26]I (303) boot: Loaded app from partition at offset 0x10000
[19:53:26]I (327) boot: Set actual ota_seq=1 in otadata[0]
[19:53:26]I (327) boot: Disabling RNG early entropy source...
[19:53:26]I (338) cpu_start: Unicore app
[19:53:26]I (347) cpu_start: Pro cpu start user code
[19:53:26]I (347) cpu_start: cpu freq: 160000000 Hz
[19:53:26]I (347) app_init: Application information:
[19:53:26]I (350) app_init: Project name:     shelly2pm-test
[19:53:26]I (356) app_init: App version:      2025.4.0
[19:53:26]I (360) app_init: Compile time:     Apr 21 2025 19:49:21
[... 77 lines cut ...]

[VV][esp-idf:000]: I (719) wifi_init: accept mbox: 6

[VV][esp-idf:000]: I (728) wifi_init: tcpip mbox: 32

[VV][esp-idf:000]: I (728) wifi_init: udp mbox: 6

[VV][esp-idf:000]: I (738) wifi_init: tcp mbox: 6

[VV][esp-idf:000]: I (748) wifi_init: tcp tx win: 5760

[VV][esp-idf:000]: I (758) wifi_init: tcp rx win: 5760

[VV][esp-idf:000]: I (759) wifi_init: tcp mss: 1440

[VV][esp-idf:000]: I (769) wifi_init: WiFi IRAM OP enabled

[VV][esp-idf:000]: I (779) wifi_init: WiFi RX IRAM OP enabled

[C][wifi:061]: Starting WiFi...
[C][wifi:062]:   Local MAC: XX:XX:XX:XX:XX:XX
[V][esp32.preferences:059]: nvs_get_blob('434255655'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][wifi_esp32:222]: Enabling STA.
[VV][esp-idf:000][wifi]: I (800) phy_init: pI (30) boot: ESP-IDF 5.3.1 2nd stage bootloader
```

stikonas, 2025-05-28:
@florix121 Hmm, people here https://community.home-assistant.io/t/do-you-have-a-working-esphome-config-for-shelly-3gen-2pm/833575/7 don't seem to have you problem, maybe worth trying that config?

florix121, 2025-05-31:
Thanks for the hint @stikonas, but I still seem to be doing something wrong. I first tried the config without any changes. Since the Shelly did not connect to the WiFi and did not open an AP, I then added `hardware_uart: UART0` to the `logger` to be able to read the log. Unfortunately, the Shelly again restarts after `Starting WiFi.` and gets into a bootloop.

stikonas, 2025-05-31:
How are you powering them?

stikonas, 2025-05-31:
If it is just UART power, than it is simply not enough to power both chip and WiFi. Though make sure you don't connect mains and UART simultaneously.

florix121, 2025-05-31:
Oh, of course I hadn't thought of that. Thank you for pointing that out. With the configuration from here https://community.home-assistant.io/t/do-you-have-a-working-esphome-config-for-shelly-3gen-2pm/833575/7 and connected to mains, everything works without any problems.

Thanks again to everyone who has helped me here!