esphome/issues #6952: Cannot set humidity offset for BME280 BME680 senors

domerich, 2025-04-19:
### The problem

Hi guys, I bought an accurate hydrometer in order to check the BME280 and BM680 sensors that I am using in my home. While I can adjust temperature offset, how can I deal with humidity offset? There is is like a 20% absolute shift in rH. 

platform: bme680_bsec

[image]

### Which version of ESPHome has the issue?

 Firmware: 4.3.12 (ESPHome 2024.12.4) 

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

domerich, 2025-04-19:
it seems one way is to offset the temperature, as it will offset everything else. but i have read it is also possible to change the humidity readings through a sensor filter, how does this work? I also came across this [article](https://community.home-assistant.io/t/bme680-temperature-offset-impacts-the-humidity/802716/12).

swoboda1337, 2025-04-21:
https://esphome.io/components/sensor/index.html#sensor-filters