esphome/issues #6955: Can't connect to ESPHome API for xxxx.local: Timeout while resolving IP address for ['xxxx.local']

percy10442, 2025-04-21:
### The problem

After upgrading to ESPHome 4.0, I couldn't resolve the IP address in the ESPHome panel, so I couldn't upgrade. However, everything works fine and I can upgrade using the web server. Additionally, Home Assistant control is functioning normally.

### Which version of ESPHome has the issue?

2025.4.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### YAML Config

```yaml
substitutions:
  devicename: "corridor1f2"

esphome:
  name: $devicename
  friendly_name: 1樓走廊前

esp8266:
  board: esp12e

# Enable logging
logger:
  baud_rate: 0
# Enable Home Assistant API
api:
  encryption:
    key: "vdNwn6bfc5mVpGXxhwr9qpV49wFL4gwV2Wgb4rGFw3U="

ota:
  - platform: esphome
    password: "2084b6542efc893469cbbef7e0d9b6af"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${devicename} Fallback Hotspot"
    password: "LdSDQV7kVMfE"

captive_portal:

web_server:
  port: 80

http_request:
  useragent: esphome/device
  id: http_request_data
  verify_ssl: false

i2c:
  sda: 13
  scl: 14
  scan: true
  id: bus_a

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:

text_sensor:
  - platform: template
    id: corridor1f1_Presence2    
  - platform: ld2410
    version:
      name: "firmware version"
    mac_address:
      name: "mac address"

sensor:
  - platform: bh1750
    name: "${devicename} illuminance"
    id: ill
    address: 0x23
    update_interval: 1s
  - platform: ld2410
    moving_distance:
      name : ${devicename} Moving Distance
    still_distance:
      name: ${devicename} Still Distance
    moving_energy:
      name: ${devicename} Move Energy
    still_energy:
      name: ${devicename} Still Energy
    detection_distance:
      name: ${devicename} Detection Distance

binary_sensor:
  - platform: ld2410
    has_target:
      name: ${devicename} Presence
  - platform: gpio
    pin: 12
    name: "${devicename} Presence2"
    id: Presence2
    device_class: motion
    on_press:
      then:
        - if:
            condition:
              lambda: 'return id(ill).state < 2.2;'
            then:
              - switch.turn_on: ${devicename}_Relay
    on_release:
      then:
        - http_request.get:
            url: http://corridor1f1.local/binary_sensor/corridor1f1_presence2
            capture_response: true
            on_response:
            - if:
                condition:
                    lambda: return response->status_code == 200;
                then:
                  - lambda: |-
                      json::parse_json(body, [](JsonObject root) -> bool {
                          id(corridor1f1_Presence2).publish_state(root["state"]);
                          return true;
                      });
                  - if:
                      condition:
                        - lambda: return id(corridor1f1_Presence2).state != "ON";     
                      then:        
                        - switch.turn_off: ${devicename}_Relay
                        - http_request.post: http://corridor1f1.local/switch/corridor1f1_relay/turn_off
                else: 
                    - lambda: |-
                        id(corridor1f1_Presence2).publish_state("ERROR");       
                    - switch.turn_off: ${devicename}_Relay                    
            on_error:
              then:
                - lambda: |-
                    id(corridor1f1_Presence2).publish_state("ERROR");       
                - switch.turn_off: ${devicename}_Relay         

  - platform: gpio
    pin: 4
    name: "${devicename} motion"
    id: motion
    device_class: motion
    on_press:
      then:
        if:
          condition:
            and:
                - binary_sensor.is_on: Presence2
                - lambda: return id(ill).state < 2.2;
          then:
              - switch.turn_on: ${devicename}_Relay

switch:
  - platform: gpio
    name: "${devicename} Relay"
    pin: 5
    id: ${devicename}_Relay
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

[image]

[image]

1nutzername, 2025-04-26:
I experienced the same with my ESP32 devices. I rebuilt my firmware and tried to flash it via cable with the help of the following website: https://web.esphome.io/?dashboard_install. When I flash the "OTA" image, my device doesn't connect to the Wi-Fi, so OTA doesn't work, and my device cannot be found. On the other hand, when I flash the "factory" image, it successfully connects to my Wi-Fi. 

I've also tried ESPHome v3.3, 3.0, and 2.0, the issue persists.

The device I'm testing with is a very small fan control unit with the following dependency tree:
```
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
```

BR

DAB-LABS, 2025-05-02:
ESPHome runs on mDNS. Your network isn’t resolving the mDNS to an ip address. It’s not an ESPHome issue.