esphome/issues #6957: MQTT prevents Improv Serial from loading on ESP32 C3

MikkoTervala, 2025-04-21:
### The problem

When MQTT is enabled, startup logs get cut and improv_serial does not work. Adding a delay to MQTT startup fixes the issue.

Failing configuration:

``` yaml
esphome:
  name: lighthouse
  friendly_name: Lighthouse
  name_add_mac_suffix: true

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

improv_serial:

wifi:
  ap:

mqtt:
  id: mqtt_client
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password

logger:

```

With this configuration, many configuration startup logs go missing

Improv does not work, `An error occurred. Improv Wi-Fi Serial not detected`

[image]

```
[20:33:00]ESP-ROM:esp32c3-api1-20210207
[20:33:00]Build:Feb  7 2021
[20:33:00]rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
[20:33:00]Saved PC:0x40380ff6
[20:33:00]SPIWP:0xee
[20:33:00]mode:DIO, clock div:1
[20:33:00]load:0x3fcd5820,len:0x1704
[20:33:00]load:0x403cc710,len:0x938
[20:33:00]load:0x403ce710,len:0x2ef4
[20:33:00]entry 0x403cc710
[20:33:00]I (24) boot: ESP-IDF 5.1.5 2nd stage bootloader
[20:33:00]I (24) boot: compile time Apr 21 2025 20:32:17
[20:33:00]I (25) boot: chip revision: v0.4
[20:33:00]I (27) boot.esp32c3: SPI Speed      : 80MHz
[20:33:00]I (32) boot.esp32c3: SPI Mode       : DIO
[20:33:00]I (37) boot.esp32c3: SPI Flash Size : 4MB
[20:33:00]I (41) boot: Enabling RNG early entropy source...
[20:33:00]I (47) boot: Partition Table:
[20:33:00]I (50) boot: ## Label            Usage          Type ST Offset   Length
[20:33:00]I (58) boot:  0 otadata          OTA data         01 00 00009000 00002000
[20:33:00]I (65) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[20:33:00]I (73) boot:  2 app0             OTA app          00 10 00010000 001c0000
[20:33:00]I (80) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[20:33:00]I (88) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[20:33:00]I (95) boot: End of partition table
[20:33:00]I (99) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=27760h (161632) map
[20:33:00]I (133) esp_image: segment 1: paddr=00037788 vaddr=3fc91600 size=02990h ( 10640) load
[20:33:00]I (136) esp_image: segment 2: paddr=0003a120 vaddr=40380000 size=05ef8h ( 24312) load
[20:33:00]I (144) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=aafdch (700380) map
[20:33:00]I (259) esp_image: segment 4: paddr=000eb004 vaddr=40385ef8 size=0b6cch ( 46796) load
[20:33:00]I (274) boot: Loaded app from partition at offset 0x10000
[... 442 lines cut ...]
Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1

### Component causing the issue

improv_serial mqtt

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-04-21:
I think the problem is that mqtt is blocking setup if it can't connect.

MikkoTervala, 2025-04-22:
Yes this does seem to be the case. I verified by using a configuration where MQTT loads during boot. When a WiFi connection has been configured and the device connects, all components load and improv_serial is shown in the logs.

I made a PR to improve the docs on this https://github.com/esphome/esphome-docs/pull/4838

Thank you for your help!

closed 2025-04-22 by MikkoTervala

DjordjeMandic, 2025-05-16:
Can you test if same happens with this https://github.com/esphome/esphome/pull/8325 added to your config?

```yaml
external_components:
  # equivalent shorthand for GitHub pull request
  - source: github://pr#8325
    components: [ mqtt ]
```

MikkoTervala, 2025-05-22:
> Can you test if same happens with this [esphome/esphome#8325](https://github.com/esphome/esphome/pull/8325) added to your config?
> 
> external_components:
>   # equivalent shorthand for GitHub pull request
>   - source: github://pr#8325
>     components: [ mqtt ]

I would be happy to test this! I'm not available now, but will test it out later next week. Will report back when I have been able to try it.

jamesmyatt, 2025-07-09:
Thanks for fixing this. Caused me *huge* headaches when I changed my home network and wifi set-up.