esphome/issues #6964: Speaker Audio Media Player on esp32s3

mad-tunes, 2025-04-24:
### The problem

After tons of trouble keeping my smart speaker/voice assistant working with recent esphome and HA core updates, I was really excited to stumble across mention of Speaker Audio Media Player and to give it a go.
Unfortunately I haven't been able to get it to make any sound at all, using a local .flac as in the example here:
(https://esphome.io/components/media_player/speaker#media-player-speaker-play-on-device-media-file-action)

I'm using the same hardware connected in the same ways that worked OK for esphome and HA core versions from near the end of last year (using the arduino framework as everything-media player seemed to break with idf ages ago now).

### Which version of ESPHome has the issue?

2025.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.3

### What platform are you using?

ESP32-IDF

### Board

esp32-s3

### Component causing the issue

Speaker Audio Media Player

### YAML Config

```yaml
substitutions:
  friendly_name: Jarvis2
  host_name: jarvis2

  led_brightness:     25%       # RGB LED brightness for simple on/off
  min_led_brightness: 20%       # Min LED brghtness for effects
  max_led_brightness: 40%       # Max LED brghtness for effects

  log_level:      DEBUG         # NONE, ERROR, WARN, INFO, DEBUG (Default), VERBOSE, VERY_VERBOSE

  static_ip:      192.168.5.36  # Static ID details
  static_gateway: 192.168.5.254
  static_subnet:  255.255.255.0
  static_dns:     192.168.5.2

  vol_up_pin:     GPIO12        # Button for Vol+
  vol_down_pin:   GPIO11        # Button for Vol-

  amp_din_pin:    GPIO04        # Amp DIN
  amp_bclk_pin:   GPIO05        # Amp BCLK
  amp_lrc_pin:    GPIO06        # Amp LRC

  rgb_led_pin:    GPIO21        # RGB LED pin, GPIO21 for onboard LED
  no_leds:        "1"           # Number of RGB LEDs, 1 for onboard LED

api:                                    # Enable Home Assistant API.
  encryption:
    key: !secret api_encryption_key

ota:
  platform: esphome # For 2024.6.1+
  password: !secret ota_password

safe_mode: # For 2024.6.1+

[... 357 lines cut ...]
22:49:01	[D]	[ring_buffer:034][1;31m[speaker_task][0;36m	
Created ring buffer with size 48000
22:49:01	[D]	[i2s_audio.speaker:111]	
Starting Speaker
22:49:01	[D]	[i2s_audio.speaker:116]	
Started Speaker
22:49:01	[D]	[speaker_media_player:426]	
State changed to IDLE
22:49:01	[D]	[speaker_mixer:323]	
Stopping speaker mixer
22:49:02	[D]	[i2s_audio.speaker:123]	
Stopping Speaker
22:49:02	[D]	[i2s_audio.speaker:129]	
Stopped Speaker
```

### Additional information

Other than basics like board definition, wifi setup, debug stuff I've added after first finding the issue, I've copied/pasted the example.

Should this be working on an esp32-s3?
Can anyone spot a mistake I've made?
Is there anything that anyone can suggest to get this working?

Thanks for any help that anyone can offer

George1422, 2025-04-30:
Your configuration is very similar to mine, we must have found similar resources to mimic. https://github.com/esphome/issues/issues/6980  I can say my config works, with a caveat in that if the media player is uses then VA will fail. But both VA and media player work well as long as you don't use them together. 

The only real differences I see mine has compiler options at the top I don't see in your configuration. 

For the mixer I define the number of channels
`num_channels: "2"`

And a few other setting in the media player. From what you have it "should" work. 

I also have the debug level cranked up on mine trying to understand why and where its failing with my issue. Since you are not using VA it shouldn't impact you.

mad-tunes, 2025-04-30:
I raised similar on the [HA forum](https://community.home-assistant.io/t/speaker-audio-media-player-in-esp32-s3-zero/881958) and was pointed towards a better setup that works pretty well for me, [here](https://community.home-assistant.io/t/esp32-s3-and-voice-assistant-config/878963).

Seems to work pretty much perfectly except that if I have music playing and send an announcement, the music's ducked down but the announcement doesn't play. I rarely play music through my speaker though, so that isn't a dealbreaker for me.

My config is now this:
`# For
# ESP32S3-Zero board https://www.aliexpress.com/item/1005006524672028.html
# MAX98357 I2S 3W amp https://www.aliexpress.com/item/1005006382608935.html
# INMP441 I2S mic

substitutions:
  friendly_name: Jarvis
  host_name: jarvis

  #pins
  amp_lrc_pin:        GPIO06        # LRC on Max98357
  amp_bclk_pin:       GPIO05        # BCLK on Max98357
  mic_ws_pin:         GPIO08        # WS on INMP441
  mic_ck_pin:         GPIO09        # SLK on INMP441
  rgb_led_pin:        GPIO21        # on-board LED
  amp_din_pin:        GPIO04        # DIN on Max98357
  mic_din_pin:        GPIO07        # SD on INMP441

  no_leds:            "1"           # Number of RGB LEDs, 1 for onboard LED
  led_brightness:     25%           # RGB LED brightness for simple on/off
  min_led_brightness: 20%           # Min LED brghtness for effects
  max_led_brightness: 40%           # Max LED brghtness for effects

  assist_mic_gain:          31dBFS  # 31dBFS # Assistant mic gain. Between 0dBFS and 31dBFS
  assist_noise_supression:  "4"     # Assistant noise suppression. Between 0 and 4 inclusive
  assist_vol_multiplier:    "8"     # "15" # Assistant volume multiplier. Must be larger than 0. Defaults to 1 (disabled)

  log_level:          ERROR         # NONE, ERROR, WARN, INFO, DEBUG (Default), VERBOSE, VERY_VERBOSE

  static_ip:          192.168.5.35  # Static IP address
  static_gateway:     192.168.5.254 # Static gateway address
  static_subnet:      255.255.255.0 # Static subnet
  static_dns:         192.168.5.2   # Static DNS address

  # Sounds
  # connected_sound:         "http://192.168.5.11:8123/local/Voice_assistant/jarvis_connected.mp3"
  # stop_listening_sound:    "http://192.168.5.11:8123/local/Voice_assistant/ill_stop_listening.mp3"
  # start_listening_sound:   "http://192.168.5.11:8123/local/Voice_assistant/im_listening.mp3"
  pip_sound:               "http://192.168.5.11:8123/local/Voice_assistant/pip.mp3"
  test_sound:              "http://192.168.5.11:8123/local/Voice_assistant/testing_1_2_3_testing.mp3"

  # Phases of the Voice AssistantGPIO8
  # IDLE: The voice assistant is ready to be triggered by a wake-word
  voice_assist_idle_phase_id: '1'
  # LISTENING: The voice assistant is ready to listen to a voice command (after being triggered by the wake word)
  voice_assist_listening_phase_id: '2'
  # THINKING: The voice assistant is currently processing the command
  voice_assist_thinking_phase_id: '3'
  # REPLYING: The voice assistant is replying to the command
  voice_assist_replying_phase_id: '4'
  # NOT_READY: The voice assistant is not ready 
  voice_assist_not_ready_phase_id: '10'
  # ERROR: The voice assistant encountered an error
  voice_assist_error_phase_id: '11'  
  # MUTED: The voice assistant is muted and will not reply to a wake-word
  voice_assist_muted_phase_id: '12'

#psram:
#  mode: octal
#  speed: 80MHz
psram:
  mode: quad
  speed: 80MHz

packages:
[... 603 lines cut ...]
    on_press:
      - media_player.play_media: $test_sound
  - platform: template
    name: Wake flac
    id: wake_flac
    icon: "mdi:speaker-play"
    disabled_by_default: true           # Shows entity in HA, but disabled by default
    on_press:
      - script.execute:
          id: play_sound
          priority: true
          sound_file: !lambda return id(wake_word_triggered_sound);
  - platform: template
    name: Pip flac
    id: pip_flac
    icon: "mdi:speaker-play"
    disabled_by_default: true           # Shows entity in HA, but disabled by default
    on_press:
      - script.execute:
          id: play_sound
          priority: true
          sound_file: !lambda return id(pip_sound);`
packages: doesn't include anything exotic, just the basics that I have common to most/all esphome .yamls

I would still like to understand what makes the difference with the media player, but haven't found it yet. It'd also be good if the example 'just worked', as anyone might expect.

George1422, 2025-04-30:
> Seems to work pretty much perfectly except that if I have music playing and send an announcement, the music's ducked down but the announcement doesn't play. I rarely play music through my speaker though, so that isn't a dealbreaker for me.

This sounds like the exact problem I'm having. 

Could you try this if you have the time: turn up the logging debug level and redeploy the image to your player. Leave the log running. Then in another web page send an announcement to verify it working. You will see a lot of text in the log that is OK. Just to prove the announcements work. Now send some music to the player and verify music works. Stop the music and then send another announcement. Does that announcement come through without the music playing? In the log do you see any red text that talks about out of memory? My announcements are coming through the voice assistant, but I think the data path is the same.

mad-tunes, 2025-04-30:
I havent watched the logs at the time, but can confirm that things play OK whenever I have either announcements or music playing -just not when I have both at the same time.
I'm fiddling with it at the moment, and when I try both I get this in logs:
`[10:46:40][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_ERR_NO_MEM
[10:46:40][E][speaker_media_player:339]: The announcement pipeline's file reader encountered an error.`

I've tried task_stack_in_psram: true, but still get the same. Also stripped anything I can out of the .yaml (like the buttons to play test files and light effects) but thats not made the difference either.

mad-tunes, 2025-04-30:
Maybe try the yaml from that HA forum thread, as I'm not having all the problems you mention.  I'm not sure what makes the difference tbh, but that seems to work well except for music and announcements at the same time. No repeated announcements (with the latest esphome and HA core versions) and va doesn't stop working after playing music.

I always like to understand what's making the difference if I can but hey, if it works..

kahrendt, 2025-05-22:
If you have little or no PSRAM, you should adjust the ``buffer_size`` setting to be much smaller than the default (the default assumes you have a lot of PSRAM).

If music stays ducked, make sure you undo the ducking with ``mixer_speaker.apply_ducking`` setting it back to 0 when an announcement is finished.

George1422, 2025-05-22:
@kahrendt can you be a bit more specific to what buffer_size parameter are you referring to? I'm still having the issue with my system. I think the OP has his solved.

mad-tunes, 2025-05-23:
Thanks @kahrendt
I never did get this figured out.
What might be suitable for use on a board with 2mb PSRAM?

George1422, 2025-05-26:
After 2 months of f'ing around with this (and the repeating echos of the Arduino DIY media players) I think I have something that works and is stable for at least 2 days.  Using the current version of Home Assistant 2025.5.3 and the latest version of ESPHome 2025.5.0.

Just to repeat my issue (which seems to be akin to what the OP posted). I'm using an ESP32-S3FH4R2 processor, this has 2mb of psram. I could either play music or I could use assistant but not both. I could play music (only) for days, or I could use assist for days. But if I tried to use assist after playing music the ESP32 would throw an error about unable to allocate memory. (the exact error should be above).

Based on the `buffer_size` comment above I started looking into what component had a buffer_size parameter, and that was the media player. I also found that several modules had the ability to allocate psram instead of main memory. So I'm not 100% sure which change helped or both solved the issue. 

Here is the relevant section of the code:
```
psram:
  mode: quad
  speed: 80MHz

speaker:
  - platform: i2s_audio
    id: speaker_id
    dac_type: external
    i2s_audio_id: i2s_out
    i2s_dout_pin: ${amp_din_pin}
    channel: stereo
    sample_rate: 48000
    timeout: never
    bits_per_sample: 32bit
    buffer_duration: 100ms

  - platform: mixer
    id: mixer_speaker_id
    output_speaker: speaker_id
    num_channels: "2"
    task_stack_in_psram: true        # use psram and not main memory
    source_speakers:
      - id: announcement_spk_mixer_input
        timeout: never
      - id: media_spk_mixer_input
        timeout: never

  - platform: resampler
    id: media_spk_resampling_input
    output_speaker: media_spk_mixer_input
    task_stack_in_psram: true        # use psram and not main memory
    sample_rate: 48000
    bits_per_sample: 16

  - platform: resampler
    id: announcement_spk_resampling_input
    output_speaker: announcement_spk_mixer_input
    task_stack_in_psram: true        # use psram and not main memory
    sample_rate: 48000
    bits_per_sample: 16

media_player:
  - platform: speaker
    name: "Media Player"
    internal: False
    id: media_player_id
    # task_stack_in_psram: true        # has psram option but decided to use main memory
    buffer_size: 500000                       # The default buffer size is: 1000000 I cut it in half and it didn't error out
    media_pipeline:
        speaker: media_spk_resampling_input
        format: FLAC
        num_channels: 2
        sample_rate: 48000
    announcement_pipeline:
        speaker: announcement_spk_resampling_input
        format: FLAC
        num_channels: 1
        sample_rate: 48000
```

I'm sure there can still be some optimization and something else hidden that will fail, but for now I'm just happy that both side work and I can take a break from messing with these for a while. If/When you add this code, make sure you pick a clean build so that all of the modules are compiled with the psram option. 

Will it work for you? I don't know. It did solve my issue )for now) so I thought I would document what worked for me, here.

mad-tunes, 2025-05-26:
Cheers @George1422 , that buffer size got ducked announcements working OK too.

GrumpyMeow, 2025-11-16:
Might this be wrong?
` i2s_dout_pin: $amp_din_pin`