esphome/issues #6965: ADC on ESP32-H2 does not compile since that device is lacking ADC2

baal86, 2025-04-25:
### The problem

I am trying to use the ADC sensor on the ESP32-H2

Compilation fails with many errors claiming that ADC2 related definitions are missing. This is correct, since the H2 only has one ADC. This is an issue with the esphome implementation of the ADC not handling the esp-idf API correctly for this part.

### Which version of ESPHome has the issue?

2025.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.3

### What platform are you using?

ESP32

### Board

esp32-h2-devkitm-1

### Component causing the issue

adc

### YAML Config

```yaml
esphome:
  name: water-level 

esp32:
  board: esp32-h2-devkitm-1
  framework:
    type: esp-idf
    version: latest

sensor:
  - platform: adc
    pin: GPIO01
    name: "Water Level"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
src/esphome/components/adc/adc_sensor.h:55:21: error: 'adc2_channel_t' has not been declared
   55 |   void set_channel2(adc2_channel_t channel) {
      |                     ^~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.h:96:3: error: 'adc2_channel_t' does not name a type; did you mean 'adc_channel_t'?
   96 |   adc2_channel_t channel2_{ADC2_CHANNEL_MAX};
      |   ^~~~~~~~~~~~~~
      |   adc_channel_t
src/esphome/components/adc/adc_sensor.h:55:21: error: 'adc2_channel_t' has not been declared
   55 |   void set_channel2(adc2_channel_t channel) {
      |                     ^~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.h:96:3: error: 'adc2_channel_t' does not name a type; did you mean 'adc_channel_t'?
   96 |   adc2_channel_t channel2_{ADC2_CHANNEL_MAX};
      |   ^~~~~~~~~~~~~~
      |   adc_channel_t
src/esphome/components/adc/adc_sensor.h:99:3: error: 'esp_adc_cal_characteristics_t' does not name a type
   99 |   esp_adc_cal_characteristics_t cal_characteristics_[SOC_ADC_ATTEN_NUM] = {};
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.h: In member function 'void esphome::adc::ADCSensor::set_channel1(adc1_channel_t)':
[... 127 lines cut ...]
  137 |           adc2_config_channel_atten(this->channel2_, ADC_ATTEN_DB_0);
      |                                           ^~~~~~~~~
      |                                           channel1_
src/esphome/components/adc/adc_sensor_esp32.cpp:138:30: error: 'class esphome::adc::ADCSensor' has no member named 'channel2_'; did you mean 'channel1_'?
  138 |           adc2_get_raw(this->channel2_, ADC_WIDTH_MAX_SOC_BITS, &raw0);
      |                              ^~~~~~~~~
      |                              channel1_
src/esphome/components/adc/adc_sensor_esp32.cpp:148:60: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  148 |   uint32_t mv12 = esp_adc_cal_raw_to_voltage(raw12, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_12_COMPAT]);
      |                                                            ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor_esp32.cpp:148:19: error: 'esp_adc_cal_raw_to_voltage' was not declared in this scope
  148 |   uint32_t mv12 = esp_adc_cal_raw_to_voltage(raw12, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_12_COMPAT]);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor_esp32.cpp:149:58: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  149 |   uint32_t mv6 = esp_adc_cal_raw_to_voltage(raw6, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_6]);
      |                                                          ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor_esp32.cpp:150:58: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  150 |   uint32_t mv2 = esp_adc_cal_raw_to_voltage(raw2, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_2_5]);
      |                                                          ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor_esp32.cpp:151:58: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  151 |   uint32_t mv0 = esp_adc_cal_raw_to_voltage(raw0, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_0]);
      |                                                          ^~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/water-level/src/esphome/components/adc/adc_sensor_esp32.cpp.o] Error 1
========================== [FAILED] Took 5.39 seconds ==========================
```

AndrewMarkley, 2025-04-25:
I'm getting the same compilation error with an ESP32-C6-DevKitM-1

```yaml
esphome:
  name: name

esp32:
  board: esp32-c6-devkitm-1
  variant: ESP32C6
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
    sdkconfig_options:
      CONFIG_OPENTHREAD_ENABLED: n
      CONFIG_ENABLE_WIFI_STATION: y
      CONFIG_ESPTOOLPY_FLASHSIZE_16MB: y

sensor:
  - platform: adc
    pin: GPIO01
    name: "Pressure 1"
    update_interval: 0.5s
```

baal86, 2025-04-26:
As a workaround I am using the fork by @lboue . I did not check why it works, it looks like a re-write of the ESP32 ADC subsystem to also fix the deprecated calibration.

```
external_components:
  - source:
      type: git
      url: https://github.com/lboue/esphome
      ref: adc_oneshot
    components: [ adc ]
    refresh: 0s 
```

AndrewMarkley, 2025-04-26:
yep that worked for me as well, thanks!

closed 2025-09-17 by github-actions[bot]