esphome/issues #6971: Events are triggered falsely on ESP board restart

mariusz-schimke, 2025-04-26:
### The problem

I recently noticed that when I restart my ESP board, Home Assistant treats template events as if they were triggered at that moment. This causes automations relying on those events to fire immediately after the restart. One of these automations opens my garage door. Although I generally have lower trust in homemade devices, I thought the chances of the door opening due to a bug were low — but I was wrong. :)

I restart the board via the restart button in Home Assistant. Following the restart, the MailboxEmptied event is triggered, resulting in a push notification being sent. The device is configured to send logs to my syslog server, but no relevant entries appear after the restart. However, I am unsure whether the device is fully initialized at the time the log would be generated and transmitted, so I cannot fully rely on this observation.

Previously, after restarting the board, related events in Home Assistant would become unavailable and would not trigger any automations.

I am uncertain whether this is the correct project to report this issue to. If it would be more appropriate to raise it in the Home Assistant repository, I would appreciate your guidance. Thank you!

### Which version of ESPHome has the issue?

ESPHome Device Builder 2025.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.4

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

template event

### YAML Config

```yaml
substitutions:
  mailbox_events_new_mail: NewMail
  mailbox_events_mailbox_emptied: MailboxEmptied

event:
  - platform: template
    id: mailbox_event
    name: "Mailbox"
    icon: mdi:mailbox-up
    event_types:
      - $mailbox_events_new_mail
      - $mailbox_events_mailbox_emptied
    on_event:
      then:
        - lambda: |-
            ESP_LOGI("mailbox", "Event '%s' triggered", event_type.c_str());
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

Additionally, I have noticed unusual behavior on another board following recent firmware updates. Right after updating, a GPIO output controlling a relay becomes active, causing an attached device to turn on. This behavior did not occur previously, and I am currently investigating the cause.

ssieb, 2025-04-27:
What triggers the event?  Get serial logs from boot.

mariusz-schimke, 2025-04-27:
Oh, I'm sorry, but I can't connect to it physically now. I'm trying to check what circumstances may lead to that, based on the code. Apologies, the logic is a bit lengthy, but basically, if the global variable `gate_mailbox_retrieval_door_was_open` is false, which it should after a restart (according to its restore_value setting), there is no condition that could lead to triggering the mailbox emptied event.

Unless the `notify_mailbox_emptied` script runs by itself on startup and the `mailbox_new_mail_indicator_switch` template switch it is on with its setting `restore_mode: RESTORE_DEFAULT_OFF`, but based on the UI, it's not.

So I'm wondering, and will check it later, if it is not the last event triggered just being repeated. **I have two sources of events from this board, and both of them are shown as fired in HA right after I restart the board.** If that's the case, is it a problem of HA or maybe the add-on that communicates with the board? 

```yaml
globals:
  - id: gate_mailbox_retrieval_door_was_open
    type: bool
    restore_value: false
    initial_value: "false"

binary_sensor:
  - platform: gpio
    id: gate_mailbox_slot_cover_open_indicator_input
    name: "Mailbox slot cover is open"
    device_class: opening
    icon: mdi:mailbox-open-up
    pin:
      mcp23xxx: io_port_expander
      number: $gate_mailbox_slot_cover_open_indicator_input
      inverted: false
      mode: INPUT_PULLUP
    filters:
      - delayed_on_off: $gate_relay_debounce_period
    on_press:
      then:
        - logger.log:
            format: Mailbox slot cover is open
            tag: mailbox
            level: INFO
    on_release:
      then:
        - logger.log:
            format: Mailbox slot cover is closed
            tag: mailbox
            level: INFO
        - if:
            condition:
              binary_sensor.is_off: gate_mailbox_retrieval_door_open_indicator_input
            then:
              - if:
                  condition:
                    lambda: |-
                      return id(gate_mailbox_retrieval_door_was_open);
                  then:
                    - globals.set:
                        id: gate_mailbox_retrieval_door_was_open
                        value: "false"
                    - script.execute: notify_mailbox_emptied
                  else:
                    - script.execute: notify_new_mail_in_mailbox

  - platform: gpio
    id: gate_mailbox_retrieval_door_open_indicator_input
    name: "Mailbox retrieval door is open"
    device_class: opening
    icon: mdi:mailbox-open
    pin:
      mcp23xxx: io_port_expander
      number: $gate_mailbox_retrieval_door_open_indicator_input
      inverted: false
      mode: INPUT_PULLUP
    filters:
      - delayed_on_off: $gate_relay_debounce_period
    on_press:
      then:
        - logger.log:
            format: Mailbox retrieval door is open
            tag: mailbox
            level: INFO
        - globals.set:
            id: gate_mailbox_retrieval_door_was_open
            value: "true"
    on_release:
      then:
        - if:
            condition:
              binary_sensor.is_on: gate_mailbox_slot_cover_open_indicator_input
            then:
              - logger.log:
                  format: Mailbox retrieval door is closed
                  tag: mailbox
                  level: INFO
            else:
              - logger.log:
                  format: Mailbox retrieval door open indicator switch is pressed with the door still open
                  tag: mailbox
                  level: INFO
```

```yaml
  - id: notify_mailbox_emptied
    mode: single
    then:
      - if:
          condition:
            switch.is_on: mailbox_new_mail_indicator_switch
          then:
            - switch.turn_off: mailbox_new_mail_indicator_switch
            - event.trigger:
                id: mailbox_event
                event_type: $mailbox_events_mailbox_emptied
            - rtttl.play:
                id: buzzer
                rtttl: $rtttl_mailbox_emptied
      - script.execute: mailbox_recently_filled_or_emptied

  - id: mailbox_recently_filled_or_emptied
    mode: restart
    then:
      - delay: $gate_mailbox_prevent_duplicate_events_period
```

mariusz-schimke, 2025-04-27:
@ssieb, let's forget the previous comment. I can confirm that HA just repeats the event that was the last event before the device was restarted.

Here's the sequence of actions (let's concentrate on the `Mailbox` event source).

1. This what the UI shows initially (MailboxEmptied event triggered last week in the first picture, but the history shows two events 8 minutes before):

[image]
[image]

2. Now I open the mailbox'` slot cover – the `NewMail` event is triggered:

[image]
[image]

3. I'm waiting a few seconds and press the restart button in the UI. The time on the first image still shows the right time the last event was **in fact** last triggered, but when I open history, it indicates new events that appear right before and right after the restart:

[image]
[image]

Since this event is attached to an automation, I'm getting en email about new mail after I restart the board.

mariusz-schimke, 2025-04-27:
But long story short, every restart generated two events, so I suppose the first one happens before the restart (one without a dot), and the second one after the restart.

[image]

mariusz-schimke, 2025-04-27:
There is one more thing that I just realized. About a month ago I started storing data in the recorder DB. Previously I had all entities filtered, so nothing was stored there, but I removed the filter recently. This means that the previous states of entities are probably restored now from the DB. But is there any reason a historical event should be re-fired when restored from the DB? An event is something that happened in the past and shouldn't repeat just like that.

mariusz-schimke, 2025-04-30:
@ssieb, do you think it has anything to do with the ESPHome add-on or is it something to be reported in the HA repository?

mariusz-schimke, 2025-05-04:
I investigated the problem further using a very simple setup. I think we can now ignore all my previous messages as this is easily reproducible based on the configuration below:

ESPHome Template event:

```yaml
event:
  - platform: template
    id: test_event
    name: "Test event"
    event_types:
      - test_event_1
      - test_event_2
    on_event:
      then:
        - lambda: |-
            ESP_LOGI("test-event", "Event '%s' triggered", event_type.c_str());
```

ESPHome API:

```yaml
api:
  services:
    - service: trigger_test_event_1
      then:
        - event.trigger:
            id: test_event
            event_type: test_event_1
```

Home Assistant automation:

```yaml
triggers:
  - entity_id:
      - event.my_test_device_test_event
    trigger: state
    attribute: event_type
    to: test_event_1
```

As I can see now from the automation, there's no way to identify this as an event as such. It's just an attribute of an entity named `event.name`, so I assume that when the state gets reloaded from the DB, the HA automation sees it as a state change to `test_event_1`, and that triggers the automation. I don't know a way with this set of automation configuration options to prevent this. Since I'm not looking for event changes but for events as such, I can't use the `from` option.

Am I trying to achieve something it wasn't designed for?

closed 2025-05-04 by mariusz-schimke