esphome/issues #6975: api.connected remains true when device is deleted or disabled from home assistant

MaJerle, 2025-04-27:
### The problem

When you disable or delete device from Home Assistant in the `ESPHome integration`, `api.connected` condition remains `true`.

The device prints the following log:

```
[11:45:41][D][api.connection:233]: Home Assistant 2025.4.4 (192.168.0.30) requested disconnected
```

But all the further conditions that are linked to `not api.connected` do not work. They only start to work again if I restart ESPHome device.

When device is not in HA yet, is powered and detected by the *ESPHome* integration inside HA, esphome log prints:

```
[11:47:42][D][api:103]: Accepted 192.168.0.30
[11:47:42][D][api.connection:1801]: Home Assistant 2025.4.4 (192.168.0.30): Connected successfully
```

Which seems correct.

### Which version of ESPHome has the issue?

2025.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.4

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

api

### YAML Config

```yaml
# Inputs
binary_sensor:
  - platform: gpio
    name: "Input #${id}"
    id: input_${id}
    pin:
      pcf8574: pcf8574_hub_input
      number: ${pcf_input_channel_number}
      mode:
        input: true
    filters:
      - invert:
      - delayed_on_off: 50ms

    # On press
    on_press:
      then:
        - logger.log: "on_press"
        - if:
            condition:
              - not:
                  api.connected:
            then:
              - switch.toggle: switch_${id}
```

### Anything in the logs that might be useful for us?

```txt

```

MaJerle, 2025-04-27:
Further tests, for this basic configuration:

```yaml
# Inputs
binary_sensor:
  - platform: gpio
    ..other setings....

    # On press
    on_press:
      then:
        - if:
            condition:
              - api.connected:
            then:
              - logger.log: "API CONNECTED"
            else:
              - logger.log: "API NOT CONNECTED"
        - if:
            condition:
              - not:
                  api.connected:
            then:
              - logger.log: "Not check: API NOT CONNECTED"
            else:
              - logger.log: "Not check: API CONNECTED"
```

At start, when HA has device in the *ESPHome* integration and I activate the binary sensor:

```
[13:49:55][D][main:2888]: API CONNECTED
[13:49:55][D][main:2904]: Not check: API CONNECTED
```

When device is *disabled* in the *ESPHome*, or when it is deleted from the integration:

```
[13:50:24][D][api.connection:233]: Home Assistant 2025.4.4 (192.168.0.30) requested disconnected
```

And when I press the button again:

```
[13:50:28][D][main:2888]: API CONNECTED
[13:50:28][D][main:2904]: Not check: API CONNECTED
```

[image]

ssieb, 2025-04-27:
The wireless logs use the api, so if you're watching the logs like that, then the api is connected.

MaJerle, 2025-04-27:
> The wireless logs use the api, so if you're watching the logs like that, then the api is connected.

I use the link with local esphome for logs, outside HA. But I tried to remove this log link and it works.

closed 2025-04-27 by MaJerle