esphome/issues #6978: [Docs] Information about yaml is impossible to find

spacemanspiff2007, 2025-04-29:
### The problem

Information about the yaml features and how to create/use them is impossible to find.
[Substitions](https://esphome.io/components/substitutions.html#substitute-include-variables) is under "supported microcontrollers"

[image]

But if I click it then a link to the page does not appear.
But substitutions also describes the `!include` operation which is not described anywhere else.

Imho there should be a dedicated `yaml` section which describes these things.

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

jzucker2, 2025-04-29:
Like this?

https://esphome.io/guides/faq#tips-for-using-esphome

spacemanspiff2007, 2025-04-30:
Yes, there's another bit of information which I missed. 
It's not clear at all that the entry "tips for using esphome" is describing the yaml structure and features.
Also I think this should not be in faq but rather in an own section "working with yaml files" or something like that.