esphome/issues #6984: Device Class: wind_direction

JulianDHall, 2025-05-01:
### The problem

Device classes for ESPHome refers to the HA documentation which supports “wind_direction”.  ESPHome does not recognise “wind_direction” and show error on compiling.  Device classes are not aligned with HA.

### Which version of ESPHome has the issue?

2025.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.4

### What platform are you using?

ESP32

### Board

QUINLED

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

DerTomm, 2025-05-11:
I'm facing the same issue. This is the relevant part of my config:

```
sensor:
    name: "FineOffset WH1080 Wind Direction"
    id: wh1080_wind_direction
    device_class: wind_direction
```

The esphome verificaton then tells me:

```
 Unknown value 'wind_direction', valid options are 'apparent_power', 'aqi', 'atmospheric_pressure', 'battery', 'carbon_dioxide', 'carbon_monoxide', 'conductivity', 'current', 'data_rate', 'data_size', 'date', 'distance', 'duration', '', 'energy', 'energy_storage', 'frequency', 'gas', 'humidity', 'illuminance', 'irradiance', 'moisture', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'ozone', 'ph', 'pm1', 'pm10', 'pm25', 'power', 'power_factor', 'precipitation', 'precipitation_intensity', 'pressure', 'reactive_power', 'signal_strength', 'sound_pressure', 'speed', 'sulphur_dioxide', 'temperature', 'timestamp', 'volatile_organic_compounds', 'volatile_organic_compounds_parts', 'voltage', 'volume', 'volume_flow_rate', 'volume_storage', 'water', 'weight', 'wind_speed'.
```

closed 2025-09-24 by github-actions[bot]

LorbusChris, 2025-09-24:
Can this be re-opened?