esphome/issues #6990: as3935_i2c binary_sensor.h: No such file or directory

manuel-rw, 2025-05-03:
### The problem

When compiling using the [as3935 sensor](https://esphome.io/components/sensor/as3935.html), it fails with the below error message:
```
In file included from src\esphome\components\as3935_i2c\as3935_i2c.cpp:1:
src\esphome\components\as3935_i2c\as3935_i2c.h:7:10: fatal error: esphome/components/binary_sensor/binary_sensor.h: No such file or directory
    7 | #include "esphome/components/binary_sensor/binary_sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\sen54-test\src\esphome\components\as3935_i2c\as3935_i2c.cpp.o] Error 1
```

### Which version of ESPHome has the issue?

2025.4.1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

d1_mini_lite

### Component causing the issue

as3935

### YAML Config

```yaml
esphome:
  name: sen54-test

esp8266:
  board: d1_mini_lite
  framework:
    version: recommended

logger:

api:
  password: ""

ota:
  - platform: esphome
    password: ""

web_server:
  port: 80

sensor:
  - platform: as3935
    lightning_energy:
      name: Lightning Energy
    distance:
      name: Distance Storm

as3935_i2c:
  irq_pin: GPIO16

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: true
  id: bus_a
  frequency: 200kHz

wifi:
  ssid: "REDACTED"
  password: "REDACTED"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
In file included from src\esphome\components\as3935_i2c\as3935_i2c.cpp:1:
src\esphome\components\as3935_i2c\as3935_i2c.h:7:10: fatal error: esphome/components/binary_sensor/binary_sensor.h: No such file or directory
    7 | #include "esphome/components/binary_sensor/binary_sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\sen54-test\src\esphome\components\as3935_i2c\as3935_i2c.cpp.o] Error 1
=============================================================================================== [FAILED] Took 11.45 seconds ===============================================================================================
```

### Additional information

  **System:**
    OS: Windows 11 10.0.26100
    CPU: (4) x64 Intel(R) N100
    Memory: 4.98 GB / 11.78 GB
  **Binaries:**
    Node: 22.15.0 - C:\nvm4w\nodejs\node.EXE
    npm: 10.9.2 - C:\nvm4w\nodejs\npm.CMD
  **Managers:**
    pip3: 25.1.1 - ~\AppData\Local\Programs\Python\Python313\Scripts\pip3.EXE
  **Utilities:**
    Git: 2.49.0.
    Curl: 8.12.1 - C:\WINDOWS\system32\curl.EXE
  **IDEs:**
    VSCode: 1.99.3 - C:\Users\XXXXX\AppData\Local\Programs\Microsoft VS Code\bin\code.CMD
  **Languages:**
    Python: 3.13.3
  **Browsers:**
    Edge: Chromium (133.0.3065.92)

Command to reproduce:
```bash
 esphome compile .\test.yaml      
```

clydebarrow, 2025-05-03:
As a workaround just add a binary_sensor - that will be sufficient to bring in the file in.

```
binary_sensor:
  - platform: template
    id: b_sens
```

manuel-rw, 2025-05-03:
Thanks, I can confirm that adding the binary sensor worked.
Would you mind to explain why the ID must be ``b_sens`` and why it fixes the import?

Sadly, the sensor is still not working:
```log
[16:14:52][I][app:029]: Running through setup()...
[16:14:52][I][i2c.arduino:218]: Performing I2C bus recovery
[16:14:52][C][as3935:010]: Setting up AS3935...
[16:14:52][C][as3935:013]:   IRQ Pin: GPIO16
[16:14:52][E][component:119]: Component as3935 was marked as failed.
[16:14:52][E][component:164]: Component as3935 set Error flag: unspecified
[16:14:52][W][as3935_i2c:013]: read_byte failed - increase log level for more details!
[16:14:52][E][component:119]: Component as3935 was marked as failed.
[16:14:52][W][as3935_i2c:013]: read_byte failed - increase log level for more details!
[16:14:52][E][component:119]: Component as3935 was marked as failed.
[16:14:52][W][as3935_i2c:013]: read_byte failed - increase log level for more details!
[16:14:52][E][component:119]: Component as3935 was marked as failed.
[16:14:52][W][as3935_i2c:013]: read_byte failed - increase log level for more details!
[16:14:52][E][component:119]: Component as3935 was marked as failed.
[16:14:52][W][as3935_i2c:013]: read_byte failed - increase log level for more details!
[16:14:52][E][component:119]: Component as3935 was marked as failed.
[16:14:52][W][as3935_i2c:013]: read_byte failed - increase log level for more details!
[16:14:52][E][component:119]: Component as3935 was marked as failed.
[16:14:52][W][as3935_i2c:013]: read_byte failed - increase log level for more details!
[16:14:52][I][as3935:285]: Starting oscillators calibration...
[16:14:53][E][component:119]: Component as3935 was marked as failed.
[16:14:53][W][as3935_i2c:013]: read_byte failed - increase log level for more details!
[16:14:53][E][component:119]: Component as3935 was marked as failed.
[16:14:53][W][as3935_i2c:013]: read_byte failed - increase log level for more details!
[16:14:53][E][component:119]: Component as3935 was marked as failed.
[16:14:53][W][as3935_i2c:013]: read_byte failed - increase log level for more details!
[16:14:53][W][as3935_i2c:029]: Writing register failed!
[16:14:53][W][as3935_i2c:029]: Writing register failed!
```

I observed that the I2C signal looks to be intact (acquired from the AS3935 pins):

[image]

I tried the following modules, none work:
- CJMCU AS3935 (purple module from AliExpress)
- DFRobot Gravity AS3935
- Custom PCB of my own using the AS3935, directly copied from the reference schematic (see page 9 in the [manual](https://www.mouser.com/datasheet/2/588/AS3935_Datasheet_v4-1214699.pdf)),

Would you like me to open a new issue or comment on #1411 , #5580 or #5581? At the moment, I cannot use SPI due to design limitations.

clydebarrow, 2025-05-03:
> Would you mind to explain why the ID must be b_sens and why it fixes the import?

The id doesn't have to be anything in particular, it's merely the presence of *any* binary_sensor in the config that ensures the C++ files are included in the build.

As to why it's not working, that's a support question, not appropriate for an issue here - your best bet would be to ask in the Discord general support channel (Discord link is in the docs.)

It's likely to be something to do with i2c addressing, clock speed or pullup resistors. I can't be sure, but from the pic it does look like you have the address DIP switches in the wrong position.

closed 2025-05-05 by jesserockz