esphome/issues #6991: cstdint not included for host build on Fedora 42

EnAppelsin, 2025-05-03:
### The problem

When trying to compile a trivial SDL project for testing LVGL layouts, compilation fails due to numerous issues with <cstdint> not being included in various files

### Which version of ESPHome has the issue?

2025.4.1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

host

### Component causing the issue

core

### YAML Config

```yaml
# Example configuration entry
esphome:
  name: sdl

host:

display:
  - platform: sdl
    show_test_card: true
    dimensions:
      width: 320
      height: 172
```

### Anything in the logs that might be useful for us?

```txt
Processing sdl (build_flags: -DSDL_BYTEORDER=4321, -DUSE_HOST, -D_GNU_SOURCE=1, -D_REENTRANT, -I/usr/include/SDL2, -Wno-sign-compare, -Wno-unused-but-set-variable, -Wno-unused-variable, -fno-exceptions, -lSDL2, -std=c++17; lib_deps: ; platform: platformio/native)
------------------------------------------------------------------------------------------------------------------------------------------------------------
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
g++ -o .pioenvs/sdl/src/esphome/components/display/display.o -c -std=c++17 -Wno-sign-compare -Wno-unused-but-set-variable -Wno-unused-variable -fno-exceptions -DPLATFORMIO=60118 -DSDL_BYTEORDER=4321 -DUSE_HOST -D_GNU_SOURCE=1 -D_REENTRANT -Isrc -I/usr/include/SDL2 src/esphome/components/display/display.cpp
g++ -o .pioenvs/sdl/src/esphome/components/display/display_buffer.o -c -std=c++17 -Wno-sign-compare -Wno-unused-but-set-variable -Wno-unused-variable -fno-exceptions -DPLATFORMIO=60118 -DSDL_BYTEORDER=4321 -DUSE_HOST -D_GNU_SOURCE=1 -D_REENTRANT -Isrc -I/usr/include/SDL2 src/esphome/components/display/display_buffer.cpp
g++ -o .pioenvs/sdl/src/esphome/components/display/rect.o -c -std=c++17 -Wno-sign-compare -Wno-unused-but-set-variable -Wno-unused-variable -fno-exceptions -DPLATFORMIO=60118 -DSDL_BYTEORDER=4321 -DUSE_HOST -D_GNU_SOURCE=1 -D_REENTRANT -Isrc -I/usr/include/SDL2 src/esphome/components/display/rect.cpp
g++ -o .pioenvs/sdl/src/esphome/core/helpers.o -c -std=c++17 -Wno-sign-compare -Wno-unused-but-set-variable -Wno-unused-variable -fno-exceptions -DPLATFORMIO=60118 -DSDL_BYTEORDER=4321 -DUSE_HOST -D_GNU_SOURCE=1 -D_REENTRANT -Isrc -I/usr/include/SDL2 src/esphome/core/helpers.cpp
g++ -o .pioenvs/sdl/src/esphome/core/ring_buffer.o -c -std=c++17 -Wno-sign-compare -Wno-unused-but-set-variable -Wno-unused-variable -fno-exceptions -DPLATFORMIO=60118 -DSDL_BYTEORDER=4321 -DUSE_HOST -D_GNU_SOURCE=1 -D_REENTRANT -Isrc -I/usr/include/SDL2 src/esphome/core/ring_buffer.cpp
g++ -o .pioenvs/sdl/src/esphome/core/string_ref.o -c -std=c++17 -Wno-sign-compare -Wno-unused-but-set-variable -Wno-unused-variable -fno-exceptions -DPLATFORMIO=60118 -DSDL_BYTEORDER=4321 -DUSE_HOST -D_GNU_SOURCE=1 -D_REENTRANT -Isrc -I/usr/include/SDL2 src/esphome/core/string_ref.cpp
g++ -o .pioenvs/sdl/src/main.o -c -std=c++17 -Wno-sign-compare -Wno-unused-but-set-variable -Wno-unused-variable -fno-exceptions -DPLATFORMIO=60118 -DSDL_BYTEORDER=4321 -DUSE_HOST -D_GNU_SOURCE=1 -D_REENTRANT -Isrc -I/usr/include/SDL2 src/main.cpp
In file included from src/esphome/core/string_ref.cpp:1:
src/esphome/core/string_ref.h:74:9: error: ‘uint8_t’ does not name a type
   74 |   const uint8_t *byte() const { return reinterpret_cast<const uint8_t *>(base_); }
      |         ^~~~~~~
src/esphome/core/string_ref.h:8:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
    7 | #include "esphome/core/defines.h"
  +++ |+#include <cstdint>
    8 |
*** [.pioenvs/sdl/src/esphome/core/string_ref.o] Error 1
In file included from src/esphome/components/display/rect.h:3,
                 from src/esphome/components/display/display.h:6,
                 from src/esphome/components/display/display.cpp:1:
src/esphome/core/helpers.h: In function ‘constexpr T esphome::byteswap(T)’:
src/esphome/core/helpers.h:141:22: error: ‘uint8_t’ does not name a type [-Wtemplate-body]
  141 |     reinterpret_cast<uint8_t *>(&m)[i] = reinterpret_cast<uint8_t *>(&n)[sizeof(T) - 1 - i];
      |                      ^~~~~~~
src/esphome/core/helpers.h:13:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
   12 | #include "esphome/core/optional.h"
  +++ |+#include <cstdint>
   13 |
```

### Additional information

[Output of `esphome -l DEBUG compile sdl.yml`](https://github.com/user-attachments/files/20023357/compile.log)

tzahari, 2025-05-08:
Similar problem on **Mac OS X 15.4.1**
I did my setup described on https://esphome.io/components/display/sdl.html

I installed with `brew install sdl2 libsodium` and linked it successfully.

And tried ESPHome in a venv or via brew. Both failed.

my esphome run output
```
INFO ESPHome 2025.4.1
INFO Reading configuration test-sdl.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing sdl (platform: platformio/native)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
No dependencies
Compiling .pioenvs/sdl/src/esphome/components/display/display.o
Compiling .pioenvs/sdl/src/esphome/components/display/display_buffer.o
Compiling .pioenvs/sdl/src/esphome/components/display/rect.o
Compiling .pioenvs/sdl/src/esphome/components/host/core.o
Compiling .pioenvs/sdl/src/esphome/components/host/gpio.o
Compiling .pioenvs/sdl/src/esphome/components/host/preferences.o
Compiling .pioenvs/sdl/src/esphome/components/mdns/mdns_component.o
Compiling .pioenvs/sdl/src/esphome/components/mdns/mdns_esp8266.o
In file included from In file included from src/esphome/components/display/rect.cppsrc/esphome/components/display/display_buffer.cpp::11In file included from :
src/esphome/components/host/core.cppIn file included from :3src/esphome/components/host/gpio.cpp:
:src/esphome/core/hal.h3::
:
2src/esphome/components/host/preferences.cpp::310::10 :fatal error:  'string' file not foundfatal error: 
In file included from src/esphome/components/display/rect.h:3src/esphome/components/display/display_buffer.hIn file included from :
In file included from src/esphome/core/helpers.hsrc/esphome/components/display/display.cpp::31::
src/esphome/components/display/display.h'filesystem' file not found:103:
10: fatal error: 'cstdarg' file not found
src/esphome/components/host/gpio.h    ::    3 | #include     3< | c5s:
t:3:2 src/esphome/core/hal.h:2:10: fatal error: d'string' file not found
a10rg:> #    2 | #inclfatal error: u'cstdarg' file not foundd
e
       <| s         ^~~~~~~~~t
r    i3n | g#>fatal error: 'cmath' file not foundi | n#cilnucdleu d<ec s<tsdt

iarngc>l
u      d| e         ^~~~~~~~~ 
<filesystem>
      |          ^~~~~~~~~~~~
    3 | #include <cmath>
r      i|          ^~~~~~~
      |          ^~~~~~~~
ng>
      |          ^~~~~~~~
1 error generated.
1 error generated.
In file included from src/esphome/components/mdns/mdns_component.cpp:3:
src/esphome/components/mdns/mdns_component.h:4:10: fatal error: 'string' file not found
    4 | #include <string>
      |          ^~~~~~~~
*** [.pioenvs/sdl/src/esphome/components/host/gpio.o] Error 1
*** [.pioenvs/sdl/src/esphome/components/display/rect.o] Error 1
1 error generated.
*** [.pioenvs/sdl/src/esphome/components/host/preferences.o] Error 1
1 error generated.
1 error generated.
*** [.pioenvs/sdl/src/esphome/components/mdns/mdns_component.o] Error 1
*** [.pioenvs/sdl/src/esphome/components/display/display_buffer.o] Error 1
1 error generated.
1 error generated.
*** [.pioenvs/sdl/src/esphome/components/display/display.o] Error 1
*** [.pioenvs/sdl/src/esphome/components/host/core.o] Error 1
===================================================================== [FAILED] Took 0.65 seconds =====================================================================
```

ZDBioHazard, 2025-05-21:
I'm on Gentoo with ESPHome 2025.5.0 in a venv and GCC 15.1.

I was setting up a similar minimal LVGL test build as OP, and this was the minimum I needed to modify to get it to work.

```diff
--- a/esphome/core/helpers.h
+++ b/esphome/core/helpers.h
@@ -11,6 +11,10 @@
 
 #include "esphome/core/optional.h"
 
+#ifdef USE_HOST
+#include <cstdint>
+#endif
+
 #ifdef USE_ESP8266
 #include <Esp.h>
 #endif
--- a/esphome/core/string_ref.h
+++ b/esphome/core/string_ref.h
@@ -6,6 +6,10 @@
 #include <string>
 #include "esphome/core/defines.h"
 
+#ifdef USE_HOST
+#include <cstdint>
+#endif
+
 #ifdef USE_JSON
 #include "esphome/components/json/json_util.h"
 #endif  // USE_JSON
```

Other similar fixes are probably needed as other components are pulled in, but this is enough for the LVGL demo.

ZDBioHazard, 2025-05-23:
Looks like this is a GCC 15 thing: https://gcc.gnu.org/gcc-15/porting_to.html#header-dep-changes

There are probably other missing includes floating around the codebase, but I'm not familiar enough with the build system to do a comprehensive build to test. Everything in my limited test configs seem to work with just the two edits.

tzahari, 2025-05-23:
@ZDBioHazard with your patch it getting better but I have still more compiler errors:

Any hints what to do?

```
INFO ESPHome 2025.5.0
INFO Reading configuration test-sdl.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing sdl (platform: platformio/native)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
No dependencies
Compiling .pioenvs/sdl/src/esphome/components/display/display.o
Compiling .pioenvs/sdl/src/esphome/components/display/display_buffer.o
Compiling .pioenvs/sdl/src/esphome/components/display/rect.o
Compiling .pioenvs/sdl/src/esphome/components/host/core.o
Compiling .pioenvs/sdl/src/esphome/components/host/gpio.o
Compiling .pioenvs/sdl/src/esphome/components/host/preferences.o
Compiling .pioenvs/sdl/src/esphome/components/mdns/mdns_component.o
Compiling .pioenvs/sdl/src/esphome/components/mdns/mdns_esp32.o
In file included from src/esphome/components/display/display_buffer.cpp:1:
src/esphome/components/display/display_buffer.h:3:10: fatal error: 'cstdarg' file not found
    3 | #include <cstdarg>
      |          ^~~~~~~~~
In file included from src/esphome/components/display/display.cpp:1:
src/esphome/components/display/display.h:3:10: fatal error: 'cstdarg' file not found
    3 | #include <cstdarg>
      |          ^~~~~~~~~
In file included from src/esphome/components/display/rect.cpp:1:
In file included from src/esphome/components/display/rect.h:3:
src/esphome/core/helpers.h:3:10: fatal error: 'cmath' file not found
    3 | #include <cmath>
      |          ^~~~~~~
1 error generated.
In file included from src/esphome/components/host/core.cpp:3:
src/esphome/core/hal.h:2:10: fatal error: 'string' file not found
    2 | #include <string>
      |          ^~~~~~~~
1 error generated.
1 error generated.
In file included from src/esphome/components/host/gpio.cpp:3:
In file included from src/esphome/components/host/gpio.h:5:
src/esphome/core/hal.h:2:10: fatal error: 'string' file not found
    2 | #include <string>
      |          ^~~~~~~~
*** [.pioenvs/sdl/src/esphome/components/display/display_buffer.o] Error 1
*** [.pioenvs/sdl/src/esphome/components/display/rect.o] Error 1
*** [.pioenvs/sdl/src/esphome/components/display/display.o] Error 1
1 error generated.
*** [.pioenvs/sdl/src/esphome/components/host/gpio.o] Error 1
src/esphome/components/host/preferences.cpp:3:10: fatal error: 'filesystem' file not found
    3 | #include <filesystem>
      |          ^~~~~~~~~~~~
1 error generated.
1 error generated.
*** [.pioenvs/sdl/src/esphome/components/host/core.o] Error 1
*** [.pioenvs/sdl/src/esphome/components/host/preferences.o] Error 1
In file included from src/esphome/components/mdns/mdns_component.cpp:3:
src/esphome/core/application.h:3:10: fatal error: 'string' file not found
    3 | #include <string>
      |          ^~~~~~~~
1 error generated.
*** [.pioenvs/sdl/src/esphome/components/mdns/mdns_component.o] Error 1
===================================================================== [FAILED] Took 0.47 seconds =====================================================================
```

ZDBioHazard, 2025-05-23:
@tzahari I think you are having a different issue.

Your build seems to be having trouble finding requested header files, so you are probably missing a dev library package, or your compiler is looking for headers in the wrong places.

The mentioned headers are ones typically provided by the platform libc, which I believe should have been installed alongside your compiler. You'll need to somehow make sure your compiler/toolchain is installed and configured correctly.

Running the build with `esphome -l DEBUG compile ...` will show the compiler commands and list some of the header directories used, so that's a good place to start looking.

I'm not familiar with the MacOS/homebrew toolchain, hopefully you can figure out a solution.

closed 2025-10-06 by github-actions[bot]