esphome/issues #7005: Boot loop caused by i2c scanning

MaxMyzer, 2025-05-10:
### The problem

When `scan: True` is in the i2c config, the device repeatably boot loops until it gets kicked into safe mode. During this time, there is no serial output at all. I have been running into this while trying to communicate with the onboard BME280. Both addresses fail to communicate with it, so I was attempting to scan for it. 

I believe this may have to do with the i2c power pin pullup, but I am not sure.

Before I get lectured, I was planning to move all my credentials to `secrets` later on, for now please don't mind `removed`

### Which version of ESPHome has the issue?

2025.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.1

### What platform are you using?

ESP32

### Board

Adafruit Industries ESP32-S2 Feather with BME280 Sensor

### Component causing the issue

i2c

### YAML Config

```yaml
esphome:
  name: mxmy-climate
  friendly_name: Max Feather
  comment: "BME280 test run"
  platformio_options:
    upload_speed: 115200
  on_boot: 
    - priority: 90
      then:
        - output.turn_on: pull_high_a
        - output.turn_on: pull_high_b

  project:
    name: "mxmy.climate"
    version: "1.0.0"

esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: [removed]
  port: 6053

web_server: 
  port: 8080

# time:
#   - platform: homeassistant
#     id: homeassistant_time

ota:
  - platform: esphome
    password: [removed]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "climate-fallback"
    password: [removed]

captive_portal:

# Example configuration entry for ESP32
i2c:
  sda: GPIO3
  scl: GPIO4
  scan: False
  id: stemma_qt

sensor:
  - platform: bme280_i2c
    temperature:
      name: "BME280 Temperature"
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x77

  - platform: max17043
    battery_voltage:
      name: "Battery Voltage"
    battery_level:
      name: "Battery"

light:
   # built in neopixel
  - platform: neopixelbus
    type: GRB
    variant: WS2811
    pin: GPIO33
    num_leds: 1
    name: "NeoPixel_Onboard"
   # built in LED
  - platform: status_led
    name: "Switch state"
    pin: GPIO13

output:
  # In Arduino & CircuitPython, these are pulled high by default. 
  # Need to manually pull high here. 
  # More info: https://learn.adafruit.com/adafruit-esp32-s2-feather/pinouts
   # i2c power pin pullup
  - platform: gpio
    id: pull_high_a
    pin: GPIO7
   # Neopixel power
  - platform: gpio
    id: pull_high_b
    pin: GPIO21
```

### Anything in the logs that might be useful for us?

```txt
The serial connection immediately closes when it is not in safe mode. However, here is the log when scanning is disabled, in case it offers clues

[W][component:157]: Component max17043.sensor set Warning flag: unspecified
[C][wifi:048]: Setting up WiFi...
[C][wifi:061]: Starting WiFi...
[C][wifi:062]:   Local MAC: 84:F7:03:EF:4A:48
[D][wifi:482]: Starting scan...
[W][component:157]: Component wifi set Warning flag: scanning for networks
[W][component:239]: Component max17043.sensor took a long time for an operation                          (1999 ms).
[W][component:240]: Components should block for at most 30 ms.
[D][wifi:497]: Found networks:
[I][wifi:541]: - '...' (70:58:...) ▂▄▆█
[D][wifi:542]:     Channel: 1
[D][wifi:543]:     RSSI: -55 dB
[I][wifi:541]: - '...' (70:58:...) ▂▄▆█
[D][wifi:542]:     Channel: 11
[D][wifi:543]:     RSSI: -69 dB
[I][wifi:313]: WiFi Connecting to '...'...
[D][esp-idf:000][wifi]: E (11151) wifi:
[D][esp-idf:000][wifi]: Association refused temporarily, comeback time 1024 mSec
[D][esp-idf:000][wifi]:

[W][wifi_esp32:569][arduino_events]: Event: Disconnected ssid='...' bssid=70:58:... reason='Association Failed'
[W][wifi:653]: Error while connecting to network.
[D][wifi:697]: Retrying with hidden networks...
[I][wifi:313]: WiFi Connecting to '...'...
[I][wifi:617]: WiFi Connected!
[C][wifi:428]:   Local MAC: 84:F7:03:EF:4A:48
[C][wifi:433]:   SSID: '...'
[C][wifi:436]:   IP Address: 192.168.1.44
[C][wifi:440]:   BSSID: 70:58:...
[C][wifi:441]:   Hostname: 'mxmy-climate'
[C][wifi:443]:   Signal strength: -47 dB ▂▄▆█
[C][wifi:447]:   Channel: 1
[C][wifi:448]:   Subnet: 255.255.255.0
[C][wifi:449]:   Gateway: 192.168.1.1
[C][wifi:450]:   DNS1: 192.168.1.2
[C][wifi:451]:   DNS2: 9.9.9.9
[D][wifi:626]: Disabling AP...
[C][web_server:238]: Setting up web server...
[C][api:026]: Setting up Home Assistant API server...
[I][app:062]: setup() finished successfully!
[W][component:172]: Component wifi cleared Warning flag
[W][component:157]: Component api set Warning flag: unspecified
[I][app:100]: ESPHome version 2025.4.1 compiled on May  9 2025, 18:51:52
[I][app:102]: Project mxmy.climate version 1.0.0
[C][status_led:065]: Status Led Light:
[C][status_led:066]:   Pin: GPIO13
[C][wifi:600]: WiFi:
[C][wifi:428]:   Local MAC: 84:F7:03:EF:4A:48
[C][wifi:433]:   SSID: '...'
[C][wifi:436]:   IP Address: 192.168.1.44
[C][wifi:440]:   BSSID: 70:58:...
[C][wifi:441]:   Hostname: 'mxmy-climate'
[C][wifi:443]:   Signal strength: -47 dB ▂▄▆█
[C][wifi:447]:   Channel: 1
[C][wifi:448]:   Subnet: 255.255.255.0
[C][wifi:449]:   Gateway: 192.168.1.1
[C][wifi:450]:   DNS1: 192.168.1.2
[C][wifi:451]:   DNS2: 9.9.9.9
[C][logger:177]: Logger:
[C][logger:178]:   Max Level: DEBUG
[C][logger:179]:   Initial Level: DEBUG
[C][logger:181]:   Log Baud Rate: 115200
[C][logger:182]:   Hardware UART: USB_CDC
[C][i2c.arduino:071]: I2C Bus:
[C][i2c.arduino:072]:   SDA Pin: GPIO3
[C][i2c.arduino:073]:   SCL Pin: GPIO4
[C][i2c.arduino:074]:   Frequency: 50000 Hz
[C][i2c.arduino:089]:   Recovery: failed, SCL is held low on the bus
[C][gpio.output:010]: GPIO Binary Output:
[C][gpio.output:011]:   Pin: GPIO7
[C][gpio.output:010]: GPIO Binary Output:
[C][gpio.output:011]:   Pin: GPIO21
[C][light:092]: Light 'NeoPixel_Onboard'
[C][light:094]:   Default Transition Length: 1.0s
[C][light:095]:   Gamma Correct: 2.80
[C][light:092]: Light 'Switch state'
[C][bme280_i2c.sensor:025]:   Address: 0x76
[C][bme280.sensor:182]: BME280:
[E][bme280.sensor:185]: Communication with BME280 failed!
[C][bme280.sensor:194]:   IIR Filter: OFF
[C][bme280.sensor:195]:   Update Interval: 60.0s
[C][bme280.sensor:197]:   Temperature 'BME280 Temperature'
[C][bme280.sensor:197]:     Device Class: 'temperature'
[C][bme280.sensor:197]:     State Class: 'measurement'
[C][bme280.sensor:197]:     Unit of Measurement: '°C'
[C][bme280.sensor:197]:     Accuracy Decimals: 1
[C][bme280.sensor:198]:     Oversampling: 16x
[C][bme280.sensor:199]:   Pressure 'BME280 Pressure'
[C][bme280.sensor:199]:     Device Class: 'pressure'
[C][bme280.sensor:199]:     State Class: 'measurement'
[C][bme280.sensor:199]:     Unit of Measurement: 'hPa'
[C][bme280.sensor:199]:     Accuracy Decimals: 1
[C][bme280.sensor:200]:     Oversampling: 16x
[C][bme280.sensor:201]:   Humidity 'BME280 Humidity'
[C][bme280.sensor:201]:     Device Class: 'humidity'
[C][bme280.sensor:201]:     State Class: 'measurement'
[C][bme280.sensor:201]:     Unit of Measurement: '%'
[C][bme280.sensor:201]:     Accuracy Decimals: 1
[C][bme280.sensor:202]:     Oversampling: 16x
[E][component:082]:   Component bme280_base is marked FAILED:
[C][max17043:076]: MAX17043:
[C][max17043:077]:   Address: 0x36
[C][max17043:081]:   Update Interval: 60.0s
[C][max17043:082]:   Battery Voltage 'Battery Voltage'
[C][max17043:082]:     Device Class: 'voltage'
[C][max17043:082]:     State Class: 'measurement'
[C][max17043:082]:     Unit of Measurement: 'V'
[C][max17043:082]:     Accuracy Decimals: 3
[C][max17043:083]:   Battery Level 'Battery'
[C][max17043:083]:     Device Class: 'battery'
[C][max17043:083]:     State Class: 'measurement'
[C][max17043:083]:     Unit of Measurement: '%'
[C][max17043:083]:     Accuracy Decimals: 3
[C][captive_portal:089]: Captive Portal:
[C][web_server:285]: Web Server:
[C][web_server:286]:   Address: mxmy-climate.local:8080
[C][mdns:116]: mDNS:
[C][mdns:117]:   Hostname: mxmy-climate
[C][esphome.ota:073]: Over-The-Air updates:
[C][esphome.ota:074]:   Address: mxmy-climate.local:3232
[C][esphome.ota:075]:   Version: 2
[C][esphome.ota:078]:   Password configured
[C][safe_mode:018]: Safe Mode:
[C][safe_mode:020]:   Boot considered successful after 60 seconds
[C][safe_mode:021]:   Invoke after 10 boot attempts
[C][safe_mode:023]:   Remain in safe mode for 300 seconds
[C][api:140]: API Server:
[C][api:141]:   Address: mxmy-climate.local:6053
[C][api:143]:   Using noise encryption: YES
[D][api:103]: Accepted 192.168.1.193
[W][component:172]: Component api cleared Warning flag
[D][status_led:030]: Restoring light state OFF
[D][api.connection:1801]: Home Assistant 2025.5.1 (192.168.1.193): Connected successfully
```

ssieb, 2025-05-10:
Your problem is `Recovery: failed, SCL is held low on the bus`
The scan is timing out on every address which takes too long and triggers the watchdog timer.  If you used uart logging instead of USB logging, you would be able to see that.
You can reduce the i2c timeout setting to avoid this.

closed 2025-05-10 by ssieb

MaxMyzer, 2025-05-10:
>  You can reduce the i2c timeout setting to avoid this.

I added `timeout: 10ms`, and it still bootlooped. In case `reduce` was a typo, I also tried `timeout: 10s`. Neither of these worked, both booted into safe mode after 10 boot attempts.

ssieb, 2025-05-10:
If you need more help, come ask on discord.

MaxMyzer, 2025-05-10:
SOLVED: the `on_boot.priority` of `90` that enables the bus is FAR too late to be able to be used ([Docs](https://esphome.io/components/esphome.html#on-boot)). Changed it to `750` and it worked flawlessly.