esphome/issues #7006: LD2450 doesn't use the set presence timeout

pottkopp, 2025-05-10:
### The problem

The LD2450 doesn't use the set presence timeout value. 
Instead it uses the following times to reset the presence states:
If Multi-Target is enabled: 36 seconds
If Multi-Target is disabled: 120 seconds

### Which version of ESPHome has the issue?

2025.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.0

### What platform are you using?

ESP32-IDF

### Board

esp32-c3-devkitm-1

### Component causing the issue

LD2450

### YAML Config

```yaml
substitutions:
  name: apollo-mtr-1-f8b110
  friendly_name: Apollo MTR-1 Küche f8b110
packages:
  ApolloAutomation.MTR-1: github://ApolloAutomation/MTR-1/Integrations/ESPHome/MTR-1_BLE.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: !secret ek_mtr1_kuche

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

In the log it doesn't say something like "Timeout of XX seconds reached. Resetting presence states."
It just says when a new timeout value is set by the user.
In Home Assistant, the web interface and the log the set timeout value is shown correctly. So it seems to be saved on the device.

I have two Apollo MTR-1 and both show this behavior.
Software details of both sensors:
Firmware (reported by HLKRadarTool): 2.04.23101915
Firmware (reported by ESPHome): 25.4.11.1
I tried using the MTR-1.yaml as well as MTR-1_BLE.yaml without a difference regarding this issue.

I tried:
- setting low (3 seconds) and high (1000 seconds) values
- rebooting the esp after setting the value in Home Assistant
- setting the value via the web interface
- factory resetting the esp and leaving it at the default value (5 seconds)
- HLK-LD25 fw version 2.02.23090617
- enabling/disabling BT and/or the led doesn't change the actual timeout it uses

My suspicion:
It checks the wrong memory address to load the presence timeout value. 
Instead of the one it's actually saved in it seems like it checks the address where (part of) the state of the Multi Target option is saved.

Workaround:
Create helpers in Home Assistant for the needed presence states and set the timeout in automations for the helpers to be (re)set.

Let me know if you need more details or if I should test something.

bharvey88, 2025-05-29:
When trying to duplicate I am unable to, however for me it seems to just.. not respect the timeout at all and as soon as i leave a zone the occupancy changes to zero.

https://github.com/ApolloAutomation/MTR-1/blob/f180e54e52cebfeb22fbd54f7d52885da4d0c7fd/Integrations/ESPHome/Core.yaml#L87 i dont see anything in the mtr-1 yaml that would cause this, so I believe it's a bug in the ld2450 component.

SaharMMesri, 2025-06-22:
Chiming in to say I am experiencing this same problem with two Apollo MTR-1 devices in my garage. I set the timeout to 5s and then later 15s but in both cases presence changes to 0 as soon as I leave the room, no delay at all. Again this applies to both sensors in the garage.

kbx81, 2025-06-24:
Just to be totally clear, we are specifically talking about these values not respecting the timeout value?:

[image]

If so, what is the expected behavior when a moving target transitions to a still target and vice-versa?

pottkopp, 2025-06-24:
> Just to be totally clear, we are specifically talking about these values not respecting the timeout value?:

Yes, plus the distance, speed etc values of the respective target.
> If so, what is the expected behavior when a moving target transitions to a still target and vice-versa?

The expected behavior is:
If a target is completely stationary not even moving a millimeter¹ (for the sensor) then the sensor waits for [timeout_value] (default: 5s) and after that it "forgets" the target resetting all of its values and subtracting it from the presence target count.
So it's not about the transition from moving target to stationary target and vice-versa (those work perfectly well), but it's about the transition from completely stationary target to "gone".

¹ I'm being so specific here because if a target just moves a few millimeters it still counts as a stationary target.

pottkopp, 2025-06-24:
Ah ok, so it isn't possible (yet) to change the timeout of the target values like target X distance to other values than the 36s if Multi-Target is enabled and 120s if Multi-Target is disabled right?
The presence timeout only applies after the target counts get set to 0. 
Then it holds the binary sensors in their current state for another [presence_timeout] seconds.

At least that's the behavior I'm getting now with the changes you suggested in my config.

Are there also binary sensors for the zones? At least a "has_target" one? 
That would make the presence timeout a lot more useful in my opinion.

kbx81, 2025-06-24:
Semi-related: you may want to take a look at using some filters to fine-tune sensor behavior.
- https://esphome.io/components/sensor/#sensor-filters
- https://esphome.io/components/binary_sensor/#binary-sensor-filters