esphome/issues #7008: Update to ESPHome 2025.4.1 impossible and also downgrade to 2025.2.2

emefff, 2025-05-10:
### The problem

After upgrading ESPHome to 2025.4.1 from 2025.2.2, upgrading the first device resulted in a lot of connection errors, for most of the libraries a second mirror was chosen. Updating the first device to 2025.4.1 seemed possible. Choosing a second EP for upgrading resulted in these strange errors show in the attached image. The error doesn't stop, just repeats like crazy until stop button is pressed. 
Then reverting to ESPHome 2025.2.2 via backup and trying to downgrade the first upgraded device, the one that seemed successful, results in the very same error! Very strange behaviour. 
However, all affected devices continue working, but it seems impossible to update in the future.
It looks like a  connection issue, I am behind a pfSense, however that was never a problem with ESPHome or HA. I can only imagine, that some of the links are in a blocklist (but then, why and how?). 
Thanks for any ideas or advice.

### Which version of ESPHome has the issue?

2025.4.1, 2025.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.1

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

no component, whole device affected

### YAML Config

```yaml
no YAML, because YAML seems not important. Affected YAMLs worked for years.
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

[image]

ProFiLeR4100, 2025-05-12:
Looks like you've used all your public quota platformio.org or sent too much connections and because of that server refused opening a connection for your IP.

emefff, 2025-05-12:
I just tried updating from 2025.2.2 to the new 2025.4.2.

With the first device updating I get a similar error like above but the 'Looking for a mirror...' seems to work and the device is updated. See the log file from esphome--6 device:

[logs_esphome--6_run.txt](https://github.com/user-attachments/files/20162105/logs_esphome--6_run.txt)

Then, as soon as repeat the same on another device, say esphome--5, I get the same error as above. I can't export a log, the text runs down like crazy. I have to press the STOP button. Then, ESPHome marks the device as updated. Not good.

emefff, 2025-05-12:
I retried updating the second device now, 5 hours later. I still get the connection refused, but a mirror is found. So my second device is updated now. It seems I have to wait for 4 or 5 hours until this blocking is cleared.

It probably means, it'll take a few days to update all my devices.

Clearly, there's something wrong in how the repositories are downloaded from platformio. First of all, the mirror could always work, and second, why is the same page called over and over again until I get 'connection refused' instead of downloading the mirror on a second device just a few minutes later? So, it happened again on the (in total) third device. Tomorrow I will update this device and then wait......

ProFiLeR4100, 2025-05-13:
1) PlatformIO uptime robot link in the footer is outdated, so can't check current status.
https://stats.uptimerobot.com/ZWBXVu5NGx
2) It is an [old issue](https://github.com/platformio/platformio-core/issues/1658), but it is mentioned here that if you disconnect WAN it will compile from cache, you can try that.

emefff, 2025-05-13:
Thanks again. I had no idea of it being an old issue. I tried, but it did not read any local copies, the log below took about 10min only to be stuck in the end. Also during updating the devices, it seems every device does its own download. So my theory is, that this maybe only works at the device level, not 'across' devices (would mean one device downloads the repo and can share these data with other devices in need for the same program versions). 

[logs_espcam-2_run.txt](https://github.com/user-attachments/files/20187291/logs_espcam-2_run.txt)

I didn't touch pfSense for nearly two years now, it runs for months usually, I only reboot occasionally. Also, it hasn't had updates for a long time, since people are waiting for CE2.8 for over a year now. Only blocking lists etc. are updated daily.

emefff, 2025-05-27:
This issue seems to be resolved with 2025.5.1