esphome/issues #7009: framework-type "arduino" leads to broken build

snailcatcher, 2025-05-10:
### The problem

Hi there,
I try to flash esphome on [one of these boards](https://www.amazon.de/diymore-Development-ESP32-S3FH4R2-Dual-Core-Prozessor-Hauptfrequenz/dp/B0F1CR5RHV). With the *esp-idf* framework it seems to work, and I can connect the esp to zto homeassistant, but with *arduino* the esp is not even responding after it is flashed.

This ends in a bootloop or something similar and I can not even few the logs of my esp
```
substitutions:
  device_name: esp-heizung
  friendly_name: ESP Heizung

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}
  platformio_options:
    upload_speed: 115200

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 4MB
  variant: ESP32S3
  framework:
    type: arduino

logger:
```

This works fine
```
substitutions:
  device_name: esp-heizung
  friendly_name: ESP Heizung

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}
  platformio_options:
    upload_speed: 115200

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 4MB
  variant: ESP32S3
  framework:
    type: esp-idf

logger:
```

Only difference is the *esp32.framework.type*.
I want to use some intgrations which just work with *arduino* so it is no solution for me to use *esp-idf*.

Thank you!

### Which version of ESPHome has the issue?

2025.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.1

### What platform are you using?

ESP32

### Board

ESP32-S3 based on ESP32-S3FH4R2

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.4.1
INFO Reading configuration /config/esphome/esp-heizung.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp-heizung (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
No dependencies
RAM:   [          ]   4.2% (used 13640 bytes from 327680 bytes)
Flash: [=         ]  14.7% (used 269061 bytes from 1835008 bytes)
========================= [SUCCESS] Took 6.13 seconds =========================
INFO Successfully compiled program.
esptool.py v4.8.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded Flash 4MB (XMC), Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: b4:3a:45:ad:a7:e0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00010000 to 0x00051fff...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Compressed 269456 bytes to 151656...
Wrote 269456 bytes (151656 compressed) at 0x00010000 in 2.4 seconds (effective 904.6 kbit/s)...
Hash of data verified.
SHA digest in image updated
Compressed 15040 bytes to 10364...
Wrote 15040 bytes (10364 compressed) at 0x00000000 in 0.3 seconds (effective 454.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.1 seconds (effective 450.1 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 600.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM0 with baud rate 115200

After this it does not continue for the broken build
```

ssieb, 2025-05-10:
What integrations require arduino?
Do you have a USB uart adapter?

snailcatcher, 2025-05-11:
Hi @ssieb 

Its not in the official release, but in a PR:
https://github.com/esphome/esphome/pull/4453

No the ESP has a jlink onboard which I use for flashing and reading the logs.

ssieb, 2025-05-11:
Why are you using jlink when the board has a USB port and uart pins?  The arduino framework might not have support for that.

snailcatcher, 2025-05-11:
Oh sorry I mean JTAG not jlink. I use the USB port which is on the board itself. So I have not build some weird stuff.

I connect it via the usb ports of my Homeassistant server.
[image]

So the main problem is, that I have the most simple config I can think of (see above) and it is working with *esp-idf*, but when I use *arduino* the ESP is not even booting correctly.<

I think this is not intended behaviour, isn't it?
If I can help fixing this issue I will do, but I don't know at this moment how I can provide more information.

ssieb, 2025-05-12:
You need to get serial logs to find out why it's not booting.

snailcatcher, 2025-05-12:
It it possible to get them without an other than the ESP board itself?
And if yes how?

ssieb, 2025-05-12:
You need a USB uart adapter connected to the TX pin and GND on the board.
I suggest you come ask on discord for further help.

closed 2025-09-24 by github-actions[bot]