esphome/issues #7011: Konnected Alarm Panel Pro Not Updating

heedmm, 2025-05-12:
### The problem

Konnected Alarm Panel is working normally but will not accept an update beyond 2025.4.0. 

### Which version of ESPHome has the issue?

2025.4.1 & 2025.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.1

### What platform are you using?

ESP32

### Board

Konnected Alarm Panel Pro

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-05-12:
You need to provide more info than that, but it's almost certainly the same issue that their components are not compatible with the latest esphome changes.

closed 2025-05-12 by ssieb

portigui, 2025-05-16:
Please see https://community.konnected.io/t/attempting-to-update-with-esphome-2025-4-1/39678 to find out more about this error.