esphome/issues #7015: LD2420 does not work on GPIO43 and GPIO44 of the ESP32-S3

patrick3399, 2025-05-12:
### The problem

Other UART sensors work when connected to GPIO43 and GPIO44.
The LD2420 works when connected to other GPIOs.

### Which version of ESPHome has the issue?

2025.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.1

### What platform are you using?

ESP32

### Board

ESP32-S3-DevKitC

### Component causing the issue

LD2420

### YAML Config

```yaml
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
uart:
  tx_pin: GPIO43
  rx_pin: GPIO44
  baud_rate: 115200
  id: uart_2420
ld2420:
text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware
binary_sensor:
  - platform: ld2420
    has_target:
      name: Presence
```

### Anything in the logs that might be useful for us?

```txt

```

swoboda1337, 2025-05-13:
```

patrick3399, 2025-05-13:
@swoboda1337 Yes, I disable the HW UART logger.

swoboda1337, 2025-05-15:
You could also try to add this option:
```
esp32:
  framework:
    sdkconfig_options:
      CONFIG_ESP_CONSOLE_NONE: "y"
```

ssieb, 2025-05-15:
Is there a reason you need to use those pins?

patrick3399, 2025-05-17:
Because the existing device uses GPIO43 and GPIO44 to connect external sensors, they must be used.

swoboda1337, 2025-05-17:
Did you try with CONFIG_ESP_CONSOLE_NONE?

patrick3399, 2025-05-17:
@swoboda1337 It still doesn't work, but I noticed that if I change the TX pin to a different one, it works properly.

closed 2025-12-09 by github-actions[bot]