esphome/issues #7017: when sen5x (air-1) is power cycled, it reports impossible out-of-range NOX values

diablodale, 2025-05-12:
### The problem

upstreaming issue https://github.com/ApolloAutomation/AIR-1/issues/59

air-1 sensor that uses the send5x it sends multiple impossible and invalid values for NOX to home assistant.

Air-1 factory firmware: 24.10.11.1 (ESPHome 2024.12.2)

NOX is an index and the only valid values are integers between 1-500 inclusively. https://sensirion.com/media/documents/9F289B95/6294DFFC/Info_Note_NOx_Index.pdf

[image]

These impossible out-of-range NOX values are then saved into the statistics and history which leads to borked 🙈 data, charts, and reports.

This is likely due to the low-level ESPHome module not correctly handling power-on. Official Sensirion https://sensirion.com/media/documents/2B6FC1F3/649C3D0E/PS_AN_Read_RHT_VOC_and_NOx_RAW_signals_v2_D1.pdf

> RAW NOx: During the first 10..11 seconds after power-on or device reset, this value will be 0xFFFF as well.

I suspect the ESP code needs to filter the data. When it gets values outside the range of 1-500 inclusively, to not report data or to report a silent error like unavailable.

Note that I have seen values other than FFFF reported to home assistant. Looking at the SQL tables directly for the NOX data, I see a value 3277
in the states table...

[image]

in the short term statistics table

[image]

and statistics table

[image]

You'l notice the error data is 3277. That is likely 32767 which is being scaled by .1 to 3276.7 and then rounded to 3277.
and of course, 32767 is 0x7FFF

Their bug is likely here https://github.com/esphome/esphome/blob/bec9d914199035fe765fc8fdd746ced566eadca2/esphome/components/sen5x/sen5x.cpp#L361
where they are doing exactly as I guessed. Scaling by 10 and checking for 0xFFFF.
I'll bet the sen5x hardware is sending 0x7FFF and therefore the xFFFF check fails.

The fix is to more safely filter the values. Any value outside 1-500 is invalid and therefore it should be filtered to be within those limits.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.1

### What platform are you using?

ESP32

### Board

apollo air-1

### Component causing the issue

sen5x

### YAML Config

```yaml
https://github.com/ApolloAutomation/AIR-1/tree/main/Integrations/ESPHome
```

### Anything in the logs that might be useful for us?

```txt

```

diablodale, 2025-05-12:
Likely also bugs in the same code place https://github.com/esphome/esphome/blob/bec9d914199035fe765fc8fdd746ced566eadca2/esphome/components/sen5x/sen5x.cpp#L353 for other data types. The official sensirion pdf writes

> 0xFFFF for uint16, 0x7FFF for int16

Humidity and Temperature are int16. Therefore, the checks for 0xFFFF are wrong on lines 353 and 356. They should check for 0x7FFF.

But I think the better approach is still to limit lower and upper to only valid values as that PDF is wrong with NOx and might be wrong with other categories and their magic numbers.

kbx81, 2025-05-13:
I was digging around Sensiron's repos with example code for the SEN5X this afternoon and, while it seems many of their examples lack any validation of the values, [this one](https://github.com/Sensirion/embedded-i2c-sen5x/blob/4584621fc5a4efbcbe04b1712a759ff657a522fb/sen5x_i2c_example_usage.c#L164-L185) does not.

I'll at least get a PR going to update our code to match that example and (hopefully) this will alleviate the issue you're seeing, @diablodale.

kbx81, 2025-05-13:
@diablodale please test the PR by adding it as an external component to your AIR-1's config. I left a comment on the PR with what you need to add to do so.

I'm not seeing this behavior (wild values) on my AIR-1 so I'm not able to confirm it myself. Of course, we can still add some extra validation based on expected VOC/NOx ranges, but I'm curious if this small fix alone at least addresses your initial issue.

closed 2025-05-18 by jesserockz