esphome/issues #7020: Linking fails on RP2040 with ESPHome 2025.4.2

emefff, 2025-05-14:
### The problem

Linking fails on my one and only device with RP2040 with the current ESPHome 2025.4.2.
The issue in #7008 is also visible on this device, but error in the end is new.

### Which version of ESPHome has the issue?

2025.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.1

### What platform are you using?

RP2040

### Board

rpipicow

### Component causing the issue

linker

### YAML Config

```yaml
probably not important
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs/esphome--14/FrameworkArduino/wiring_digital.cpp.o
Compiling .pioenvs/esphome--14/FrameworkArduino/wiring_private.cpp.o
Compiling .pioenvs/esphome--14/FrameworkArduino/wiring_pulse.cpp.o
Compiling .pioenvs/esphome--14/FrameworkArduino/wiring_shift.cpp.o
Archiving .pioenvs/esphome--14/libFrameworkArduino.a
Linking .pioenvs/esphome--14/firmware.elf
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `std':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0xda): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x124): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x17c): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `__sfp':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:201:(.text+0x1ea): undefined reference to `__retarget_lock_init_recursive'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esphome--14/firmware.elf] Error 1
========================= [FAILED] Took 154.77 seconds =========================
```

ssieb, 2025-05-14:
Why would you assume that the yaml isn't important?  At least show the `rp2040:` section.
I don't get any errors compiling for rp2040.

emefff, 2025-05-14:
rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

ssieb, 2025-05-14:
Yes, but it's no longer valid.  Remove the framework section.

closed 2025-05-14 by ssieb

emefff, 2025-05-14:
Now it links but it doesn't reconnect to the API, isn't that great?