esphome/issues #7021: RP2040 not reconnecting to API after changing framework

emefff, 2025-05-14:
### The problem

Follow up on #7020. The framework part in the .yaml was removed. But the RP2040 doesn't reconnect to the API anymore. It's completely gone.

### Which version of ESPHome has the issue?

2025.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5

### What platform are you using?

RP2040

### Board

rpipicow

### Component causing the issue

?

### YAML Config

```yaml
esphome:
  name: esphome--14

rp2040:
  board: rpipicow

# Enable logging
logger:
  
# Enable Home Assistant API
api:
  reboot_timeout: 1min 
  encryption:
    key: ***** REDACTED *****

ota:
  platform: esphome
  password: ***** REDACTED *****

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  output_power: 20dB #8.5-20db
  manual_ip:
    static_ip: ***** REDACTED *****
    gateway: ***** REDACTED *****
    subnet: 255.255.255.0
  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Esphome--14 Fallback Hotspot"
    password: ***** REDACTED *****

status_led:
  pin:
    number: GPIO25 

one_wire:
  - platform: gpio
    pin: GPIO5

i2c: 
  sda: GPIO20 
  scl: GPIO21 
  scan: true
  id: bus_a    

sensor:
  - platform: dallas_temp
    address: 0x230922503ac7d228
    name: "Schlafzimmer oben Pico Onboard Dallas Temperatur"
    accuracy_decimals: 1  
    filters:
      - filter_out: nan
      - filter_out: 85.0   
  - platform: wifi_signal
    name: "Schlafzimmer oben Pico WiFi Signal"
    update_interval: 300s # defaults to 60s
    accuracy_decimals: 1
  - platform: uptime
    name: "Schlafzimmer oben Pico Uptime"
    update_interval: 300s # defaults to 60s
    filters:
      - lambda: return x / 3600;
    unit_of_measurement: "h"
    accuracy_decimals: 1
  - platform: bmp280_i2c
    temperature:
      name: "Schlafzimmer oben Pico BMP280 Temperatur"
      oversampling: 16x
    pressure:
      name: "Schlafzimmer oben Pico BMP280 Druck"
    #humidity:
    #  name: "Schlafzimmer 2 Pico BME280 Feuchtigkeit"
    address: 0x76
    update_interval: 60s
  - platform: bh1750
    name: "Schlafzimmer oben Pico BH Lichtstärke"
    address: 0x23
    update_interval: 60s

button:
  - platform: restart
    name: "Schlafzimmer oben Restart"
    id: schlafzimmer2_restart_button
    disabled_by_default: true
    on_press:
      then:
        - logger.log: Button Pressed
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.4.2
INFO Reading configuration /config/esphome/esphome--14.yaml...
INFO Starting log output from 192.168.178.165 using esphome API
WARNING Can't connect to ESPHome API for esphome--14 @ 192.168.178.165: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.178.165', port=6053))]: [Errno 113] Connect call failed ('192.168.178.165', 6053) (SocketAPIError)
INFO Trying to connect to esphome--14 @ 192.168.178.165 in the background
```

ssieb, 2025-05-14:
That reboot timeout is very short.  Try getting serial logs to see what's happening.

emefff, 2025-05-14:
It's mounted at a height of 4m+ near the roof. I can't get to it easily.

ssieb, 2025-05-14:
Can you ping it?

emefff, 2025-05-14:
(base) user@HP-Z8-G4-Workstation:~$ ping 192.168.178.165
PING 192.168.178.165 (192.168.178.165) 56(84) bytes of data.
From 192.168.178.65 icmp_seq=1 Destination Host Unreachable
From 192.168.178.65 icmp_seq=2 Destination Host Unreachable
From 192.168.178.65 icmp_seq=3 Destination Host Unreachable
From 192.168.178.65 icmp_seq=4 Destination Host Unreachable
From 192.168.178.65 icmp_seq=5 Destination Host Unreachable
From 192.168.178.65 icmp_seq=6 Destination Host Unreachable

ssieb, 2025-05-14:
I don't know.  This platform change happened a long time ago and I'm not aware of any other reports of this kind of issue.  I guess you'll have to get it.
The OTA update was successful?

emefff, 2025-05-14:
Yes, that was successful. It was one of the last lines I received from the device. I will maybe try later, it involves a very long ladder... :-(

emefff, 2025-05-14:
I flashed the device via USB, now it's working. It always had a bad connection although it's only 5m from the next AP. It's sitting on concrete, maybe that's a bad influence.

closed 2025-05-14 by emefff