esphome/issues #7023: Light: transitions use wrong initial color with color_correct enabled

aveao, 2025-05-14:
### The problem

Background (feel free to skip):

I have a light that is remotely controlled that is set with a cronjob every minute, going through home assistant API (it adjusts the LED strip on my work table so that if I'm not on my desk it turns off, and if I'm in a video call, the brightness goes up to 100%). Most minutes it is just set to the same value as the last one.

Earlier today I switched to using `color_correct` instead of setting the correction in my script setting the light, and as a result now my light turns slightly green every minute for the default transition length. I thought I was going crazy, but nope, I can clearly reproduce it by manually setting it and looking at the light, and as such had to report it.
---

Clear and concise description of what the problem:

With `color_correct` enabled, the transition does not seem to account for the color correction that was already applied, which seems to lead to incorrect colors during the transition period. This is most visible when setting a light to the same value it is already at, with a significant `color_correct` (in my case: 10% on blue channel).

### Which version of ESPHome has the issue?

2025.4.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

esp32dev

### Component causing the issue

light (base)

### YAML Config

```yaml
Most concise form I tested:

  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    pin: GPIO22
    num_leds: 56
    chipset: ws2812
    name: "ave desk light"
    id: "ave_desk_light"
    color_correct: [100%, 100%, 90%]
```

### Anything in the logs that might be useful for us?

```txt
Nothing particularly useful sadly.

[D][light:036]: 'ave desk light' Setting:
[D][light:051]:   Brightness: 50%
[D][light:055]:   Color brightness: 99%
[D][light:058]:   Red: 100%, Green: 96%, Blue: 87%
[D][light:085]:   Transition length: 0.5s
[D][light:036]: 'ave desk light' Setting:
[D][light:051]:   Brightness: 50%
[D][light:055]:   Color brightness: 99%
[D][light:058]:   Red: 100%, Green: 96%, Blue: 87%
[D][light:085]:   Transition length: 0.5s
```

### Additional information

My educated guess is that this lerp:

https://github.com/esphome/esphome/blob/191afd3e6926300342f178d08c7035ed739f9072/esphome/components/light/light_color_values.h#L86-L100

which is used by the transition transformer:

https://github.com/esphome/esphome/blob/191afd3e6926300342f178d08c7035ed739f9072/esphome/components/light/transformers.h#L54

and is configured with current_values as opposed to remote_values:

https://github.com/esphome/esphome/blob/191afd3e6926300342f178d08c7035ed739f9072/esphome/components/light/light_state.cpp#L235C18-L242

which is supposed to contain the internal color state, not the externally set and reported one. I suspect internal color state's color values are set with color correction (255, 255, 229 etc), whereas the target is set without (255, 255, 255). Why it is visible when the correction should be done during transition as well is not immediately clear to me.

markusressel, 2025-05-24:
I am seeing the same issue.

I am not using the `color_correct` option to change the individual color channels, but simply to limit the output power of the strip (f.ex. `color_correct: [ 70%, 70%, 70%, 70% ]`). My light is adapted every minute (like with OP), and each call causes a slight dip in brightness and change in color (think, not as clear to me).

I though about using the `min_power` and `max_power` options of the `output` component, but since I am using an addressable strip with the `esp32_rmt_led_strip` thats not an option. I do have a `power_supply` component, but that also doesn't seem to provide options to limit the power.

The device I am using is a [QuinLED Dig2Go](https://quinled.info/quinled-dig2go/).

markusressel, 2025-10-29:
I just noticed #7236 

So I guess if we want this to get any attention, we have to move this to the esphome/esphome issue tracker?
https://github.com/esphome/esphome/issues

I am definitely still seeing this behavior, so I assume there were not made any efforts yet to investigate.
@aveao would you move/recreate this issue to the other repository, to hopefully get more attention?