esphome/issues #7026: LD2450 stops detecting randomly after x hours

reylinux, 2025-05-16:
### The problem

Hi I have an issue with my presence Sensor using LD2450. When it works, it is great, however it stops detecting anything after awhile.

The example below, last detection was 8 hours ago. There was no power cycle, wifi has been solid, no other drop off. I've also set an automation to put the coordinates back if the detection zone is changed to disabled.

I happen to have 3x sensors using this component and it seems that only one component is affected. The rest has been solid and very rarely dropped off.

I have done multiple factory reset and none helped. They fixed it temporarily, and then at some point broke down again.

Please let me know if I need to post something else. This is the first time I post an issue on ESPhome 

[image]

[image]

### Which version of ESPHome has the issue?

2025.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.1

### What platform are you using?

ESP32

### Board

Apollo MTR-1

### Component causing the issue

LD2450

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

rosstroha, 2025-05-28:
@reylinux I'm having the same issue. Would you mind sharing the automation you wrote to reset the values? It would be very helpful!

reylinux, 2025-05-28:
> [@reylinux](https://github.com/reylinux) I'm having the same issue. Would you mind sharing the automation you wrote to reset the values? It would be very helpful!

Hey, sure. Look below. Probably easier to set up yourself since it contains unique IDs. 

https://pastebin.com/DUZg4sWY

rosstroha, 2025-05-29:
Thank you so much!

kbx81, 2025-06-23:
@reylinux @steelcube12 @rosstroha I would like to confirm the LD2450 firmware version you each have on each of your device(s).
Also, would you each confirm that you are enabling multi-target tracking (or not) and/or have one or more zones configured for the sensor?
Thanks!

kbx81, 2025-06-24:
> confirm the LD2450 firmware version

@steelcube12 25.4.11.1 is not the LD2450 firmware version. Please check again. 😇

bdraco, 2025-07-05:
If the restart button fixes it, it might just be a firmware bug and we could add a watchdog like

```cpp
 // Add to ld2450.h:
  uint32_t last_data_received_ = 0;
  uint32_t watchdog_timeout_ = 30000; // 30 seconds

  // Add to handle_periodic_data_:
  this->last_data_received_ = App.get_loop_component_start_time();

  // Add to loop():
  if (App.get_loop_component_start_time() - this->last_data_received_ > this->watchdog_timeout_) {
    ESP_LOGW(TAG, "No data received for %d ms, restarting sensor", this->watchdog_timeout_);
    this->restart_and_read_all_info();
    this->last_data_received_ = App.get_loop_component_start_time(); // Reset to prevent continuous restarts
  }
```