esphome/issues #7031: Bug Report: I2S Media Player repeats last phrase every 3–5 minutes after playback (ESPHome 2025.4.x)

wolters1313, 2025-05-20:
### Bug Report: I2S Media Player repeats last phrase every 3–5 minutes after playback (ESPHome 2025.4.x)

**ESPHome Version**: 2025.4.2  
**Home Assistant Core Version**: 2025.4.2  
**Board**: ESP32 (esp32dev)  
**Framework**: Arduino (also tested with ESP-IDF)  
**Integration**: `media_player.i2s_audio`

----

#### 🔍 Problem Description:

After updating to ESPHome 2025.4.1 and 2025.4.2, the `media_player` based on `i2s_audio` does not enter an idle state after playing a TTS message.

Instead, **it repeats the last phrase automatically every ~3–5 minutes**, even without any new playback requests.

----

#### ✅ Previously Working:

This worked correctly in:
- ESPHome 2024.11.x
- ESPHome 2024.12.x

----

#### 🧪 Workarounds Tried:

- ✅ Downgrade to 2024.12.x: fixes the issue
- ❌ Arduino framework: no effect
- ❌ `idle_timeout`, `on_idle`, `media_player.stop`: not working or causes config errors
- ❌ Sample rate / bit depth changes: no effect

----

#### 📄 YAML Config:

```yaml
esp32:
  board: esp32dev
  framework:
    type: arduino

i2s_audio:
  i2s_lrclk_pin: GPIO33
  i2s_bclk_pin: GPIO26

media_player:
  platform: i2s_audio
  name: ESPHome I2S Media Player
  dac_type: external
  i2s_dout_pin: GPIO27
  mode: mono

NdK73, 2025-05-25:
Can confirm: same here (althought different board: I'm using an esp32-eth01 with mic and spk on a breadboard).

Moreover:
  - first interaction with va works, but after that nothing is recognized till a reboot
  - esp-idf framework does not work anymore (before last update it required "version: 4.4.8" and "platform_version: 5.4.0" but now with those settings it does not compile): va never recognizes the wake word
  - nothing appears in the log when last announce/answer gets repeated

wolters13, 2025-05-31:
Hello, I just wanted to follow up and confirm that this issue is still present in Home Assistant Core version 2025.5.3.

The I2S media player on ESP32 continues to repeat the last TTS message every ~3 minutes, instead of returning to idle after playback.

Temporary workarounds (like playing a dot or silence after the message) don’t fully resolve the issue.

Is there any update or acknowledgment on this bug? It would be great to know if it’s being looked into.

Thanks a lot for your work on ESPHome!