esphome/issues #7032: Can't connect to ESPHome API for dsmr-reader

carabao72, 2025-05-20:
### The problem

INFO ESPHome 2025.4.2
INFO Reading configuration /config/esphome/smartmeter.yaml...
INFO Starting log output from 192.168.1.45 using esphome API
WARNING Can't connect to ESPHome API for dsmr-reader @ 192.168.1.45: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.45', port=6053))]: [Errno 113] Connect call failed ('192.168.1.45', 6053) (SocketAPIError)
INFO Trying to connect to dsmr-reader @ 192.168.1.45 in the background

### Which version of ESPHome has the issue?

ESPHome 2025.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2025.5.2, supervisor-2025.05.1

### What platform are you using?

ESP8266

### Board

board: d1_mini

### YAML Config

```yaml
# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxx"

substitutions:
  name: dsmr-reader
  friendly_name: DSMR Reader

esphome: 
  name: dsmr-reader

esp8266: 
  board: d1_mini

ota:
  platform: esphome
  password: "xxxxxx"

wifi:
  ssid: "xxxxx"
  password: "xxxxxxxx"

# Optional manual IP
  manual_ip: 
    static_ip: 192.168.1.45
    gateway: 192.168.1.1
    subnet: 255.255.255.128
    dns1: 192.168.1.1

  ap:
    ssid: "xxxxxx"
    password: "xxxxxxxx"

captive_portal:

logger:
  # Serial logging is disabled by setting the logger baud rate to 0.
  # Otherwise the logger will occupy the hardware UART, making it unavailable
  # for receiving smart meter data on pin D7 (GPIO13).
  baud_rate: 0
  level: VERBOSE
  
uart:
  rx_pin: D7
  baud_rate: 115200
  rx_buffer_size: 1500

dsmr:
  request_pin: D5
  request_interval: 15s

sensor:
  - platform: dsmr
    energy_delivered_tariff1:
      name: ${friendly_name} Energy Delivered Tariff 1
      id: Delivered_Tariff1
      state_class: total_increasing
    energy_delivered_tariff2:
      name: ${friendly_name} Energy Delivered Tariff 2
      id: Delivered_Tariff2
      state_class: total_increasing
    energy_returned_tariff1:
      name: ${friendly_name} Energy Returned Tariff 1
      id: Returned_Tariff1
    energy_returned_tariff2:
      name: ${friendly_name} Energy Returned Tariff 2
      id: Returned_Tariff2
    power_delivered:
      name: ${friendly_name} Power Consumed
    power_returned:
      name: ${friendly_name} Power Returned
    electricity_failures:
      name: ${friendly_name} Electricity Failures
    electricity_long_failures:
      name: ${friendly_name} Electricity Long Failures
    voltage_l1:
      name: ${friendly_name} Voltage L1
    current_l1:
      name: ${friendly_name} Current L1
    voltage_l2:
      name: ${friendly_name} Voltage L2
    current_l2:
      name: ${friendly_name} Current L2
    voltage_l3:
      name: ${friendly_name} Voltage L3
    current_l3:
      name: ${friendly_name} Current L3    
    power_delivered_l1:
      name: ${friendly_name} Power Delivered L1
    power_returned_l1:
      name: ${friendly_name} Power Returned L1
    power_delivered_l2:
      name: ${friendly_name} Power Delivered L2
    power_returned_l2:
      name: ${friendly_name} Power Returned L2
    power_delivered_l3:
      name: ${friendly_name} Power Delivered L3
    power_returned_l3:
      name: ${friendly_name} Power Returned L3
    active_energy_import_current_average_demand:
      name: ${friendly_name} Current Average Quarterly Demand for Peak Tarrif Belgium     
    active_energy_import_maximum_demand_running_month:
      name: ${friendly_name} Current Month Maximum Quarterly Demand for Peak Tarrif Belgium
    active_energy_import_maximum_demand_last_13_months:
      name: ${friendly_name} 13 Month Maximum Quarterly Demand for Peak Tarrif Belgium

  - platform: wifi_signal
    name: "WiFi Signal Strenght"
    id: wifi_signal_db
    unit_of_measurement: dBm
    update_interval: 60s

  - platform: copy
    source_id: wifi_signal_db
    name: "WiFi Signal (percent)"
    id: wifi_signal_pct
    unit_of_measurement: '%'
    accuracy_decimals: 0
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    entity_category: "diagnostic"

  - platform: template
    name: ${friendly_name} Energy Delivered Total
    unit_of_measurement: kWh
    lambda: |-
      return (id(Delivered_Tariff1).state + id(Delivered_Tariff2).state);
  - platform: template
    name: ${friendly_name} Energy Returned Total
    unit_of_measurement: kWh
    lambda: |-
      return (id(Returned_Tariff1).state + id(Returned_Tariff2).state);
      
text_sensor:
  - platform: dsmr
    identification:
      name: ${friendly_name} Identification
    p1_version_be:
      name: ${friendly_name} Version BE
    message_long:
      name: ${friendly_name} Message Long  
  - platform: wifi_info
    ip_address:
      name: "IP Address"
    ssid:
      name: "WiFi SSID"
    bssid:
      name: "WiFi BSSID"
```

### Anything in the logs that might be useful for us?

```txt

```

jesserockz, 2025-05-22:
Please connect to the device via usb and get the serial/console logs and paste them here

jesserockz, 2025-05-22:
Those are your system logs of the usb being plugged in, but not the logs coming from the device itself. 
You need to use a terminal/console to read the serial port output. 
If you are running ESPHome locally, you can use `esphome logs your-config.yaml --device /dev/ttyUSB0`

carabao72, 2025-05-28:
Hello

Apologies, problem solved. I did not see that the port "P1" of the meter had been disabled (by mistake or because of a technical problem) remotely by the power grid manager. After contacting it, port "P1" was reactivated and everything works as before.