esphome/issues #7033: SSD1322 OLED Display does not display properly

Enrique-Cordoba-Saez, 2025-05-20:
### The problem

[image]

### Which version of ESPHome has the issue?

Unknown

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Unknown

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32

### Component causing the issue

SSD1322 OLED Display

### YAML Config

```yaml
esphome:
  name: mi-esp32
  friendly_name: Mi ESP32
  on_boot:
    priority: -100
    then:
      - lambda: |-
          pinMode(21, OUTPUT); // Fuerza SDA HIGH
          digitalWrite(21, HIGH);
          delay(10);
          for (int i = 0; i < 10; i++) {
            pinMode(22, OUTPUT);
            digitalWrite(22, HIGH);
            delay(2);
            digitalWrite(22, LOW);
            delay(2);
          }
          pinMode(22, INPUT);
          pinMode(21, INPUT); // Devuelve SDA a modo normal

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  level: DEBUG
  
api:
  encryption:
    key: "sPNJb0u/fdvDRFGbsv8YlsZ5DTWl154hrsSALoLdce8="
  reboot_timeout: 0s
    
ota:
  platform: esphome
  password: "173f4d052c240e3716de7d5d29a8ffb9"

wifi:
  networks:
[... 391 lines cut ...]
      name: "Dioxido de Nitrogeno"
      id: "sens_NO2"
    carbon_monoxide:
      name: "Monoxido de Carbono"
      id: "sens_CO"
    hydrogen:
      name: "Hidrogeno"
      id: "sens_h2"
    ethanol:
      name: "Etanol"
      id: "sens_C2H5OH"
    methane:
      name: "Metano"
      id: "sens_ch4"
    ammonia:
      name: "amoniaco"
      id: "sens_nh3"    
    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-05-20:
Looks like that model is not suitable for your display.
Can you provide a link to the display?
What happens if you don't rotate it?

Enrique-Cordoba-Saez, 2025-05-21:
Oh sorry, I didnt mean to copy that line, the result is pretty much the same, that white band to the left is still there and the text also stays mirrored. Also I tried with another screen that used to work with that same board and still it is the same.

old screen: https://tcclcd.en.made-in-china.com/product/ZyVQONcjpBWJ/China-3-12-Inch-256-64-LCD-Display-SSD1322-Driver-16-Pin-Green-OLED-Module.html

this screen: https://es.aliexpress.com/item/1005005853598865.html?spm=a2g0o.order_list.order_list_main.5.d283194d0FO4It&gatewayAdapt=glo2esp

ssieb, 2025-05-21:
The old screen used to work in esphome?  Which version?

Enrique-Cordoba-Saez, 2025-05-21:
I am not sure of the version, this is the docker compose file I used:

services:

  esphome:
    container_name: esphome
    image: esphome/esphome
    devices:
      - COM3
    environment:
      ESPHOME_DASHBOARD_USE_PING: "true"
    volumes:
      - C:\Users\Kry0m\Desktop\compose\esphome\config:/config
     # - /etc/localtime:/etc/localtime:ro
    restart: always
    privileged: true
    ports:
      - "6052:6052"

ssieb, 2025-05-22:
Are you saying that the old screen used to work, but doesn't now?

closed 2025-10-05 by github-actions[bot]