esphome/issues #7035: ESPHome version 2025.5.0 compiled warning

gevezeus, 2025-05-21:
### The problem

/config/.esphome/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
/config/.esphome/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:73: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)

### Which version of ESPHome has the issue?

2025.5.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2025.05

### What platform are you using?

ESP8266

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

randybb, 2025-05-21:
Just a warning and it is related to the old framework which is esphome using for these ancient mcu's.

closed 2025-05-21 by randybb

onlize, 2025-05-21:
I have the same issue.

rinyheuvelman, 2025-06-03:
Same issue, though "only a warning", it irritates me seeing these kind of errors since I'm using esp's for quit a lot of stuff for quit a long time :(