esphome/issues #7036: HA AddOn: Update All should skip offline devices.

SynoUser-NL, 2025-05-21:
### The problem

Hi,

In the Home Assistant ESPhome AddOn there is an option Update All in the right upper corner. When clicking this, the update all process appears to start with the first alphabetical (named) device. If this (or any device in the proces I assume) is offline, the proces stops or waits.
Please make it an option to skip any offline devices so that time & resources are spent only on active devices.

### Which version of ESPHome has the issue?

multiple

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5

### What platform are you using?

ESP32

### Board

Not relevant

### Component causing the issue

Not relevant

### YAML Config

```yaml
Not relevant
```

### Anything in the logs that might be useful for us?

```txt
Not relevant
```

randybb, 2025-05-21:
It is not an issue as it is designed so :) This is more a feature request and there are multiple feature requests asking for the same thing, for example this one https://github.com/esphome/feature-requests/issues/1363

SynoUser-NL, 2025-05-21:
One could argue that multiple Feature Requests asking for the same thing can be considered a bug\issue.. 
One could also argue that, although it is designed to do so, this is a design issue.. and thus still an issue..

Wether this is an issue or an FR, that's really just a label. Apparently more people feel this is something that needs to be addressed.

jesserockz, 2025-05-22:
Closing as this is not a bug/issue but a FR and exists already.

> One could argue that multiple Feature Requests asking for the same thing can be considered a bug\issue

No, it is a feature request that is no different from others, just a bit more popular. Nobody has stepped up and made the PR though 🤷 

Jesse

closed 2025-05-22 by jesserockz