esphome/issues #7037: Tuya MCU not working anymore in Duux smart fan

jkufner, 2025-05-21:
### The problem

After the upgrade from 2025.4.2 to 2025.5.0 and updating my ESP32 in Duux fan, it stopped working. When switching oscillation modes and other things, I get an error: [[tuya:628](https://github.com/esphome/esphome/blob/dd47d063b50d3df6e751cce473082f4fc8738802/esphome/components/tuya/tuya.cpp#L628)]	`Attempt to set datapoint 1 with incorrect type` 

Another weird thing is that when turning the fan on or off, it updates most of other datapoints too (both oscillations, mode, ...). It turns on the oscillations and switches mode when the fan is turned on, and disables them when it is turned off.

I downgraded from 2025.5.0 back to 2025.4.2 and the fan works flawlessly again with exactly the same yaml config.

### Which version of ESPHome has the issue?

2025.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.2

### What platform are you using?

ESP32-IDF

### Board

M5nano-c6

### Component causing the issue

tuya

### YAML Config

```yaml
``
#
# https://devices.esphome.io/devices/Duux-Whisper-Flex-Smart-Fan-DXCF10
# https://templates.blakadder.com/duux_DXCF10.html
# https://community.home-assistant.io/t/experience-integrating-duux-products/386403/107
# https://docs.m5stack.com/en/core/M5NanoC6
#

esphome:
  name: "duux-whisper-flex-fan"
  friendly_name: Duux Whisper Flex Fan

##
## Connection to the fan:
##
## Fan TX ---yellow--> GPIO2
## Fan RX <---white--- GPIO1
##

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 4MB
  framework:
    type: esp-idf
    #version: 5.3.1
    version: 5.3.1
    #version: latest
    platform_version: 6.9.0 # Need at least 6.4 for ESP32-C6
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

# Enable logging
logger:

[... 280 lines cut ...]
[15:23:52][D][select:015]: 'Oscillate horizontally': Sending state off (index 0)
[15:23:52][D][uart_debug:114]: <<< 55:AA:03:07:00:05:04:04:00:01:00:17
[15:23:52][D][tuya:335]: Datapoint 1 update to ON
[15:23:52][D][fan:120]: 'Duux Whisper Flex Fan' - Sending state:
[15:23:52][D][fan:121]:   State: ON
[15:23:52][D][fan:123]:   Speed: 2
[15:23:52][D][tuya:355]: Datapoint 2 update to 2
[15:23:52][D][select:015]: 'Mode': Sending state Sleep mode (index 2)
[15:23:52][D][tuya:355]: Datapoint 3 update to 1
[15:23:52][D][fan:120]: 'Duux Whisper Flex Fan' - Sending state:
[15:23:52][D][fan:121]:   State: ON
[15:23:52][D][fan:123]:   Speed: 2
[15:23:52][D][tuya:355]: Datapoint 4 update to 0
[15:23:52][D][select:015]: 'Oscillate horizontally': Sending state off (index 0)
[15:23:52][D][tuya:355]: Datapoint 5 update to 0
[15:23:52][D][select:015]: 'Oscillate vertically': Sending state off (index 0)
[15:23:52][D][tuya:355]: Datapoint 6 update to 0
[15:23:52][D][select:015]: 'Timer': Sending state off (index 0)
[15:23:52][D][uart_debug:114]: <<< 55:AA:03:07:00:05:01:01:00:01:01:12:55:AA:03:07:00:05:02:04:00:01:02:17:55:AA:03:07:00:05:03:04:00:01:01:17:55:AA:03:07:00:05:04:04:00:01:00:17:55:AA:03:07:00:05:05:04:00:01:00:18:55:AA:03:07:00:05:06:04:00:01:00:19
[15:23:53][D][sensor:093]: 'Wifi Signal': Sending state -62.00000 dBm with 0 decimals of accuracy
[15:23:59][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[15:23:59][D][uart_debug:114]: <<< 55:AA:03:00:00:01:01:04
``
``
```

randybb, 2025-05-21:
Maybe related to this change https://github.com/esphome/esphome/pull/8393

jkufner, 2025-05-21:
> Maybe related to this change [esphome/esphome#8393](https://github.com/esphome/esphome/pull/8393)

I tried to use int_datapoint instead of enum_datapoint. It did not help.

Cossid, 2025-05-21:
I don't think it's related to https://github.com/esphome/esphome/pull/8393, the datapoint with the error is datapoint 1 which is part of the fan entity, none of the selects use that datapoint, and none of the underlying datapoint code was changed in that PR, select was just modified to use already existing datapoint code.

That being said, no code changed in the underlying Tuya Fan or TuyaDatapoint code that I can see either.

@jkufner can you post the tuya config dump from the boot log?

Cossid, 2025-05-21:
I think I've spotted the bug, it stemmed from a suggestion having a format error, and I corrected it the wrong way it appears.  I'll post a PR soon.

jkufner, 2025-05-21:
> I think I've spotted the bug, it stemmed from a suggestion having a format error, and I corrected it the wrong way it appears. I'll post a PR soon.

I've added the parenthesis directly into my venv install and it fixed the issue. Thank you!

Fun fact: Many linters require the developer to provide these parenthesis around assignments explicitly to avoid such bugs :)

Cossid, 2025-05-21:
> Fun fact: Many linters require the developer to provide these parenthesis around assignments explicitly to avoid such bugs :)

Python isn't my native programming language and I'm still learning bits and pieces, and a code suggestion was recommended, but it had a typo in it, and my unfamiliarity lead me to correct it the wrong way.  After re-reading it's more obvious, but still a pretty strange one.  At least it was an easy fix in the end.

closed 2025-05-22 by jesserockz

plyatov, 2025-06-01:
I have encountered same issue.
Can you please be more detailed on how to fix this issue exactly?

plyatov, 2025-06-01:
> I have encountered same issue. Can you please be more detailed on how to fix this issue exactly?

Sorry, just found it at https://github.com/esphome/esphome/pull/8871/files.

Cossid, 2025-06-02:
> I have encountered same issue. Can you please be more detailed on how to fix this issue exactly?

The patch is already included in the released version 2025.5.1