esphome/issues #7039: 1-wire no more dallas sensor found

iddiek, 2025-05-21:
### The problem

with the latest update 2025.5.0 the one-wire module no longer detects my dallas sensors. I tried deleting the sensor and rescanning but it found nothing. Restored 2025.4.1 and all OK.

### Which version of ESPHome has the issue?

2025.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.2

### What platform are you using?

ESP8266

### Board

d1 mini

### Component causing the issue

1-wire + dallas

### YAML Config

```yaml
substitutions:
  devicename: termostato cameretta
  nodename: "termostato-cameretta"
  device_description: Termostato in cameretta

esphome:
  name: $nodename
  comment: ${device_description}    

esp8266:
  board: d1_mini
wifi:
  ssid: "XXXXXX"
  password: "XXXXXXXXX"
  ap:
    ssid: "$devicename Fallback"
    password: "XXXXXXXXXXX"
  fast_connect: true
captive_portal:
api:
ota:
  - platform: esphome
    password: "XXXXXXXXXXX"

time:
  - platform: homeassistant
    id: homeassistant_time

web_server:
  port: 80

logger: 
  on_message:
    level: ERROR
    then:
      - light.turn_on:
          id: termled          
          effect: strobe
one_wire:
  - platform: gpio
    pin: D2  

sensor:
  - platform: wifi_signal
    name: "wifi $devicename"
    update_interval: 60s
#  - platform: dallas
#    name: "DALLASTemperature"
#    address: 0xE73C01B556029D28
#    id: temp
#  - platform: dallas
#    address: 0xEF3C01B556970928    
#    name: "DALLASTemperature2"
#    id: temp2  
#  - platform: dallas
#    address: 0x763C01B5566BE828    
#    name: "DALLASTemperature3"
#    id: temp3
  - platform: dallas_temp
    address: 0x4E3C01B556FDE628
    name: "$devicename"
    id: temp       
    update_interval: 10s

#  - platform: dallas
#    address: 0xE73C01B556029D28
#    name: "DALLASTemperature5"
#    id: temp5        
#  - platform: dallas
#    address: 0xAE3CFFB5566B3B28
#    name: "DALLASTemperature6"
#    id: temp6 
#  - platform: dallas
#    address: 0xD23C01B556B47728
#    name: "DALLASTemperature7"
#    id: temp7        
# General INFO
#  - platform: uptime
#    name: Termostato Bagnetto Uptime

binary_sensor:
  - platform: homeassistant
    id: finestra
    entity_id: binary_sensor.finestre_cameretta
    publish_initial_state: True
    on_state:
      then:
        - delay: 15s
        - if:
            condition:
                  binary_sensor.is_on: finestra
            then:
              - climate.control:
                  id: thermo
                  mode: "off"
            else:
              - delay: 180s
              - climate.control:
                  id: thermo
                  mode: heat

switch: 
  - platform: gpio
    pin: D1
    id: relayD1
    name: "Risc $devicename"
    inverted: false
    icon: "mdi:radiator"
button:
  - platform: restart
    name: "Restart"
    entity_category: config 
    icon: "mdi:restart-alert"
    id: rest_sw

light:
  - platform: neopixelbus
    id: termled
#    name: termled $devicename
    variant: WS2812x
    pin: D7
    num_leds: 1
    type: grb
    effects:
      - strobe:
      - strobe:
          name: idle
          colors:
            - state: True
              brightness: 50%
              duration: 250ms
              green: 1
              red: 0
              blue: 0
            - state: False
              duration: 3s
      - strobe:
          name: all_off
          colors:
            - state: True
              brightness: 60%
              duration: 3s
              green: 0
              red: 0
              blue: 1
            - state: False
              duration: 500ms
      - addressable_color_wipe:  
          name: color_wipe
      - flicker:
      
climate:
  - platform: thermostat
    id: thermo
    name: "$devicename"
    sensor: temp

    startup_delay: true  
    default_preset: inverno
    on_boot_restore_from: default_preset
    preset: 
      - name: inverno
        mode: heat
        default_target_temperature_low: 21.0

    min_heating_off_time: 300s
    min_heating_run_time: 300s
    min_idle_time: 300s
    heat_deadband: 0.1
    heat_overrun: 0.1
    
    heat_action:
      - if:
          condition:
            not:
              binary_sensor.is_on: finestra
          then:
            - switch.turn_on: relayD1
            - light.turn_on: 
                id: termled
                red: 1
                green: 0
                blue: 0
                effect: flicker

    idle_action:
      - switch.turn_off: relayD1
      - light.turn_on:
          id: termled
          effect: idle

    off_mode:
      - light.turn_on:
          id: termled
          effect: all_off
    
    heat_mode:
      - if:
          condition: 
            switch.is_off: relayD1
          then:
            - light.turn_on:
                id: termled
                effect: idle
          else:
            - light.turn_on: 
                id: termled
                red: 1
                green: 0
                blue: 0
                effect: flicker
```

### Anything in the logs that might be useful for us?

```txt
scratch pad: ff.ff.ff.ff.ff.ff.ff.ff.ff (c9)
```

DaHype, 2025-05-21:
Same Issue here:

[19:17:24][C][wifi:600]: WiFi:
[19:17:24][C][wifi:428]:   Local MAC: xxxxx
[19:17:24][C][wifi:433]:   SSID: [redacted]
[19:17:24][C][wifi:436]:   IP Address: 192.168.1.169
[19:17:24][C][wifi:439]:   BSSID: [redacted]
[19:17:24][C][wifi:441]:   Hostname: 'speichertemperatur'
[19:17:24][C][wifi:443]:   Signal strength: -57 dB ▂▄▆█
[19:17:24][C][wifi:447]:   Channel: 1
[19:17:24][C][wifi:448]:   Subnet: 255.255.255.0
[19:17:24][C][wifi:449]:   Gateway: 192.168.1.1
[19:17:24][C][wifi:450]:   DNS1: 192.168.1.1
[19:17:24][C][wifi:451]:   DNS2: 0.0.0.0
[19:17:24][C][logger:224]: Logger:
[19:17:24][C][logger:225]:   Max Level: DEBUG
[19:17:24][C][logger:226]:   Initial Level: DEBUG
[19:17:24][C][logger:228]:   Log Baud Rate: 115200
[19:17:24][C][logger:229]:   Hardware UART: UART0
[19:17:24][C][gpio.one_wire:022]: GPIO 1-wire bus:
[19:17:24][C][gpio.one_wire:023]:   Pin: GPIO2
[19:17:24][W][gpio.one_wire:082]:   Found no devices!
[19:17:24][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[19:17:24][C][dallas.temp.sensor:034]:   Address: 0x8e3c56e381501528 (DS18B20)
[19:17:24][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[19:17:24][C][dallas.temp.sensor:036]:   Update Interval: 60.0s
[19:17:24][C][captive_portal:089]: Captive Portal:
[19:17:26][C][mdns:120]: mDNS:
[19:17:26][C][mdns:121]:   Hostname: speichertemperatur
[19:17:26][C][esphome.ota:073]: Over-The-Air updates:
[19:17:26][C][esphome.ota:074]:   Address: 192.168.1.62:8266
[19:17:26][C][esphome.ota:075]:   Version: 2
[19:17:26][C][esphome.ota:078]:   Password configured
[19:17:26][C][safe_mode:018]: Safe Mode:
[19:17:26][C][safe_mode:019]:   Boot considered successful after 60 seconds
[19:17:26][C][safe_mode:021]:   Invoke after 10 boot attempts
[19:17:26][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[19:17:26][C][api:170]: API Server:
[19:17:26][C][api:171]:   Address: 192.168.1.62:6053
[19:17:26][C][api:173]:   Using noise encryption: YES
[19:18:07][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[19:18:19][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[19:18:20][W][dallas.temp.sensor:128]: 'SpeicherTemp' - Scratch pad checksum invalid!
[19:18:20][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[19:18:20][D][dallas.temp.sensor:130]: Scratch pad: FD.FF.FF.FF.FF.FF.FF.FF.FF (4F)
[19:18:20][D][sensor:093]: 'SpeicherTemp': Sending state nan °C with 2 decimals of accuracy

Lstick, 2025-05-21:
Same problem.
After update today's I have error:
`[20:36:18][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[20:36:18][C][dallas.temp.sensor:034]:   Address: 0x290000055XXXXXXX (DS18B20)
[20:36:18][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[20:36:18][C][dallas.temp.sensor:036]:   Update Interval: 60.0s
...
[20:39:54][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[20:39:55][E][one_wire:023]: 1-wire bus is held low
[20:39:55][W][dallas.temp.sensor:066]: 'Bathroom water temperature' - reading scratch pad failed bus reset
[20:39:55][W][component:167]: Component dallas_temp.sensor set Warning flag: bus reset failed`

No problem with the previous version so far.

ssieb, 2025-05-21:
Are you using parasitic power mode?

iddiek, 2025-05-21:
> Are you using parasitic power mode?
No

Lstick, 2025-05-21:
> Are you using parasitic power mode?

As far as I remember, yes, Two wires at all.

Denis198806, 2025-05-21:
Same Issue here
INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/epvs-350.yaml...
INFO Starting log output from 192.168.5.58 using esphome API
INFO Successfully connected to epvs-350 @ 192.168.5.58 in 0.022s
INFO Successful handshake with epvs-350 @ 192.168.5.58 in 2.639s
[00:38:27][I][app:115]: ESPHome version 2025.5.0 compiled on May 22 2025, 00:23:42
[00:38:27][C][wifi:600]: WiFi:
[00:38:27][C][wifi:428]:   Local MAC: F4:CF:A2:ED:8B:5A
[00:38:27][C][wifi:433]:   SSID: 'Pelihds_IOT'[redacted]
[00:38:27][C][wifi:436]:   IP Address: 192.168.5.58
[00:38:27][C][wifi:439]:   BSSID: 64:D1:54:17:35:8F[redacted]
[00:38:27][C][wifi:441]:   Hostname: 'epvs-350'
[00:38:27][C][wifi:443]:   Signal strength: -35 dB ▂▄▆█
[00:38:27][V][wifi:445]:   Priority: 0.0
[00:38:27][C][wifi:447]:   Channel: 6
[00:38:27][C][wifi:448]:   Subnet: 255.255.255.0
[00:38:27][C][wifi:449]:   Gateway: 192.168.5.1
[00:38:27][C][wifi:450]:   DNS1: 192.168.5.1
[00:38:27][C][wifi:451]:   DNS2: 8.8.8.8
[00:38:27][C][logger:224]: Logger:
[00:38:27][C][logger:225]:   Max Level: VERY_VERBOSE
[00:38:27][C][logger:226]:   Initial Level: VERY_VERBOSE
[00:38:27][C][logger:228]:   Log Baud Rate: 115200
[00:38:27][C][logger:229]:   Hardware UART: UART0
[00:38:27][C][gpio.one_wire:022]: GPIO 1-wire bus:
[00:38:27][C][gpio.one_wire:023]:   Pin: GPIO5
[00:38:27][W][gpio.one_wire:082]:   Found no devices!
[00:38:27][C][switch.gpio:068]: GPIO Switch 'Epvs-350 High Relay'
[00:38:27][C][switch.gpio:070]:   Icon: 'mdi:fan-speed-2'
[00:38:27][C][switch.gpio:090]:   Restore Mode: always OFF
[00:38:27][C][switch.gpio:031]:   Pin: GPIO14
[00:38:27][C][switch.gpio:033]:   Interlocks:
[00:38:27][C][switch.gpio:037]:     Epvs-350 Low Relay
[00:38:27][C][switch.gpio:068]: GPIO Switch 'Epvs-350 Low Relay'
[00:38:27][C][switch.gpio:070]:   Icon: 'mdi:fan-speed-1'
[00:38:27][C][switch.gpio:090]:   Restore Mode: always OFF
[00:38:27][C][switch.gpio:031]:   Pin: GPIO12
[00:38:27][C][switch.gpio:033]:   Interlocks:
[00:38:27][C][switch.gpio:037]:     Epvs-350 High Relay
[00:38:27][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[00:38:27][C][switch.gpio:068]: GPIO Switch 'Epvs-350 Heat'
[00:38:27][C][switch.gpio:090]:   Restore Mode: always OFF
[00:38:27][C][switch.gpio:031]:   Pin: GPIO13
[00:38:27][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:38:27][C][dallas.temp.sensor:034]:   Address: 0x830000006ab11a28 (DS18B20)
[00:38:27][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:38:27][C][dallas.temp.sensor:036]:   Update Interval: 5.0s
[00:38:27][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:38:27][C][dallas.temp.sensor:034]:   Address: 0x3500000069fa2728 (DS18B20)
[00:38:27][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:38:27][C][dallas.temp.sensor:036]:   Update Interval: 5.0s
[00:38:27][C][speed.fan:021]: Speed Fan 'Epvs-350 Speed Fan'
[00:38:27][C][speed.fan:192]:   Speed: YES
[00:38:27][C][speed.fan:193]:   Speed count: 2
[00:38:27][C][captive_portal:089]: Captive Portal:
[00:38:27][C][mdns:120]: mDNS:
[00:38:27][C][mdns:121]:   Hostname: epvs-350
[00:38:27][V][mdns:122]:   Services:
[00:38:27][V][mdns:124]:   - _esphomelib, _tcp, 6053
[00:38:27][V][mdns:127]:     TXT: friendly_name = EPVS-350
[00:38:27][V][mdns:127]:     TXT: version = 2025.5.0
[00:38:27][V][mdns:127]:     TXT: mac = f4cfa2ed8b5a
[00:38:27][V][mdns:127]:     TXT: platform = ESP8266
[00:38:27][V][mdns:127]:     TXT: board = nodemcuv2
[00:38:27][V][mdns:127]:     TXT: network = wifi
[00:38:27][V][mdns:127]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[00:38:27][C][esphome.ota:073]: Over-The-Air updates:
[00:38:27][C][esphome.ota:074]:   Address: epvs-350.local:8266
[00:38:27][C][esphome.ota:075]:   Version: 2
[... 51 lines cut ...]
[00:38:34][VV][sensor.filter:014]: Filter(0x3fff383c)::input(nan)
[00:38:34][VV][sensor.filter:021]: Filter(0x3fff383c)::output(nan) -> SENSOR
[00:38:34][D][sensor:093]: 'in-air': Sending state nan °C with 1 decimals of accuracy
[00:38:34][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 680144295
  state: nan
  missing_state: NO
}
[00:38:37][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[00:38:38][VV][dallas.temp.sensor:122]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
[00:38:38][W][dallas.temp.sensor:128]: 'out-air' - Scratch pad checksum invalid!
[00:38:38][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[00:38:38][D][dallas.temp.sensor:130]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
[00:38:38][V][sensor:043]: 'out-air': Received new state nan
[00:38:38][D][sensor:093]: 'out-air': Sending state nan °C with 1 decimals of accuracy
[00:38:38][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 3459060448
  state: nan
  missing_state: NO
}
[00:38:38][W][component:182]: Component dallas_temp.sensor cleared Warning flag

The problem appeared after the update, as seen on the graph:

[image]

ssieb, 2025-05-21:
All of these seem to be 8266 devices.
Using parasitic power seems to give `[E][one_wire:023]: 1-wire bus is held low`.
Normal power seems to not trigger the sensor, so you get `][D][dallas.temp.sensor:130]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)`, although there's a `[D][dallas.temp.sensor:130]: Scratch pad: FD.FF.FF.FF.FF.FF.FF.FF.FF (4F)` from one person which is interesting.
If anyone that's not using parasitic mode has a logic analyzer, please contact me on discord.
See if this resolves the issue:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components: [ one_wire, gpio, dallas_temp ]
```

Denis198806, 2025-05-22:
> Все эти устройства, по-видимому, относятся к 8266-й серии. Использование паразитной мощности, по-видимому, дает `[E][one_wire:023]: 1-wire bus is held low`. Обычная мощность, по-видимому, не активирует датчик, поэтому вы получаете `][D][dallas.temp.sensor:130]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)`, хотя есть `[D][dallas.temp.sensor:130]: Scratch pad: FD.FF.FF.FF.FF.FF.FF.FF.FF (4F)` от одного человека, что интересно. Если у вас нет логического анализатора, работающего в паразитарном режиме, пожалуйста, свяжитесь со мной в Discord. Это решает проблему:
> 
> внешние_компоненты:
>  - источник:
>  тип: git
>       url: https://github.com/esphome/esphome
>       ref: 2025.4.2
>     компоненты: [ one_wire, gpio, dallas_temp ]

Unfortunately this did not solve the problem.

<details><summary>Here is my full code:</summary>
<p>

Maybe I did it wrong?

```c++
esphome:
  name: epvs-350
  friendly_name: EPVS-350

external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components: [ one_wire, gpio, dallas_temp ]

esp8266:
  board: nodemcuv2
  framework:
    version: recommended
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "***********************************"

ota:
  - platform: esphome
    password: "*****************************"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: $devicename
    password: !secret ota_password
captive_portal:

# Example configuration entry
one_wire:
  - platform: gpio
    pin: GPIO5

sensor:
  - platform: dallas_temp
    address: 0x830000006ab11a28
    name: in-air
    update_interval: 5s
    filters:
     - offset: -0.4

  - platform: dallas_temp
    address: 0x3500000069fa2728
    name: out-air
    update_interval: 5s

[... 153 lines cut ...]
[08:36:17][C][esphome.ota:074]:   Address: epvs-350.local:8266
[08:36:17][C][esphome.ota:075]:   Version: 2
[08:36:17][C][esphome.ota:078]:   Password configured
[08:36:17][C][safe_mode:018]: Safe Mode:
[08:36:17][C][safe_mode:019]:   Boot considered successful after 60 seconds
[08:36:17][C][safe_mode:021]:   Invoke after 10 boot attempts
[08:36:17][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[08:36:17][C][api:170]: API Server:
[08:36:17][C][api:171]:   Address: epvs-350.local:6053
[08:36:17][C][api:173]:   Using noise encryption: YES
[08:36:20][D][api:122]: Accepted 192.168.5.150
[08:36:20][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[08:36:20][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[08:36:21][W][dallas.temp.sensor:139]: 'out-air' - Scratch pad checksum invalid!
[08:36:21][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[08:36:21][D][sensor:093]: 'out-air': Sending state nan °C with 1 decimals of accuracy
[08:36:21][D][api.connection:1529]: Home Assistant 2025.5.2 (192.168.5.150): Connected successfully
[08:36:21][W][dallas.temp.sensor:139]: 'in-air' - Scratch pad checksum invalid!
[08:36:21][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[08:36:21][D][sensor:093]: 'in-air': Sending state nan °C with 1 decimals of accuracy

``` 

</p>
</details>

ssieb, 2025-05-22:
Ok, try adding `esp8266` to the components list in the external components section.

Denis198806, 2025-05-22:
> Ok, try adding `esp8266` to the components list in the external components section.

@ssieb  thank you very much! Everything works great.
Here is what happened:

```c++
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components: [ one_wire, gpio, dallas_temp, esp8266 ]
``` 

<details><summary>Here is the log:</summary>
<p>

```c++
INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/epvs-350.yaml...
INFO Starting log output from 192.168.5.58 using esphome API
INFO Successfully connected to epvs-350 @ 192.168.5.58 in 0.003s
INFO Successful handshake with epvs-350 @ 192.168.5.58 in 2.397s
[08:58:34][I][app:115]: ESPHome version 2025.5.0 compiled on May 22 2025, 08:53:18
[08:58:34][C][wifi:600]: WiFi:
[08:58:34][C][wifi:428]:   Local MAC: F4:CF:A2:ED:8B:5A
[08:58:34][C][wifi:433]:   SSID: 'Pelihds_IOT'[redacted]
[08:58:34][C][wifi:436]:   IP Address: 192.168.5.58
[08:58:34][C][wifi:439]:   BSSID: 64:D1:54:17:35:8F[redacted]
[08:58:34][C][wifi:441]:   Hostname: 'epvs-350'
[08:58:34][C][wifi:443]:   Signal strength: -36 dB ▂▄▆█
[08:58:34][V][wifi:445]:   Priority: 0.0
[08:58:34][C][wifi:447]:   Channel: 6
[08:58:34][C][wifi:448]:   Subnet: 255.255.255.0
[08:58:34][C][wifi:449]:   Gateway: 192.168.5.1
[08:58:34][C][wifi:450]:   DNS1: 192.168.5.1
[08:58:34][C][wifi:451]:   DNS2: 8.8.8.8
[08:58:34][C][logger:224]: Logger:
[08:58:34][C][logger:225]:   Max Level: VERBOSE
[08:58:34][C][logger:226]:   Initial Level: VERBOSE
[08:58:34][C][logger:228]:   Log Baud Rate: 115200
[08:58:34][C][logger:229]:   Hardware UART: UART0
[08:58:34][C][gpio.one_wire:020]: GPIO 1-wire bus:
[08:58:34][C][gpio.one_wire:021]:   Pin: GPIO5
[08:58:34][C][gpio.one_wire:080]:   Found devices:
[08:58:34][C][gpio.one_wire:082]:     0x830000006ab11a28 (DS18B20)
[08:58:34][C][gpio.one_wire:082]:     0x3500000069fa2728 (DS18B20)
[08:58:34][C][switch.gpio:068]: GPIO Switch 'Epvs-350 High Relay'
[08:58:34][C][switch.gpio:070]:   Icon: 'mdi:fan-speed-2'
[08:58:34][C][switch.gpio:090]:   Restore Mode: always OFF
[08:58:34][C][switch.gpio:031]:   Pin: GPIO14
[08:58:34][C][switch.gpio:033]:   Interlocks:
[08:58:34][C][switch.gpio:037]:     Epvs-350 Low Relay
[08:58:34][C][switch.gpio:068]: GPIO Switch 'Epvs-350 Low Relay'
[08:58:34][C][switch.gpio:070]:   Icon: 'mdi:fan-speed-1'
[08:58:34][C][switch.gpio:090]:   Restore Mode: always OFF
[08:58:34][C][switch.gpio:031]:   Pin: GPIO12
[08:58:34][C][switch.gpio:033]:   Interlocks:
[08:58:34][C][switch.gpio:037]:     Epvs-350 High Relay
[08:58:34][C][switch.gpio:068]: GPIO Switch 'Epvs-350 Heat'
[08:58:34][C][switch.gpio:090]:   Restore Mode: always OFF
[08:58:34][C][switch.gpio:031]:   Pin: GPIO13
[08:58:34][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[08:58:34][C][dallas.temp.sensor:034]:   Address: 0x830000006ab11a28 (DS18B20)
[08:58:34][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[08:58:34][C][dallas.temp.sensor:036]:   Update Interval: 5.0s
[08:58:34][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[08:58:34][C][dallas.temp.sensor:034]:   Address: 0x3500000069fa2728 (DS18B20)
[08:58:34][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[08:58:34][C][dallas.temp.sensor:036]:   Update Interval: 5.0s
[08:58:34][C][speed.fan:021]: Speed Fan 'Epvs-350 Speed Fan'
[08:58:34][C][speed.fan:192]:   Speed: YES
[08:58:34][C][speed.fan:193]:   Speed count: 2
[08:58:34][C][captive_portal:089]: Captive Portal:
[08:58:34][C][mdns:120]: mDNS:
[08:58:34][C][mdns:121]:   Hostname: epvs-350
[08:58:34][V][mdns:122]:   Services:
[08:58:34][V][mdns:124]:   - _esphomelib, _tcp, 6053
[08:58:34][V][mdns:127]:     TXT: friendly_name = EPVS-350
[08:58:34][V][mdns:127]:     TXT: version = 2025.5.0
[08:58:34][V][mdns:127]:     TXT: mac = f4cfa2ed8b5a
[08:58:34][V][mdns:127]:     TXT: platform = ESP8266
[08:58:34][V][mdns:127]:     TXT: board = nodemcuv2
[08:58:34][V][mdns:127]:     TXT: network = wifi
[08:58:34][V][mdns:127]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[08:58:35][C][esphome.ota:073]: Over-The-Air updates:
[08:58:35][C][esphome.ota:074]:   Address: epvs-350.local:8266
[08:58:35][C][esphome.ota:075]:   Version: 2
[08:58:35][C][esphome.ota:078]:   Password configured
[08:58:35][C][safe_mode:018]: Safe Mode:
[08:58:35][C][safe_mode:019]:   Boot considered successful after 60 seconds
[08:58:35][C][safe_mode:021]:   Invoke after 10 boot attempts
[08:58:35][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[08:58:35][C][api:170]: API Server:
[08:58:35][C][api:171]:   Address: epvs-350.local:6053
[08:58:35][C][api:173]:   Using noise encryption: YES
[08:58:37][D][dallas.temp.sensor:054]: 'in-air': Got Temperature=26.4°C
[08:58:37][V][sensor:043]: 'in-air': Received new state 26.437500
[08:58:37][D][sensor:093]: 'in-air': Sending state 26.03750 °C with 1 decimals of accuracy
[08:58:38][D][dallas.temp.sensor:054]: 'out-air': Got Temperature=25.2°C
[08:58:38][V][sensor:043]: 'out-air': Received new state 25.187500
[08:58:38][D][sensor:093]: 'out-air': Sending state 25.18750 °C with 1 decimals of accuracy
[08:58:42][D][dallas.temp.sensor:054]: 'in-air': Got Temperature=26.4°C
[08:58:42][V][sensor:043]: 'in-air': Received new state 26.437500
[08:58:42][D][sensor:093]: 'in-air': Sending state 26.03750 °C with 1 decimals of accuracy

``` 

</p>
</details>

ssieb, 2025-05-22:
That suggests it's an issue with the new low-level gpio code.  Can you try the new code with other pins and see if it's specific to certain pins?

Denis198806, 2025-05-22:
> That suggests it's an issue with the new low-level gpio code. Can you try the new code with other pins and see if it's specific to certain pins?

Yes, I can try to transfer the pins of the 1-wire bus in the evening. Once I do, I will write about the results.

ssieb, 2025-05-22:
@DaHype when was it last working?  That setting will revert all the relevant components back to the last release.
`[D][dallas.temp.sensor:130]: Scratch pad: FD.FF.FF.FF.FF.FF.FF.FF.FF (4F)` suggests a different issue.  That `FD` should be `FF`.

DaHype, 2025-05-22:
@ssieb 
Working till ESPHome Version 2025.4.2 without Problems.
So i will downgrade now, and hope it will be fixed in future Versions.
Thanks

bethgocs, 2025-05-22:
same issue

iddiek, 2025-05-22:
> Ok, try adding `esp8266` to the components list in the external components section.

It's the only one needed to get it working again

Lstick, 2025-05-22:
Confirm adding `esp8266 ` solved the issue in my case.

ssieb, 2025-05-23:
@iddiek do you mean that only using the `esp8266` component from the previous version fixed it?  You didn't even need to rollback the 1-wire parts?

iddiek, 2025-05-23:
> [@iddiek](https://github.com/iddiek) do you mean that only using the `esp8266` component from the previous version fixed it? You didn't even need to rollback the 1-wire parts?

Exactly, just list esp8266 component in external_components ref 2025.4.2

fluffy123blau, 2025-05-23:
Hi,
Various ESP8266 modules are in use here. The problem only occurs with the Wemos D1_mini. With the Nodemcuv2, it works as expected. Both with 1-Wire on D1 (GPIO5).
For the D1_mini, I use the recommended code (only with the ESP8266 specification in the components section):

```c++
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components: [ esp8266 ]
``` 

Thank you for that helpful hint

ssieb, 2025-05-23:
> Various ESP8266 modules are in use here. The problem only occurs with the Wemos D1_mini. With the Nodemcuv2, it works as expected. Both with 1-Wire on D1 (GPIO5).

Ok, that's interesting.  I don't think I have a d1 mini around.  I wonder why it would be different like that.

> I have a `esp01_1m` board and parasitic `ds18b20` with the same issue. the `external_components:` solved it, just need to have it before the `esp8266:'

The order doesn't matter.

tikismoke, 2025-05-23:
Got the same issue, but with a lot of other error scratch_pad value:
```
[21:44:14][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[21:44:15][W][dallas.temp.sensor:128]: 'rdc_onewire_temp_temperature' - Scratch pad checksum invalid!
[21:44:15][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[21:44:15][D][dallas.temp.sensor:130]: Scratch pad: 79.01.02.44.7E.FF.07.10.08 (E5)
[21:44:15][D][sensor:093]: 'rdc_onewire_temp_temperature': Sending state nan °C with 1 decimals of accuracy
[21:44:16][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[21:44:17][W][dallas.temp.sensor:128]: 'couloir_1er_onewire_temp_temperature' - Scratch pad checksum invalid!
[21:44:17][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[21:44:17][D][dallas.temp.sensor:130]: Scratch pad: 34.01.4B.46.7F.FD.08.10.45 (68)
[21:44:17][D][sensor:093]: 'couloir_1er_onewire_temp_temperature': Sending state nan °C with 1 decimals of accuracy
[21:44:19][W][dallas.temp.sensor:128]: 'bureau_onewire_temp_temperature' - Scratch pad checksum invalid!
[21:44:19][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[21:44:19][D][dallas.temp.sensor:130]: Scratch pad: 76.81.05.A2.2D.5B.05.88.BC (31)
[21:44:19][D][sensor:093]: 'bureau_onewire_temp_temperature': Sending state nan °C with 1 decimals of accuracy
[21:44:20][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[21:44:21][W][dallas.temp.sensor:128]: 'sdb_1er_onewire_temp_temperature' - Scratch pad checksum invalid!
[21:44:21][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[21:44:21][D][dallas.temp.sensor:130]: Scratch pad: ED.FF.FF.FF.FF.FF.FF.DF.6D (38)
[21:44:21][D][sensor:093]: 'sdb_1er_onewire_temp_temperature': Sending state nan °C with 1 decimals of accuracy
[21:44:22][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[21:44:23][W][dallas.temp.sensor:128]: 'parents_onewire_temp_temperature' - Scratch pad checksum invalid!
[21:44:23][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[21:44:23][D][dallas.temp.sensor:130]: Scratch pad: BF.E9.B6.6D.BE.FF.FF.FF.FF (4D)
[21:44:23][D][sensor:093]: 'parents_onewire_temp_temperature': Sending state nan °C with 1 decimals of accuracy
[21:44:24][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[21:44:25][W][dallas.temp.sensor:128]: 'tobias_onewire_temp_temperature' - Scratch pad checksum invalid!
[21:44:25][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[21:44:25][D][dallas.temp.sensor:130]: Scratch pad: 6D.DB.FF.FF.FF.FF.FF.FF.FF (92)
[21:44:25][D][sensor:093]: 'tobias_onewire_temp_temperature': Sending state nan °C with 1 decimals of accuracy
[21:44:26][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[21:44:27][W][dallas.temp.sensor:128]: 'couloir_2eme_onewire_temp_temperature' - Scratch pad checksum invalid!
[21:44:27][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[21:44:27][D][dallas.temp.sensor:130]: Scratch pad: BE.FF.FF.DF.7F.FF.F7.ED.DB (B5)
[21:44:27][D][sensor:093]: 'couloir_2eme_onewire_temp_temperature': Sending state nan °C with 1 decimals of accuracy
[21:44:28][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[21:44:29][W][dallas.temp.sensor:128]: 'sdb_2eme_onewire_temp_temperature' - Scratch pad checksum invalid!
[21:44:29][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[21:44:29][D][dallas.temp.sensor:130]: Scratch pad: FF.FF.FF.FF.FF.FF.B6.6D.BF (4D)
[21:44:29][D][sensor:093]: 'sdb_2eme_onewire_temp_temperature': Sending state nan °C with 1 decimals of accuracy
[21:44:30][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[21:44:31][W][dallas.temp.sensor:128]: 'noelyne_onewire_temp_temperature' - Scratch pad checksum invalid!
[21:44:31][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
```

lokomoti, 2025-05-24:
Same issue on Pi Pico W.

`substitutions:
  device_name: water-heating

esphome:
  name: ${device_name}
  friendly_name: Water Heating

rp2040:
  board: rpipicow
  framework:
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  platform: esphome
  password:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Water-Heating Fallback Hotspot"
    password: 

one_wire:
  - platform: gpio
    pin: 3

# i2c:
#   sda: 4
#   scl: 5
#   scan: true
#   id: bus_a
#   frequency: 400kHz

switch:
  - platform: gpio
    name: "Relay 1"
    pin: 
      number: 14
      inverted: true
    
  - platform: gpio
    name: "Solar Circulation Pump"
    pin: 
      number: 15
      inverted: true
    id: solar_circulation_pump

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s
    id: wifi_signal_db
    entity_category: "diagnostic"
    unit_of_measurement: "Signal %"
  
  - platform: dallas_temp
    address: 0x230000000d019428
    name: "Temperature Top"
    id: temp_top
    update_interval: 10s
  
  - platform: dallas_temp
    address: 0x0a0000000c57e528
    name: "Temperature Bottom"
    id: temp_bot
    update_interval: 10s

  - platform: dallas_temp
    address: 0x2a0000000d236428
    name: "Medium Temperature Leaving"
    update_interval: 10s
    

binary_sensor:
  - platform: gpio
    name: "Input 1"
    pin:
      number: 6
      mode:
        input: true
        pullup: true
      inverted: true
  
  - platform: gpio
    name: "Input 2"
    pin:
      number: 7
      mode:
        input: true
        pullup: true
      inverted: true
  
  - platform: gpio
    name: "Input 3"
    pin:
      number: 8
      mode:
        input: true
        pullup: true
      inverted: true
  
  - platform: status
    name: "${device_name} - Status"
    device_class: connectivity

`

lokomoti, 2025-05-24:
I have same issue on Pi Pico W.

Same log messages as above.

fluffy123blau, 2025-05-24:
I have added an external pull-up to the sensor's data pin (3.3V - 4.7k - GPIO5) to a module and can confirm that it works without the additional code.

prbernard, 2025-05-24:
> > I too have the identical problem and logging messages. It is not restricted to esp8266 I have it on a Pico W rp2040. I'm not using parasitic power and have two DS18B20 devices. It was working on 2025.4.2 but failed after an upgrade to 2025.5.0. The previously suggested workaround of:
> > ```
> > external_components:
> >   - source:
> >       type: git
> >       url: https://github.com/esphome/esphome
> >       ref: 2025.4.2
> >     components: [ one_wire, gpio, dallas_temp ]
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > solves the problem.
> 
> interesting. At this point the problem probably occurs for sensors that are not connected with a pull-up resistor, as suggested before. probably some modules in use have it integrated, others do not. I would like to know if you also solve the problem on the pico W with 4.7k between vcc and data, removing the part related to the external component in the code.

I've double checked, neither of the DS18B20 modules have integrated pull-up resistors.  I had added a single 4.7k pull-up to the 1-wire buss per the DS documentation for using a powered buss.  The pull-up was present with both 2025.4.2 and the upgrade to 2025.5.0.  When running with the above listed "external_components" the following behaviour is observed:
 
* If I remove one of the DS18B20 from the buss the other keeps working.
* If I remove the pull-up resistor both devices stop working.
* If I remove one DS18B20 and the pull-up resistor the other keeps working.  I'm assuming because of it being passive powered.

After removing the "external_components" section and doing a "clean build files" and re-install. Only one device is found in the scan, but it generates the "scratch pad checksum invalid!" error.
 
* Removing the found device stops the related checksum error but the other device continues to generate the same.
* Removing the pull-up resistor and the non-found device caused the single device to be found in the scan but generates the error "reading scratch pad failed bus reset" and it doesn't work.  Each update generates this error rather than the "scratch pad checksum invalid!" 

Restoring the "external_components" section and doing a "clean build files" goes back to normal with both working as expected.

janchlebek, 2025-05-25:
I had the problem as well with only 1 sensor on GPIO5 (no pull-up resistor). 
Adding esp8266 from 2024.4.2 to my config works for now:

```
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components: [ esp8266 ]
```

ssieb, 2025-05-25:
An external pull-up resistor is highly recommended.  The internal pullup is very weak.

gitglint, 2025-05-27:
Hello,

I had the same problem that after updating to 2025.5.0 my DS18B20 sensors, which were connected to several D1-Mini (ESP8266), stopped working. Even after today's update to 2025.5.1, the errors remained. 

These sensors are connected (no external pull-up resistor) to the GPIO port D4, as well as to GND and +5V of the D1-Mini.

```
one_wire:
  - platform: gpio
    pin:
      number: D4
      mode:
        input: true
        pullup: true
```
On the ESP8266 with five Dallas sensors connected to GPIO pin D4, none of the sensors have been delivering data since the update.

For the ESP with three Dallas sensors connected to GPIO pin D4, two sensors provide correct data, while one sensor provides invalid data every few minutes (air temperature above 80 degrees Celsius instead of 20 degrees Celsius).

For ESPs with only one Dallas sensor connected to GPIO pin D4, no errors occur.

But after I added
```
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components: [ esp8266 ] 
```
into the configuration, all Dallas sensors are now reliably detected again and send valid data.

Maybe this information is helpful.

Soforak, 2025-05-27:
Same problem on Raspi Pico W, even after today's update  2025.5.1, no success.
Adding the
`external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components: [ one_wire, gpio, dallas_temp ]`
and cleaning the build cahce solved the problem

freijn, 2025-05-27:
@ssieb 

This worked for me !  Many thanks.   

esp8266 with parasitic power and no external pullup.

external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components: [ one_wire, gpio,esp8266, dallas_temp ]

peterle00370, 2025-05-27:
> > [@iddiek](https://github.com/iddiek) do you mean that only using the `esp8266` component from the previous version fixed it? You didn't even need to rollback the 1-wire parts?
> 
> Exactly, just list esp8266 component in external_components ref 2025.4.2

I had the same error on 4 of my 9 ESP8266s (nodemcuv2). All were connected via 3 pins, but not all with an external pull-up.
Listing the external component esp8266 solved the problem for me.

cosote, 2025-05-28:
> All of these seem to be 8266 devices. Using parasitic power seems to give `[E][one_wire:023]: 1-wire bus is held low`. Normal power seems to not trigger the sensor, so you get `][D][dallas.temp.sensor:130]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)`, although there's a `[D][dallas.temp.sensor:130]: Scratch pad: FD.FF.FF.FF.FF.FF.FF.FF.FF (4F)` from one person which is interesting. If anyone that's not using parasitic mode has a logic analyzer, please contact me on discord. See if this resolves the issue:
> 
> external_components:
>   - source:
>       type: git
>       url: https://github.com/esphome/esphome
>       ref: 2025.4.2
>     components: [ one_wire, gpio, dallas_temp ]

I'm running 11 DS18B20 on two pins on bk72xx: board: cb3s and had same issue on all sensors.
Forcing to use 2025.4.2 on components one_wire, gpio, dallas_temp fixed it also for me 🙏.
Thx a lot!

ssieb, 2025-05-28:
@cosote do you have an external pullup on the sensors?

rinyheuvelman, 2025-05-28:
Haven't tried the above mentioned solutions yet, but since last update I've noticed that there's more than just those issues with "too large", one-wire and "syntax error": Now EPSHome device builder maxes out the RPI's CPU when compiling, where as before it hasn't been any higher than 50%. Now it compiling slows down my entire Home assistant setup!

wymes, 2025-05-28:
adding the following  also worked for me also

external_components:

  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components: [ one_wire, gpio,esp8266, dallas_temp ]

ssieb, 2025-05-28:
@rinyheuvelman that's completely off-topic from this issue, but you can set the max compile process setting in the addon.  By default it will use all the cores and you need enough RAM.  A pi is really not powerful enough for compiling.

rinyheuvelman, 2025-05-28:
Thanks for the input @ssieb, although I don't agree: it's not completely off topic, it's all related to the latest esphome update, which (in my case) did more damage than good... The update generates all kinds of "misbehavior" on every ESP that's not a ESP32 derivative. :(  
The pi never failed me since I started using it years ago; neither did I have any problem compiling in esphome. Like I said: never used over 50% of the pi's CPU and always less than 20% of it's memory.

cosote, 2025-05-29:
> I've fixed rp2040, at least for non-parasitic. Either I don't understand how to set it up or all my sensors are fake and don't work, so I can't test that.
> 
> external_components:
>   - source:
>       type: git
>       url: https://github.com/ssieb/esphome
>       ref: onewire
>     components: [ one_wire, gpio, rp2040, dallas_temp ]

Quick update on my BK7231N 120MHz, 256KB RAM, 1.03MB Flash board: none of my 11 sensors are getting anything - logs attached. If there's more I can contribute, just let me know.

Cheers, cosote.
<details><summary>Logs testing https://github.com/ssieb/esphome with [ one_wire, gpio, rp2040, dallas_temp ] ... </summary>

  ```
  INFO ESPHome 2025.5.1
  INFO Reading configuration /config/esphome/4-433-interval.yaml...
  INFO Generating C++ source...
  INFO Compiling app...
  Processing 4-433 (board: cb3s; framework: arduino; platform: libretiny @ 1.7.0)
  --------------------------------------------------------------------------------
  Warning! Non-Git installations are NOT SUPPORTED.
  HARDWARE: BK7231N 120MHz, 256KB RAM, 1.03MB Flash
   - framework-arduino-api @ 2022.8.24+sha.237b10a 
   - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
   - library-flashdb @ 1.2.0+sha.d5c892f 
   - library-freertos @ 9.0.0+sha.95cc959 
   - library-freertos-port @ 2023.5.23+sha.a917d93 
   - library-lwip @ 2.2.0-bdk+sha.48da4a2 
   - library-printf @ 6.1.0+sha.28a79bd 
   - library-uf2ota @ 5.0.0+sha.f955412 
  PLATFORM VERSIONS:
   - libretiny @ 1.7.0
   - ltchiptool @ 4.12.0
  CUSTOM OPTIONS:
   - fw_name = esphome
   - fw_version = 2025.5.1
  Dependency Graph
  |-- AsyncTCP-esphome @ 2.1.4
  |-- ESPAsyncWebServer-esphome @ 3.3.0
  |-- DNSServer @ 1.1.0
  |-- noise-c @ 0.1.6
  |-- ArduinoJson @ 6.18.5
  Compiling .pioenvs/4-433/src/main.cpp.o
  Linking .pioenvs/4-433/raw_firmware.elf
  |-- Image 1: raw_firmware.elf
  |   |-- image_bk7231n_app.0x011000.rbl
  |   |   |-- raw_firmware.bin
  |   |-- image_bk7231n_app.0x011000.crc
  |   |-- image_bk7231n_app.0x129F0A.rblh
  |   |-- image_bk7231n_app.ota.rbl
  |   |-- image_bk7231n_app.ota.ug.bin
  RAM:   [====      ]  36.9% (used 96756 bytes from 262144 bytes)
  Flash: [========= ]  92.9% (used 1006596 bytes from 1083136 bytes)
  Building UF2 OTA image
  |-- esphome_2025.5.1_cb3s_bk7231n_lt1.7.0.uf2
  |-- firmware.uf2
  |-- firmware.bin
  ========================= [SUCCESS] Took 12.86 seconds =========================
  INFO Successfully compiled program.
  INFO Connecting to 192.168.177.162 port 8892...
  INFO Connected to 192.168.177.162
  INFO Uploading /data/build/4-433/.pioenvs/4-433/firmware.uf2 (3471872 bytes)
  Uploading: [============================================================] 100% Done...
  
  INFO Upload took 25.01 seconds, waiting for result...
  INFO OTA successful
  INFO Successfully uploaded program.
  INFO Starting log output from 192.168.177.162 using esphome API
  WARNING Can't connect to ESPHome API for 4-433 @ 192.168.177.162: Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.177.162', port=6053))] (TimeoutAPIError)
[... 345 lines cut ...]
  [11:27:17][W][dallas.temp.sensor:131]: 'Vorlauf (S10)' - Scratch pad checksum invalid!
  [11:27:17][D][dallas.temp.sensor:133]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
  [11:27:17][D][sensor:093]: 'Solar (S1) ADC PT1000 Raw': Sending state 1.27144 V with 4 decimals of accuracy
  [11:27:17][D][sensor:093]: 'Solar (S1) ADC PT1000 Voltage': Sending state 1.48504 V with 4 decimals of accuracy
  [11:27:17][D][sensor:093]: 'Solar (S1) PT1000 Resistance': Sending state 1212.10010 Ω with 1 decimals of accuracy
  [11:27:17][D][dht:049]: Got Temperature=14.8°C Humidity=77.7%
  [11:27:17][D][sensor:093]: 'Aussen (S9)': Sending state 14.80000 °C with 1 decimals of accuracy
  [11:27:18][W][dallas.temp.sensor:131]: 'Rücklauf' - Scratch pad checksum invalid!
  [11:27:18][D][dallas.temp.sensor:133]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
  [11:27:21][D][sensor:093]: 'NaN Rücklauf': Sending state 6.00000  with 0 decimals of accuracy
  [11:27:21][E][one_wire:025]: no devices responded to the reset
  [11:27:21][W][dallas.temp.sensor:044]: 'Puffer oben (S16)' - requesting conversion failed bus reset
  [11:27:21][D][sensor:093]: 'NaN Vorlauf': Sending state 6.00000  with 0 decimals of accuracy
  [11:27:22][D][sensor:093]: 'Solar (S1) ADC PT1000 Raw': Sending state 1.27078 V with 4 decimals of accuracy
  [11:27:22][D][sensor:093]: 'Solar (S1) ADC PT1000 Voltage': Sending state 1.48427 V with 4 decimals of accuracy
  [11:27:22][D][sensor:093]: 'Solar (S1) PT1000 Resistance': Sending state 1210.96069 Ω with 1 decimals of accuracy
  [11:27:22][E][one_wire:025]: no devices responded to the reset
  [11:27:22][W][dallas.temp.sensor:044]: 'Puffer Solar (S2)' - requesting conversion failed bus reset
  [11:27:23][E][one_wire:025]: no devices responded to the reset
  [11:27:23][W][dallas.temp.sensor:044]: 'Puffer Solar (S3)' - requesting conversion failed bus reset
  [11:27:24][E][one_wire:025]: no devices responded to the reset
  [11:27:24][W][dallas.temp.sensor:044]: 'Puffer unten (S7)' - requesting conversion failed bus reset
  [11:27:26][W][dallas.temp.sensor:131]: 'Solar-Rücklauf' - Scratch pad checksum invalid!
  ```
</details>

ssieb, 2025-05-29:
@cosote interesting, the detection works, but not the reading.  It's likely the same issue as all the other platforms, but I'm not sure what I can about it on the BK...  I have some devices, so I'll see what I can do.

iddiek, 2025-05-30:
> I've fixed rp2040, at least for non-parasitic. Either I don't understand how to set it up or all my sensors are fake and don't work, so I can't test that.
>
> external_components:
>   - source: [...]

Your fix listed as

external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ one_wire, gpio, dallas_temp ]

Works on D1 mini board too

JonLaliberte, 2025-05-31:
Using the 2025.4.2 components worked for me on esp8266 (sonoff th16). Hopefully I remember to undo it when this gets fixed 😬

ssieb, 2025-05-31:
I finally found the issue with 8266.  This should work now:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ one_wire, gpio, esp8266, dallas_temp ]
```

iddiek, 2025-05-31:
> I don't see how that's possible, since the only significant changes were in the `rp2040` component and you said you needed to rollback the `esp8266` component to make it work. Also, it isn't fixed for me on an esp8266, so I'm still trying to figure it out.

For me the problem was fixed by gpio component in your repo but if it was forked from previous version also does the job ;)

ssieb, 2025-05-31:
If you had previously used the old version and then rebooted into the new one without power cycling, it would appear to work.

closed 2025-05-31 by kbx81

freijn, 2025-06-01:
just to be 100%

external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: dev
    components: [ one_wire, gpio, esp8266, dallas_temp ]

is what you like to be tested ?

peterle00370, 2025-06-01:
> If you had previously used the old version and then rebooted into the new one without power cycling, it would appear to work.

I'm not sure how to understand "...without power cycling, it would appear to work"   **SHOULD** I power-cycle the device?

I'm only using     **components: [ esp8266 ]**    
WITHOUT power cycling it works perfectly.

JonLaliberte, 2025-06-02:
Sorry, what is a soft reboot here? I always assumed after installing new code it triggered a hard reboot.

ssieb, 2025-06-02:
A reboot instead of a power cycle.

andypnz, 2025-06-02:
Also an issue on RP 2040 board

bs617, 2025-06-03:
> Also an issue on RP 2040 board Dallas temp sensor Works with 2025.4, not working 2025.5

Thinking my Dallas temp sensor went bad I tried replacing it with other sensor to no avail.  I then tried it with another board, to no avail.  Finally I realized all my RP2040 (PiPico W) boards with the Dallas sensor (DS18b20) were not working.  One thing I noticed are the boards I tried swapping the sensor will not find a device, however the one I haven't touched finds the device but then get the following entries when it tries to read the temps:

`[22:29:12][W][component:182]: Component dallas_temp.sensor cleared Warning flag
[22:29:13][W][dallas.temp.sensor:128]: 'Coop Temperature' - Scratch pad checksum invalid!
[22:29:13][W][component:167]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[22:29:13][D][dallas.temp.sensor:130]: Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)`

JonLaliberte, 2025-06-03:
> A reboot instead of a power cycle.

Sorry, these mean the same thing _to me_ right now. 
Soft reboot = install new code and it reboots, correct? 
Does this mean that if I make changes to the code around pin settings that those changes don't take effect unless I physically cut power?

Power cycle = Unplug it and plug it back in? Or trigger a reboot of the device via HA?

ssieb, 2025-06-03:
> Does this mean that if I make changes to the code around pin settings that those changes don't take effect unless I physically cut power?

This was a specific bug in the code I added.

> Power cycle = Unplug it and plug it back in? Or trigger a reboot of the device via HA?

Power cycle means unplug and plug it back in.  A hard reboot can mean that or pressing the reset button.  A soft reboot is the reboot button or what happens when doing an update.

ssieb, 2025-06-04:
Did you try 2025.5.2?  No external components needed.

derkrasseleo, 2025-06-04:
can confirm dallas sensor works on 2025.5.2 without external components

andypnz, 2025-06-04:
RP 2040 board, Dallas temp sensor
Working again with 2025.5.2
Thank you

janchlebek, 2025-06-04:
Confirmed Dallas working with 2025.5.2, no pull-up resistor.

wineds, 2025-07-17:
ESPHome 2025.7.1  breaks the external components workaround?  

`#external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components: [ one_wire, gpio, dallas_temp, bk72xx ]    
`
DS18B20 was working with prior release with this workaround.  Now back scratchpad errors 

`08:48:53	[W]	[component:307]	
dallas_temp.sensor cleared Warning flag
08:48:53	[W]	[component:279]	
dallas_temp.sensor set Warning flag: scratch pad checksum invalid
08:48:53	[D]	[dallas.temp.sensor:129]	
Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
08:48:53	[D]	[sensor:104]	
'room_temperature': Sending state nan °C with 1 decimals of accuracy
08:49:13	[W]	[component:307]	
dallas_temp.sensor cleared Warning flag
08:49:13	[W]	[component:279]	
dallas_temp.sensor set Warning flag: scratch pad checksum invalid
08:49:13	[D]	[dallas.temp.sensor:129]	
Scratch pad: FF.FF.FF.FF.FF.FF.FF.FF.FF (C9)
08:49:14	[D]	[sensor:104]	
'room_temperature': Sending state nan °C with 1 decimals of accuracy
`

Bluscream, 2025-09-29:
I connect + to 5V, - to G and S to D3 on a D1 Mini and i get:

```
21:41:30
[W]
[component:326]
dallas_temp.sensor cleared Warning flag
21:41:31
[E]
[one_wire:023]
1-wire bus is held low
21:41:31
[E]
[one_wire:023]
1-wire bus is held low
21:41:31
[W]
[dallas.temp.sensor:066]
'DS18B20 Temperature' - reading scratch pad failed bus reset
21:41:31
[W]
[component:296]
dallas_temp.sensor set Warning flag: bus reset failed
```