esphome/issues #7040: pmsx003 Start character 1 mismatch with 2025.5.0

MallocArray, 2025-05-21:
### The problem

After upgrading from 2025.4.x to 2025.5.0, the PMSx003 sensor is reporting a warning when it reads the sensor

```
[12:07:35][D][pmsx003:262]: Got PM0.3 Particles: 96 Count/0.1L, PM0.5 Particles: 28 Count/0.1L, PM1.0 Particles: 2 Count/0.1L, PM2.5 Particles 0 Count/0.1L, PM5.0 Particles: 0 Count/0.1L, PM10.0 Particles 0 Count/0.1L
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x4B != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x12 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x01 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x97 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0x01 != 0x42
[12:07:35][W][pmsx003:125]: Start character 1 mismatch: 0xA0 != 0x42
```

### Which version of ESPHome has the issue?

2025.5.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

AirGradient ONE v9

### Component causing the issue

pmsx003

### YAML Config

```yaml
sensor:
  - platform: pmsx003
    # Default interval of updating every second, but changing to 30 seconds to extend life but keep fan spinning
    # PMS5003 https://esphome.io/components/sensor/pmsx003.html
    type: PMSX003
    uart_id: pms5003_uart
    pm_2_5:
      name: "PM 2.5"
      id: pm_2_5
      device_class: pm25  # Added to report properly to HomeKit
      filters:
        - lambda: |-
            float result = 0.0;
            if (x == 0.0) {
              result = 0.0;
            } else if (x < 30.0) {
              result = (0.524 * x) - (0.0862 * id(humidity_raw).state) + 5.75;
            } else if (x < 50.0) {
              result = (0.786 * (x / 20 - 3/2) + 0.524 * (1 - (x / 20 - 3/2))) * x - (0.0862 * id(humidity_raw).state) + 5.75;
            } else if (x < 210.0) {
              result = (0.786 * x) - (0.0862 * id(humidity_raw).state) + 5.75;
            } else if (x < 260.0) {
              result = (0.69 * (x / 50 - 21/5) + 0.786 * (1 - (x / 50 - 21/5))) * x - (0.0862 * id(humidity_raw).state * (1 - (x / 50 - 21/5))) + (2.966 * (x / 50 - 21/5)) + (5.75 * (1 - (x / 50 - 21/5))) + (8.84 * pow(10,-4) * pow(x,2) * (x / 50 - 21/5));
            } else {
              result = 2.966 + (0.69 * x) + (8.84 * pow(10,-4) * pow(x,2));
            }
            if (result <= 0.0) {
              return 0.0;
            } else {
              return result;
            }
    pm_1_0:
      name: "PM 1.0"
      id: pm_1_0
      device_class: pm1  # Added to report properly to HomeKit
    pm_10_0:
      name: "PM 10.0"
      id: pm_10_0
      device_class: pm10  # Added to report properly to HomeKit
    pm_0_3um:
      name: "PM 0.3"
      id: pm_0_3um
      filters:
    update_interval: 30s
```

### Anything in the logs that might be useful for us?

```txt

```

jesserockz, 2025-05-22:
Looking at the PR, this was silently ignoring the bytes before, and is now logging a warning which might not be the best thing @ximex

ximex, 2025-05-22:
will check later today

M66B, 2025-06-02:
For comparison, a "good" log:

```
[15:23:03][D][uart_debug:158]: >>> "BM\xE4\x00\x01\x01t"
[15:23:33][D][uart_debug:158]: <<< "BM\x00\x04\xE4\x00\x01wBM\x00\x1C\x00\x03\x00\x03\x00\x03\x00\x03\x00\x03\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x97\x00\x01TBM\x00\x1C\x00\x02\x00\x03\x00\x03\x00\x02\x00\x03\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x97\b\x01ZBM\x00\x1C\x00\x03\x00\x04\x00\x04\x00\x03\x00\x04\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x97\b\x01`BM\x00\x1C\x00\x03\x00\x04\x00\x04\x00\x03\x00\x04\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
[15:23:33][D][uart_debug:158]: <<< "\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x97\b\x01`BM\x00\x1C\x00\x03\x00\x04\x00\x04\x00\x03\x00\x04\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x97\b\x01`BM\x00\x1C\x00\x03\x00\x04\x00\x04\x00\x03\x00\x04\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x97\b\x01`BM\x00\x1C\x00\x03\x00\x04\x00\x04\x00\x03\x00\x04\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x97\b\x01`\x03\x00\x04\x00\x04\x00\x03\x00\x04\x00\x04\x02|\x00\xBB\x0
[15:23:33][D][uart_debug:158]: <<< "\x04\x00\x03\x00\x04\x00\x04\x02g\x00\xB9\x00\x15\x00\x01\x00\x01\x00\x00\x97\x00\x02\x91BM\x00\x1C\x00\x03\x00\x04\x00\x04\x00\x03\x00\x04\x00\x04\x02g\x00\xB9\x00\x15\x00\x01\x00\x01\x00\x00\x97\x00\x02\x91BM\x00\x1C\x00\x03\x00\x04\x00\x04\x00\x03\x00\x04\x00\x04\x02g\x00\xB9\x00\x15\x00\x01\x00\x01\x00\x00\x97\x00\x02\x91"
[15:23:33][D][uart_debug:158]: >>> "BM\xE2\x00\x00\x01q"
[15:23:34][D][pmsx003:226]: Got PM1.0 Standard Concentration: 3 µg/m³, PM2.5 Standard Concentration 4 µg/m³, PM10.0 Standard Concentration: 4 µg/m³, PM1.0 Concentration: 3 µg/m³, PM2.5 Concentration 4 µg/m³, PM10.0 Concentration: 4 µg/m³
[15:23:34][D][sensor:094]: '<1.0µm': Sending state 3.00000 µg/m³ with 0 decimals of accuracy
[15:23:34][D][sensor:094]: '<2.5µm': Sending state 4.00000 µg/m³ with 0 decimals of accuracy
[15:23:34][D][sensor:094]: '<10.0µm': Sending state 4.00000 µg/m³ with 0 decimals of accuracy
[15:23:34][D][pmsx003:266]: Got PM0.3 Particles: 627 Count/0.1L, PM0.5 Particles: 187 Count/0.1L, PM1.0 Particles: 22 Count/0.1L, PM2.5 Particles 1 Count/0.1L, PM5.0 Particles: 1 Count/0.1L, PM10.0 Particles 0 Count/0.1L
[15:23:34][D][uart_debug:158]: <<< "BM\x00\x1C\x00\x03\x00\x04\x00\x04\x00\x03\x00\x04\x00\x04\x02s\x00\xBB\x00\x16\x00\x01\x00\x01\x00\x00\x97\x00\x02\xA0"
[15:23:34][D][uart_debug:158]: >>> "BM\xE4\x00\x00\x01s"
```

MikeBP13, 2025-06-03:
I'm getting the same error as OP with esphome 2025.5.2 on the same type of device (different configuration).

MikeBP13, 2025-06-19:
Does the 2025.6.0 version fix anything? Thanks.

M66B, 2025-06-19:
Seems okay now.

MikeBP13, 2025-06-20:
The issue is still present in esphome 2025.6.0.

ximex, 2025-06-20:
Pls test this PR if this fix the problem.
https://github.com/esphome/esphome/pull/8989

MikeBP13, 2025-06-21:
With esphome 2025.6.0, the PMS5003 stopped sending values out of the sudden on 3 identical devices (same HW, same config). One of the devices resumed after 7h, while the other 2 seem stuck sending data at random intervals or not at all. The fans are spinning on all sensors.

MikeBP13, 2025-06-22:
After several hours of testing after a clean restart, the behavior of the sensors is erratic. Sometimes data is sent (the absolute values doesn't seem to be correct) and sometimes the sensors report 0. No errors are logged.

MikeBP13, 2025-06-22:
> Pls test this PR if this fix the problem. [esphome/esphome#8989](https://github.com/esphome/esphome/pull/8989)

Now, I've switched to this implementation. Will test in the following hours to see if it works.

MikeBP13, 2025-06-27:
> > Pls test this PR if this fix the problem. [esphome/esphome#8989](https://github.com/esphome/esphome/pull/8989)
> 
> Now, I've switched to this implementation. Will test in the following hours to see if it works.

It appears to be working fine with esphome 2025.6.0 after a manual restart, but somehow flaky with esphome 2025.6.1 - 2 of the 3 devices tested stopped reporting values for several hours without any error in the logs. They have been working for a couple of hours, then went silent. One of them resumed after 7-8 hours.

I've also noticed that a manual power off/power on must be applied after OTA update to get the devices working reliably.

Another thing is that the refresh interval is not always respected. I have configured 30s. Sometimes values are reported exactly at 30s mark, sometimes after 1-2 min.

MallocArray, 2025-06-27:
@MikeBP13 
For what it is worth, there is a PR pending that helps address the refresh interval not behaving as expected
https://github.com/esphome/esphome/pull/8989

While testing this PR I also noticed needing to do a full power cycle to get the device to report sometimes, but I hadn't seen it prior to testing the PR.

MallocArray, 2025-07-04:
I wonder if the need for a power cycle to get values back from the sensor is related to this open issue: https://github.com/esphome/issues/issues/6344

MallocArray, 2025-08-22:
Can this still be looked at and this issue migrated to the new location?

bednim, 2025-09-07:
@boussouira - I hope so - VCC pin on ESP32 d1mini board :)

bednim, 2025-09-07:
@boussouira sadly issue remains at 3.3v :( whats you code? mines very basic - not sure if BMSEC component would break it? or perhaps esp-idf instead of arduino?

bednim, 2025-12-27:
> Any updates? Seeing these errors on a recent sensor I built.

Hey have you updated to most recent public version of esphome? My device suddenly started working!

MikeBP13, 2025-12-28:
The initial reported issue is still present on my side as well and has been for the last esphome updates...