esphome/issues #7042: ESPHome fails to load external ADF pipeline sink platform `db_sound_meter`

wjohnson2250, 2025-05-21:
### The problem

**Description**:

I'm developing a custom ESPHome ADF pipeline **sink platform** called `db_sound_meter`. This platform is designed to process audio frames from an I²S microphone and compute a sound level (in decibels), effectively acting as a virtual sound pressure level meter. It is not a full component domain, but a platform-only integration intended for use strictly within the `adf_pipeline:` YAML block.

I'm trying to implement a custom ADF pipeline **sink platform** (`db_sound_meter`) using ESPHome's `external_components` mechanism. The component is structured according to the `esphome_audio` model (e.g. `i2s_audio`) and is used solely under the `adf_pipeline:` key in YAML. However, ESPHome fails to load the platform and incorrectly reports that `__init__.py` is missing — even though the platform implementation is provided at `components/db_sound_meter/adf_pipeline/__init__.py`, and the root-level `__init__.py` is present.

---

### 📦 Component Overview (following the `esphome_audio` model)

**Repository**: Structure follows the `esphome_audio` model, based on `i2s_audio`. Source available on request.

**Structure:**
```
components/
  db_sound_meter/
    __init__.py               ← present but empty (no CONFIG_SCHEMA)
    component.yaml            ← domain: db_sound_meter, dependencies: [adf_pipeline]
    adf_pipeline/
      __init__.py             ← contains:
                                - CONFIG_SCHEMA
                                - to_code()
                                - cg.class_() registration for a subclass of ADFPipelineSink
```

**ESPHome YAML**:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/wjohnson2250/esphome_audio_sink.git
      ref: main
    components: [db_sound_meter]

adf_pipeline:
  - platform: db_sound_meter
    type: audio_out
    id: sound_meter_sink
```

---

### ✅ Verified:
- `components/db_sound_meter/__init__.py` exists
- ESPHome version `2025.4.2` is used
- `db_sound_meter` is never referenced as a top-level YAML domain
- `print()` statements inside `db_sound_meter/adf_pipeline/__init__.py` are never executed

---

### ❗Error:
```
Could not find __init__.py file for component db_sound_meter.
(search path: /data/external_components/.../components/db_sound_meter/__init__.py)
```

This suggests ESPHome fails to resolve `db_sound_meter` correctly despite its usage only under `adf_pipeline:` and the correct platform structure being present.

---

### 🔍 Key Questions for the Community:

1. Does ESPHome support external components that are platform-only — i.e. not top-level domains?
2. Why does ESPHome reject `db_sound_meter` when the YAML does not reference it directly as a component block?
3. Is there a way to register a valid ADF pipeline platform (like `i2s_audio`) without requiring a top-level `CONFIG_SCHEMA` or stub implementation?

---

### 💡 Suggestions:
- Support for platform-only components that implement valid platforms under existing domains (like `adf_pipeline`) without requiring stub top-level `__init__.py` logic
- Better error messaging when ESPHome is attempting to validate external components solely based on top-level presence

Thank you for any clarification or workarounds.

### Which version of ESPHome has the issue?

2025.4.2 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.4.4

### What platform are you using?

ESP32

### Board

AZ-Delivery

### Component causing the issue

i2c

### YAML Config

```yaml
See description
```

### Anything in the logs that might be useful for us?

```txt

```

jzucker2, 2025-05-22:
How does this differ from the new `sound_level` component? 

https://esphome.io/components/sensor/sound_level

wjohnson2250, 2025-05-22:
Thank you for the pointer to the new sound_level component — it’s a great fit
for analog microphones connected to the ADC.

However, this project specifically targets *digital I²S microphones* like the
*INMP441*, which:

   -

   Transmit raw 24-bit PCM audio over I²S (not analog voltage)
   -

   Require *clocked digital interfaces*, not ESP32’s ADC
   -

   Are *very inexpensive*, typically costing *less than £1 GBP / $1 USD*
   -

   Are widely used in embedded and consumer audio designs

Unfortunately, the sound_level sensor in ESPHome:

   -

   Only supports analog input (pin:), e.g. for LM393, KY-038, MAX4466
   -

   Does *not* support I²S input or PCM audio frame analysis
   -

   Cannot access real-time audio buffers needed to compute dB from digital
   signals

This is why the solution relies on esphome_audio and the ADF pipeline — it’s
currently the only way to:

   -

   Integrate I²S input with sample-level access
   -

   Process raw audio in a pipeline sink
   -

   Accurately compute sound level (in decibels) from digital audio data

Regards.

wjohnson2250, 2025-05-25:
After the latest updates, ESPHome can load the ADF pipeline sink platform. The original loading problem with db_sound_meter no longer occurs. Closing this issue.

closed 2025-05-25 by wjohnson2250