esphome/issues #7043: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2025.5.0)

alfwro13, 2025-05-21:
### The problem

After upgrading my device to esphome 2025.5.0 it keeps crashing with this error:

### Which version of ESPHome has the issue?

 2025.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.2

### What platform are you using?

ESP8266

### Board

esp01_1m

### YAML Config

```yaml
substitutions:
  device_name: ld2410-tv-room
  device_friendly_name: "LD2410 TV Motion Sensor"
  ip_address: 192.168.1.89
  room: "TV Room"
  board: esp01_1m

esphome:
  !include common/esphome.yaml

esp8266:
  board: ${board}

logger:
  baud_rate: 0

api:
  encryption:
    key: [redacted]

ota:
  - platform: esphome
    password: [redacted]

wifi:
  !include common/wifi.yaml

captive_portal:

uart:
  tx_pin: 15
  rx_pin: 13
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:

binary_sensor:
  - !include common/bin_sensor_status.yaml
  - platform: ld2410
    has_target:
      name: Presence
[... 201 lines cut ...]
WARNING Can't connect to ESPHome API for ld2410-tv-room @ 192.168.1.89: ld2410-tv-room @ 192.168.1.89: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2025.5.0) (EncryptionHelloAPIError)
INFO Trying to connect to ld2410-tv-room @ 192.168.1.89 in the background
INFO Successfully connected to ld2410-tv-room @ 192.168.1.89 in 0.004s
INFO Successful handshake with ld2410-tv-room @ 192.168.1.89 in 2.882s
[21:45:44][D][sensor:093]: 'Still Energy': Sending state 10.00000 % with 0 decimals of accuracy
[21:45:45][D][sensor:093]: 'Still Energy': Sending state 9.00000 % with 0 decimals of accuracy
[21:45:46][D][sensor:093]: 'Still Energy': Sending state 10.00000 % with 0 decimals of accuracy
WARNING ld2410-tv-room @ 192.168.1.89: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for ld2410-tv-room @ 192.168.1.89
WARNING Disconnected from API
INFO Successfully connected to ld2410-tv-room @ 192.168.1.89 in 0.002s
WARNING ld2410-tv-room @ 192.168.1.89: Connection error occurred: ld2410-tv-room @ 192.168.1.89: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2025.5.0)
WARNING Can't connect to ESPHome API for ld2410-tv-room @ 192.168.1.89: ld2410-tv-room @ 192.168.1.89: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2025.5.0) (EncryptionHelloAPIError)
INFO Trying to connect to ld2410-tv-room @ 192.168.1.89 in the background
INFO Successfully connected to ld2410-tv-room @ 192.168.1.89 in 0.012s
WARNING ld2410-tv-room @ 192.168.1.89: Connection error occurred: ld2410-tv-room @ 192.168.1.89: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2025.5.0)
INFO Successfully connected to ld2410-tv-room @ 192.168.1.89 in 7.213s
INFO Successful handshake with ld2410-tv-room @ 192.168.1.89 in 2.918s
[21:46:30][D][sensor:093]: 'Still Energy': Sending state 9.00000 % with 0 decimals of accuracy
WARNING ld2410-tv-room @ 192.168.1.89: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for ld2410-tv-room @ 192.168.1.89
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for ld2410-tv-room @ 192.168.1.89: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.89', port=6053))]: [Errno 111] Connect call failed ('192.168.1.89', 6053) (SocketAPIError)
INFO Trying to connect to ld2410-tv-room @ 192.168.1.89 in the background
```

alfwro13, 2025-05-21:
flashed it back with ESPHome 2025.4.0 and all good again

bdraco, 2025-05-22:
Please connect a usb cable and get a log from the console

DerEmder, 2025-05-22:
Since todays espHome update I get the same error, device is not connected with Home Assistant but I can log wirelessly. I also use the LD2410 (esp32cam). Please fix asap.  I annot log via cable because the device is built into the wall. Log follows (it keeps repeating over and over again).:

```
INFO ESPHome 2025.5.0
---
INFO Starting log output from 10.42.0.69 using esphome API
INFO Successfully connected to max @ 10.42.0.69 in 0.055s
INFO Successful handshake with max @ 10.42.0.69 in 0.110s
[22:19:36][I][app:115]: ESPHome version 2025.5.0 compiled on May 22 2025, 22:09:23
[22:19:36][C][wifi:600]: WiFi:
[22:19:36][C][wifi:428]:   Local MAC: ---
[22:19:36][C][wifi:433]:   SSID: ---
[22:19:36][C][wifi:436]:   IP Address: 10.42.0.69
[22:19:36][C][wifi:440]:   BSSID: 24:65:11:3C:5C:96[redacted]
[22:19:36][C][wifi:441]:   Hostname: 'max'
[22:19:36][C][wifi:443]:   Signal strength: -60 dB ▂▄▆█
[22:19:36][C][wifi:447]:   Channel: 11
[22:19:36][C][wifi:448]:   Subnet: 255.255.255.0
[22:19:36][C][wifi:449]:   Gateway: 10.42.0.1
[22:19:36][C][wifi:450]:   DNS1: 10.42.0.1
[22:19:36][C][wifi:451]:   DNS2: 0.0.0.0
[22:19:36][C][logger:224]: Logger:
[22:19:36][C][logger:225]:   Max Level: DEBUG
[22:19:36][C][logger:226]:   Initial Level: DEBUG
[22:19:36][C][logger:228]:   Log Baud Rate: 0
[22:19:36][C][logger:229]:   Hardware UART: UART0
[22:19:36][C][logger:233]:   Task Log Buffer Size: 768
[22:19:36][C][i2c.arduino:071]: I2C Bus:
[22:19:36][C][i2c.arduino:072]:   SDA Pin: GPIO14
[22:19:36][C][i2c.arduino:073]:   SCL Pin: GPIO13
[22:19:36][C][i2c.arduino:074]:   Frequency: 50000 Hz
[22:19:36][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[22:19:36][C][uart.arduino_esp32:151]: UART Bus 0:
[22:19:36][C][uart.arduino_esp32:152]:   TX Pin: GPIO1
[22:19:36][C][uart.arduino_esp32:153]:   RX Pin: GPIO3
[22:19:36][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[22:19:36][C][uart.arduino_esp32:157]:   Baud Rate: 256000 baud
[22:19:36][C][uart.arduino_esp32:158]:   Data Bits: 8
[22:19:36][C][uart.arduino_esp32:159]:   Parity: NONE
[22:19:36][C][uart.arduino_esp32:160]:   Stop bits: 1
[22:19:36][C][ledc.output:180]: LEDC Output:
[22:19:36][C][ledc.output:181]:   Pin GPIO4
[22:19:36][C][ledc.output:182]:   LEDC Channel: 2
[22:19:36][C][ledc.output:183]:   PWM Frequency: 1000.0 Hz
[22:19:36][C][ledc.output:184]:   Phase angle: 0.0°
[22:19:36][C][ledc.output:185]:   Bit depth: 16
[22:19:36][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Max Motion'
[22:19:36][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[22:19:37][C][gpio.binary_sensor:016]:   Pin: GPIO16
[22:19:38][C][light:092]: Light 'Max LED'
[22:19:38][C][light:094]:   Default Transition Length: 1.0s
[22:19:38][C][light:095]:   Gamma Correct: 2.80
[22:19:38][C][esp32_camera:048]: ESP32 Camera:
[22:19:38][C][esp32_camera:049]:   Name: Max Camera
[22:19:38][C][esp32_camera:050]:   Internal: NO
[22:19:38][C][esp32_camera:052]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[22:19:38][C][esp32_camera:053]:   VSYNC Pin: 25
[22:19:38][C][esp32_camera:054]:   HREF Pin: 23
[22:19:38][C][esp32_camera:055]:   Pixel Clock Pin: 22
[22:19:38][C][esp32_camera:056]:   External Clock: Pin:0 Frequency:10000000
[22:19:38][C][esp32_camera:060]:   I2C Pins: SDA:26 SCL:27
[22:19:38][C][esp32_camera:062]:   Reset Pin: -1
[22:19:38][C][esp32_camera:080]:   Resolution: 640x480 (VGA)
[22:19:38][C][esp32_camera:129]:   JPEG Quality: 10
[22:19:38][C][esp32_camera:131]:   Contrast: 0
[22:19:38][C][ld2410:075]:   Throttle_ : 1000ms
[22:19:42][D][sensor:094]: 'Max Still Energy': Sending state 2.00000 % with 0 decimals of accuracy
[22:19:43][D][sensor:094]: 'Max Still Energy': Sending state 3.00000 % with 0 decimals of accuracy
[22:19:48][D][sensor:094]: 'Max Still Energy': Sending state 2.00000 % with 0 decimals of accuracy
[22:19:49][D][sensor:094]: 'Max Still Energy': Sending state 3.00000 % with 0 decimals of accuracy
[22:19:50][D][sensor:094]: 'Max Still Energy': Sending state 2.00000 % with 0 decimals of accuracy
[22:19:53][D][sensor:094]: 'Max Still Energy': Sending state 3.00000 % with 0 decimals of accuracy
[22:19:54][D][sensor:094]: 'Max Still Energy': Sending state 2.00000 % with 0 decimals of accuracy
[22:19:55][D][sensor:094]: 'Max Still Energy': Sending state 3.00000 % with 0 decimals of accuracy
[22:19:57][D][sensor:094]: 'Max Still Energy': Sending state 2.00000 % with 0 decimals of accuracy
[22:19:58][D][sensor:094]: 'Max Still Energy': Sending state 3.00000 % with 0 decimals of accuracy
[22:20:04][D][api:122]: Accepted 10.42.0.3
[22:20:04][W][component:257]: Component api took a long time for an operation (52 ms).
[22:20:04][W][component:258]: Components should block for at most 30 ms.
[22:20:04][D][api.connection:1529]: Home Assistant 2025.5.2 (10.42.0.3): Connected successfully
WARNING max @ 10.42.0.69: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for max @ 10.42.0.69
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for max @ 10.42.0.69: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.42.0.69', port=6053))]: [Errno 111] Connect call failed ('10.42.0.69', 6053) (SocketAPIError)
INFO Trying to connect to max @ 10.42.0.69 in the background
```

DerEmder, 2025-05-22:
```
INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/max.yaml...
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 10.42.0.69 using esphome API
WARNING Can't connect to ESPHome API for max @ 10.42.0.69: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.42.0.69', port=6053))]: [Errno 111] Connect call failed ('10.42.0.69', 6053) (SocketAPIError)
INFO Trying to connect to max @ 10.42.0.69 in the background
INFO Successfully connected to max @ 10.42.0.69 in 3.111s
INFO Successful handshake with max @ 10.42.0.69 in 0.124s
[22:39:57][I][app:115]: ESPHome version 2025.5.0 compiled on May 22 2025, 22:09:23
[22:39:57][C][wifi:600]: WiFi:
WARNING max @ 10.42.0.69: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for max @ 10.42.0.69
WARNING Disconnected from API
INFO Successfully connected to max @ 10.42.0.69 in 0.039s
INFO Successful handshake with max @ 10.42.0.69 in 0.104s
[22:40:38][D][api:122]: Accepted 10.42.0.3
[22:40:38][W][component:257]: Component api took a long time for an operation (52 ms).
[22:40:38][W][component:258]: Components should block for at most 30 ms.
[22:40:38][D][api.connection:1529]: Home Assistant 2025.5.2 (10.42.0.3): Connected successfully
WARNING max @ 10.42.0.69: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for max @ 10.42.0.69
WARNING Disconnected from API
INFO Successfully connected to max @ 10.42.0.69 in 0.037s
INFO Successful handshake with max @ 10.42.0.69 in 0.100s
[22:41:18][D][api:122]: Accepted 10.42.0.3
[22:41:18][W][component:257]: Component api took a long time for an operation (52 ms).
[22:41:18][W][component:258]: Components should block for at most 30 ms.
[22:41:18][D][api.connection:1529]: Home Assistant 2025.5.2 (10.42.0.3): Connected successfully
```

That's all I get now ...

bdraco, 2025-05-22:
Thanks for sharing the logs. From what you're seeing, it does look like the device might be crashing or rebooting, especially with the repeated disconnects and "connection reset by peer" messages.

To help troubleshoot this properly, we’ll need either a minimal YAML configuration that reproduces the issue or, ideally, serial console logs that show what’s happening on boot or right before the disconnect. Without one of those, it's really difficult for us to replicate the problem on our side and figure out what's going wrong.

If you can share the config (even a simplified version) or connect to the device over USB and grab the console output, that would be a huge help.

norriscm, 2025-05-22:
I am also seeing random resets on ESP32s after upgrading to 2025.5.0. I have not yet been able to capture logs around the event (and the wakeup reason after is simply "undefined"), but unlike the reports above I am not using an esp32cam or LD2420. I am seeing resets on devices that are just WROOMs driving PWM LEDs, so the the issue (if it's the same one) is not specific to either the esp32cam board or the LD2420 integration.

I have a suspicion that the issue is related to logging, perhaps downstream of https://github.com/esphome/esphome/pull/8736 . I have not seen this issue on devices that rarely emit logs, while one of my devices started resetting every few minutes when I added additional logging to a frequently run lambda; removing that logging seemed to reduce the frequency of resets.

I am trying to capture serial logs from one of my devices with the problem. I am also disabling all logging on another device to see whether the resets continue. I will let you know if I get useful data from either approach.

bdraco, 2025-05-22:
If you do manage to get a trace on the console, if you recompile and upload the firmware on your machine that connects via USB, and run `esphome logs <xxx.yaml>` it should automatically decode the trace.

bdraco, 2025-05-22:
Even if you can't get serial logs, if you come up with a yaml config that works on a dev board that can reproduce, I'll build a device and get the trace.

DerEmder, 2025-05-23:
I almost forgot. Here is the yaml:

```
substitutions:
  firmware: '5.511'
  devicename: max
  upper_devicename: 'Max'

esphome:
  name: $devicename
  on_boot:
    priority: 600.0
    then:
      - output.set_level:
          id: flash
          level: 0%
      - delay: 500ms
      - output.turn_off: flash

esp32:
  board: esp32dev

external_components:
  - source:
      type: git
      url: https://github.com/MichaKersloot/esphome_custom_components
    components: [ esp32_camera ]

logger:
  baud_rate: 0

api:
  encryption:
    key: !secret max_api

ota:
  - platform: esphome
    password: !secret max_ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  fast_connect: true
  manual_ip:
    static_ip: 10.42.0.69
    gateway: 10.42.0.1
    subnet: 255.255.255.0
    dns1: 10.42.0.1
  ap:
    ssid: '$upper_devicename'
    password: !secret init_password

esp32_camera:
  name: '$upper_devicename Camera'
  id: cam
  external_clock:
    pin: GPIO0
    frequency: 10MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  max_framerate: 10 fps
  idle_framerate: 0.01 fps
  resolution: 640X480
[... 251 lines cut ...]
    factory_reset:
      name: '$upper_devicename Factory Reset'
    restart:
      name: '$upper_devicename Sensor Reboot'
    query_params:
      name: '$upper_devicename Get Params'

select:
  - platform: ld2410
    distance_resolution:
      name: '$upper_devicename Resolution'
    baud_rate:
      name: '$upper_devicename Baudrate'
    light_function:
      name: '$upper_devicename Light Function'
    out_pin_level:
      name: '$upper_devicename Level'

switch:
  - platform: ld2410
    engineering_mode:
      name: '$upper_devicename Config Mode'
    bluetooth:
      name: '$upper_devicename Bluetooth'
...

DerEmder, 2025-05-23:
I removed any logger or logging but still the same re-boot. I can OTA without problems but the API does not connect. I get the same on a Wemos D1 mini (8266) with an attached LD2410 and Neopixel Light. I got it to work after removing the partition scheme for an LED strip. So I assume it has something to do with memory, or better OOM.

DerEmder, 2025-05-23:
I removed the external camera component - no rebooting anymore but LD2410 does not work at all.
I removed the esp32 camera completely - everything works again, except the camera, of course.

Again - the whole system (HTU21D, LD2410 and cam) worked fine until the last espHome update.

bdraco, 2025-05-23:
external_components:
  - source:
      type: git
      url: https://github.com/MichaKersloot/esphome_custom_components
    components: [ esp32_camera ]

That's likely the problem. That repo needs to back port some of the fixes to esp32_camera made recently.

bdraco, 2025-05-23:
https://github.com/esphome/esphome/pull/8811

DerEmder, 2025-05-23:
I tried the latest PR for testing. I get:

```
[16:29:12][E][esp32_camera:123]:   Setup Failed: ESP_ERR_NOT_FOUND
[16:29:12][E][component:085]:   Component esp32_camera is marked FAILED: 

```
And still - why does my LD2410 not work anymore? I can only read values from LD2410 but not write any parameters. Very odd.

Full log:

```
INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/max.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/8811/head
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 10.42.0.69 using esphome API
INFO Successfully connected to max @ 10.42.0.69 in 0.036s
INFO Successful handshake with max @ 10.42.0.69 in 0.101s
[16:39:03][I][app:115]: ESPHome version 2025.5.0 compiled on May 23 2025, 16:27:13
[16:39:03][C][wifi:600]: WiFi:
[16:39:03][C][wifi:428]:   Local MAC: ---
[16:39:03][C][wifi:433]:   SSID: ---
[16:39:03][C][wifi:436]:   IP Address: 10.42.0.69
[16:39:03][C][wifi:440]:   BSSID: ---
[16:39:03][C][wifi:441]:   Hostname: 'max'
[16:39:03][C][wifi:443]:   Signal strength: -55 dB ▂▄▆█
[16:39:03][C][wifi:447]:   Channel: 11
[16:39:03][C][wifi:448]:   Subnet: 255.255.255.0
[16:39:03][C][wifi:449]:   Gateway: 10.42.0.1
[16:39:03][C][wifi:450]:   DNS1: 10.42.0.1
[16:39:03][C][wifi:451]:   DNS2: 0.0.0.0
[16:39:03][C][logger:224]: Logger:
[16:39:03][C][logger:225]:   Max Level: DEBUG
[16:39:03][C][logger:226]:   Initial Level: DEBUG
[16:39:03][C][logger:228]:   Log Baud Rate: 0
[16:39:03][C][i2c.arduino:071]: I2C Bus:
[16:39:03][C][i2c.arduino:072]:   SDA Pin: GPIO14
[16:39:03][C][i2c.arduino:073]:   SCL Pin: GPIO13
[16:39:03][C][i2c.arduino:074]:   Frequency: 50000 Hz
[16:39:03][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[16:39:03][C][uart.arduino_esp32:151]: UART Bus 0:
[16:39:03][C][uart.arduino_esp32:152]:   TX Pin: GPIO1
[16:39:03][C][uart.arduino_esp32:153]:   RX Pin: GPIO3
[16:39:03][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[16:39:03][C][uart.arduino_esp32:157]:   Baud Rate: 256000 baud
[16:39:03][C][uart.arduino_esp32:158]:   Data Bits: 8
[16:39:03][C][uart.arduino_esp32:159]:   Parity: NONE
[16:39:03][C][uart.arduino_esp32:160]:   Stop bits: 1
[16:39:03][C][ledc.output:180]: LEDC Output:
[16:39:03][C][ledc.output:181]:   Pin GPIO4
[16:39:03][C][ledc.output:182]:   LEDC Channel: 2
[16:39:03][C][ledc.output:183]:   PWM Frequency: 1000.0 Hz
[16:39:03][C][ledc.output:184]:   Phase angle: 0.0°
[16:39:03][C][ledc.output:185]:   Bit depth: 16
[16:39:03][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Max Motion'
[16:39:03][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[16:39:03][C][gpio.binary_sensor:016]:   Pin: GPIO16
[16:39:03][C][light:092]: Light 'Max LED'
[16:39:03][C][light:094]:   Default Transition Length: 1.0s
[16:39:03][C][light:095]:   Gamma Correct: 2.80
[16:39:03][C][esp32_camera:048]: ESP32 Camera:
[16:39:03][C][esp32_camera:049]:   Name: Max Camera
[16:39:03][C][esp32_camera:050]:   Internal: NO
[16:39:03][C][esp32_camera:052]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[... 174 lines cut ...]
[16:39:13][D][htu21d:098]: Heater Level=0
[16:39:18][D][sensor:094]: 'Max Still Energy': Sending state 3.00000 % with 0 decimals of accuracy
[16:39:21][D][sensor:094]: 'Max Still Energy': Sending state 4.00000 % with 0 decimals of accuracy
[16:39:22][D][sensor:094]: 'Max Still Energy': Sending state 3.00000 % with 0 decimals of accuracy
[16:39:30][D][sensor:094]: 'Max Still Energy': Sending state 4.00000 % with 0 decimals of accuracy
[16:39:32][D][sensor:094]: 'Max Still Energy': Sending state 6.00000 % with 0 decimals of accuracy
[16:39:33][D][sensor:094]: 'Max Still Energy': Sending state 11.00000 % with 0 decimals of accuracy
[16:39:34][D][sensor:094]: 'Max Still Energy': Sending state 16.00000 % with 0 decimals of accuracy
[16:39:35][D][sensor:094]: 'Max Still Energy': Sending state 20.00000 % with 0 decimals of accuracy
[16:39:36][D][sensor:094]: 'Max Still Distance': Sending state 449.00000 cm with 0 decimals of accuracy
[16:39:36][D][sensor:094]: 'Max Still Energy': Sending state 22.00000 % with 0 decimals of accuracy
[16:39:36][D][sensor:094]: 'Max Distance': Sending state 282.00000 cm with 0 decimals of accuracy
[16:39:37][D][sensor:094]: 'Max Still Distance': Sending state 444.00000 cm with 0 decimals of accuracy
[16:39:37][D][sensor:094]: 'Max Distance': Sending state 404.00000 cm with 0 decimals of accuracy
[16:39:38][D][sensor:094]: 'Max Still Distance': Sending state 442.00000 cm with 0 decimals of accuracy
[16:39:38][D][sensor:094]: 'Max Still Energy': Sending state 21.00000 % with 0 decimals of accuracy
[16:39:38][D][sensor:094]: 'Max Distance': Sending state 423.00000 cm with 0 decimals of accuracy
[16:39:39][D][sensor:094]: 'Max Still Distance': Sending state 441.00000 cm with 0 decimals of accuracy
[16:39:39][D][sensor:094]: 'Max Still Energy': Sending state 18.00000 % with 0 decimals of accuracy
[16:39:43][D][htu21d:058]: Got Temperature=34.7°C
[16:39:43][D][htu21d:079]: Got Humidity=21.2%
[16:39:43][D][sensor:094]: 'Max Hum': Sending state 21.20642 % with 1 decimals of accuracy
[16:39:43][D][htu21d:098]: Heater Level=0

```

DerEmder, 2025-05-23:
But anyway - the random reboots are gone due to the cam component. Thanks for that. I'll dig deeper. I assume I will never update my system anymore once everything works again :-)

DerEmder, 2025-05-23:
Forehead -> Table  

now this stupid system renamed all my entities from .max_motion or .max_led to .max_max_motion or .max_max_led and so on

I gotta love Home Assistant and espHome ....

anyway, LD2410 is working more or less but the Camera is unavailable. But thats a different issue ... the original problem is solved for me.

wired-dev, 2025-05-24:
I'm also having this issue on an esp8266 acting as a rf gateway,  the other issue I noticed was a compile time error fixing /s into /s+ two times.

bdraco, 2025-05-27:
Thanks for the update — that’s a helpful clue.

To better understand what’s going on, could you try compiling the firmware locally and running the following with the device connected?

```bash
esphome logs xxx.yaml
```

This should give us a full backtrace and help confirm whether the logger is involved in the reboots.

bdraco, 2025-06-01:
https://github.com/esphome/issues/issues/7043 should fix the original report.  The external component issues got me a bit of track with this one since we had other issues pile on there.

closed 2025-06-02 by jesserockz