esphome/issues #7046: Font size calculated differently since PR #8300

JonasB2497, 2025-05-22:
### The problem

I am using a waveshare eink display and after an update to esphome I noticed a change in font height. I could narrow it down to PR esphome/esphome#8300 but can't find any obvious issues with the new freetype implementation.

I am using the Gotham Rounded Medium font  (but noticed it with other fonts as well) which is scalable. The MWE produces the following two results. The last character in the string is the one in the font with the lowest descender. The glyphs itself keep the same size but comparing the lines (baseline and height) with the glyph view from a fontviewer (fontdrop.info in my case) the baselines seem to be correct but the height/descender line is way off with the freetype implementation.

### Which version of ESPHome has the issue?

PR esphome/esphome#8300 and newer

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

font

### YAML Config

```yaml
esphome:
  name: eink-test
  friendly_name: eink-test
  on_boot:
    priority: 250
    then:
      - logger.log: "booted"
      - script.execute: eink_refresh

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

script:
  - id: eink_refresh
    mode: restart
    then:
      - delay: 1sec
      - output.turn_on: gpio_eink_pwr
      - component.update: eink
      - delay: 1min
      - output.turn_off: gpio_eink_pwr

output:
  - platform: gpio
    id: gpio_eink_pwr
    pin: GPIO12

font:
  - file: 'fonts/Gotham Rounded Medium.otf'
    id: font_main_144
    size: 144
    glyphs: [
      '!"%()+=,-_.:°/?0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZÄÖÜ abcdefghijklmnopqrstuvwxyzäöü',
      "\u0218"
    ]

spi:
  clk_pin: GPIO13
  mosi_pin: GPIO14

external_components:
  # - source: github://pr#8261
  #   components: [font]
  - source: github://pr#8300
    components: [font]

display:
  - platform: waveshare_epaper
    id: eink
    cs_pin: GPIO33
    dc_pin: GPIO27
    busy_pin:
      number: GPIO25
      inverted: true
    reset_pin: GPIO26
    model: 7.50in-bV3
    update_interval: never
    rotation: 180
    lambda: |-
      it.print(0, 0, id(font_main_144), "ÄAäayg\u0218");
      int height = id(font_main_144).get_height();
      it.line(0, height, it.get_width(), height);
      height = id(font_main_144).get_baseline();
      it.line(0, height, it.get_width(), height);
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

using esphome/esphome#8261 [image]

using esphome/esphome#8300 [image]

[image]

clydebarrow, 2025-05-24:
As far as I can tell from the image, the font is rendered identically in either case, the difference is in the value returned by `get_height()`. I assume that the lower line on your display is the one drawing using the value of `get_height()` but it would be helpful if you could log the actual values for the measurements.

JonasB2497, 2025-05-24:
Yes that is exactly what I meant. 
Sure I forgot to include them. For pre PR esphome/esphome#8300 `get_height()` returns 145 and post PR it returns 173. The baseline stays the same at 116. These values are as well in the generated C++ constructors for the `Font` object.

---

On a side note, while researching I have noticed that when loading the font in Python freetype the Face.size struct had the following values:
```
ascender: 7424
descender: -1856
height: 11072
max_advance: 11328
x_ppem: 144
x_scale: 603980
y_ppem: 144
y_scale: 603980
```
Which looks like `y_ppem` might be the right value to use here.
`height` in the documentation[1](https://freetype.org/freetype2/docs/reference/ft2-sizing_and_scaling.html#ft_size_metrics), [2](https://freetype.org/freetype2/docs/reference/ft2-face_creation.html#ft_facerec) is denoted as:

> This value is the vertical distance between two consecutive baselines, expressed in font units. It is always positive. Only relevant for scalable formats.
If you want the global glyph height, use ascender - descender.

which sounds like the font can specify some distance between lines.

clydebarrow, 2025-05-25:
Looks like the Gotham font has rather strange metrics.

[image]

JonasB2497, 2025-05-25:
> So the value being returned by get_height() is what the font file specifies as the vertical distance between successive baselines. What value are you looking for?

The value returned by `get_height()` was previously calculated as the distance between the ascender and descender, meaning the actual height of the largest glyph below which other stuff can be places safely without any gaps. That's the value I am looking for.

> Looks like the Gotham font has rather strange metrics.

Yeah maybe this font is just broken and that's the culprit I am running into. I have tested the Roboto font which looks fine for me too.

---

Thank you for your insights. I will look out for another font which looks good on epaper without this issue.

However might it be sensible to keep the height calculation as previously and add the distance between successive baselines as a new `line_height` variable? Maybe along with some other font metrics like ascender and descender. I could do a PR for this.

clydebarrow, 2025-05-25:
Using `ascender + descender` does not guarantee to give you the height of the largest glyph, it's quite possible for glyphs to draw outside those bounds (as you see with the umlaut in Gotham.)

There was never any definition of what `get_height()` was supposed to do, and it was not used in the codebase before LVGL came along, which needs to know the line height. I think it was accidental that it usually returned a value close to the correct one, just some fonts are a bit strange. 

There would certainly be no harm in having additional getters for other font metrics and a PR to add those would be welcome but IMO the current implementation of `get_height()` is correct, though the documentation should be clarified.

closed 2025-06-23 by clydebarrow

tjhorner, 2025-06-24:
Haha, I was having issues with calculating font height a while ago and also narrowed it down to that PR. Didn't have time to investigate further at the time so I pinned my ESPHome to the last known working version. Revisited the issue today and I'm glad to see someone else also ran into it.

Thank you for raising the issue and for the font metrics PR @JonasB2497 @clydebarrow, it's exactly what I needed :)