esphome/issues #7050: ESP-IDF 4.x now longer works with 2025.5.x

bdraco, 2025-05-24:
### The problem

Two issues

1. `setuptools`  https://github.com/esphome/issues/issues/6343
2. `esp_clk_tree.h` is required but not available - now required after https://github.com/esphome/esphome/pull/8542

```
src/esphome/components/esp32/core.cpp:18:10: fatal error: esp_clk_tree.h: No such file or directory

**********************************************************************
* Looking for esp_clk_tree.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_clk_tree.h"
* Web  > https://registry.platformio.org/search?q=header:esp_clk_tree.h
*
**********************************************************************

 #include <esp_clk_tree.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs/apollo-temp-1b-124868/src/esphome/components/esp32_ble_server/ble_characteristic.o
*** [.pioenvs/apollo-temp-1b-124868/src/esphome/components/esp32/core.o] Error 1
src/esphome/components/esp32/gpio.cpp:25:80: warning: missing initializer for member 'gpio_hal_context_t::version' [-Wmissing-field-initializers]
 static const gpio_hal_context_t GPIO_HAL = {.dev = GPIO_HAL_GET_HW(GPIO_PORT_0)};
                                                                                ^
```

[image]

```
ModuleNotFoundError: No module named 'pkg_resources':
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32@5.4.0/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 62:
    env.ProcessProgramDeps()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 142:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 352:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 684:
    return method(*args, **kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32@5.4.0/builder/frameworks/espidf.py", line 29:
    import pkg_resources
```

### Which version of ESPHome has the issue?

2025.5.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

profucius, 2025-07-03:
This is also still happening to me too. @jesserockz Will you please reopen this issue, it is not yet resolved.

profucius, 2025-07-03:
```
INFO ESPHome 2025.6.3
INFO Reading configuration /config/esphome/esp32-s3-box-lite-d50b0c.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING `attenuation: 11db` is deprecated, use `attenuation: 12db` instead
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO46 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Core config, version or integrations changed, cleaning build files...
INFO Deleting /data/build/esp32-s3-box-lite-d50b0c/.pioenvs
INFO Deleting /data/build/esp32-s3-box-lite-d50b0c/.piolibdeps
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.3.1
INFO Compiling app...
Processing esp32-s3-box-lite-d50b0c (board: esp32s3box; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
INFO Installing kahrendt/ESPMicroSpeechFeatures @ 1.1.0
Unpacking  [####################################]  100%
Library Manager: ESPMicroSpeechFeatures@1.1.0 has been installed!
INFO ESPMicroSpeechFeatures@1.1.0 has been installed!
Library Manager: Installing esphome/esp-audio-libs @ 1.1.4
INFO Installing esphome/esp-audio-libs @ 1.1.4
Unpacking  [####################################]  100%
Library Manager: esp-audio-libs@1.1.4 has been installed!
INFO esp-audio-libs@1.1.4 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
ModuleNotFoundError: No module named 'pkg_resources':
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32@5.4.0/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 62:
    env.ProcessProgramDeps()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 142:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/platformio/builder/tools/piobuild.py", line 352:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 684:
    return method(*args, **kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32@5.4.0/builder/frameworks/espidf.py", line 29:
    import pkg_resources
========================== [FAILED] Took 8.13 seconds ==========================
```

randybb, 2025-07-03:
IDF4 support will be removed in 2025.7 https://github.com/esphome/esphome/pull/9145 So there is no point of fixing it.

profucius, 2025-07-03:
Or will the removal replace it with something else that will allow devices to work again?

jesserockz, 2025-07-04:
We have a working configuration for s3-box as a voice assistant here: https://github.com/esphome/wake-word-voice-assistants/blob/main/esp32-s3-box/esp32-s3-box.yaml