esphome/issues #7055: ESPHome fails to resolve esp_adc_cal in Git-less ESP-IDF environment (ADF pipeline + CONFIG_ADC_CALIBRATION_ENABLED=y)

wjohnson2250, 2025-05-25:
### The problem

📋 Overview

When building an ESPHome firmware with the esp-idf framework and Audio Development Framework (ADF) components (e.g. adf_pipeline, i2s_audio, db_sound_meter), the build fails with a CMake error:

CMake Error: Failed to resolve component 'esp_adc_cal'
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

This occurs even when CONFIG_ADC_CALIBRATION_ENABLED=y is explicitly set in sdkconfig_options.

🗒️ Configuration Context

ESPHome version: 2025.5.0 (ESPHome Add-On)

Framework: esp-idf

ADF Components used: adf_pipeline, i2s_audio, db_sound_meter (custom sink)

Platform: Home Assistant add-on, building inside /data/ (no Git repo present)

YAML fragment:

esp32:
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ADC_CALIBRATION_ENABLED: y

external_components:
  - source:
      type: local
      path: /config/esphome_audio_sink/components
    components: [adf_pipeline, i2s_audio, db_sound_meter]

❗ Error Summary

The build environment is offline and Git-less (no .git directory under /data/)

ESP-IDF ≥ v5 removed esp_adc_cal as a standalone component — its functionality now lives inside esp_adc

However, ESPHome or its components still reference esp_adc_cal.h

With CONFIG_ADC_CALIBRATION_ENABLED enabled, the build tries to resolve the deprecated esp_adc_cal as a component — fails

The Git error is benign but reveals the fallback logic: IDF tries and fails to fetch missing components from registry

🔍 Root Cause

In ESP-IDF 5.x, the calibration APIs (esp_adc_cal_xxx) are deprecated and moved into esp_adc

If ESPHome or ADF components reference those APIs, the build must explicitly include the esp_adc component

In a Git-less container (like ESPHome Builder or HA add-on), the fallback resolver cannot find esp_adc_cal, and no Git metadata exists to bypass that

✅ Desired Outcome

Make ESPHome (or ADF platform logic) compatible with ESP-IDF 5.x by ensuring esp_adc is included when CONFIG_ADC_CALIBRATION_ENABLED=y is set

Provide a portable solution for users building in Git-less environments like the Home Assistant add-on

🔧 Workarounds Explored

Disable ADC calibration (temporary):Comment out CONFIG_ADC_CALIBRATION_ENABLED to bypass the dependency.

Build outside ESPHome Add-On:Use Docker or local build environment where Git context is available.

Use older ESPHome version:Building with ESP-IDF 4.x avoids the missing esp_adc_cal component.

Inject dummy esp_adc_cal component (works offline):Create a local shim that REQUIRES esp_adc to satisfy the build system.

Avoid microphone: and sound_level: blocks:These can trigger inclusion of the legacy calibration header even if unused.

📌 References

ESP-IDF Migration Docs – esp_adc_cal removal

Home Assistant Community – Git sandbox build failures

### Which version of ESPHome has the issue?

2025.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

15.2

### What platform are you using?

ESP32

### Board

AZ-Delivery

### Component causing the issue

i2c

### YAML Config

```yaml
esphome:
  name: sm300d2
  friendly_name: SM300D2

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    # sdkconfig_options:
    #   CONFIG_ADC_CALIBRATION_ENABLED: y
    #   CONFIG_ESP_TLS_INSECURE: y

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "tgRo7gJ8voCZpjD7WQ4tgJ8ibcmjaNulP9dSx86TtDQ="

ota:
  platform: esphome
  password: "40862b0108ca63b51bdde71211d7e670"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.0.9
    gateway: 192.168.0.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sm400D2 Fallback Hotspot"
    password: "p8OgEbYdhWFH"

captive_portal:

# Add SM300D2 Sensors
uart:
  id: uart_sm300
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 9600

external_components:
  - source:
      type: local
      path: /config/esphome_audio_sink/esphome/components
    components: [adf_pipeline, i2s_audio, db_sound_meter]

i2s_audio:
  - id: i2s_in
    i2s_lrclk_pin: GPIO25
    i2s_bclk_pin: GPIO26

adf_pipeline:
  - platform: i2s_audio
    type: audio_in
    id: adf_i2s_in
    i2s_audio_id: i2s_in
    i2s_din_pin: GPIO33
    sample_rate: 16000
    bits_per_sample: 16bit
    pdm: false

  - platform: db_sound_meter
    type: audio_out
    id: sound_meter_sink

# microphone:
#   - platform: adf_pipeline
#     id: adf_microphone
#     keep_pipeline_alive: true
#     pipeline:
#       - adf_i2s_in
#       - sound_meter_sink
#       - self
      
sensor:
  - platform: template
    name: "Sound Level (INMP441 ADF)"
    id: adf_db_sensor
    unit_of_measurement: "dB"
    accuracy_decimals: 1
    update_interval: 500ms

  - platform: sm300d2
    uart_id: uart_sm300
    pm_2_5:
      name: "SM300D2 PM2.5"
      icon: "mdi:air-filter"
    pm_10_0:
      name: "SM300D2 PM10"
      icon: "mdi:air-filter"
    temperature:
      name: "SM300D2 Temperature"
      icon: "mdi:thermometer"
    humidity:
      name: "SM300D2 Humidity"
      icon: "mdi:water-percent"
    co2:
      name: "SM300D2 CO2"
      icon: "mdi:molecule-co2"
    tvoc:
      name: "SM300D2 TVOC"
      icon: "mdi:chemical-weapon"
    formaldehyde:
      name: "SM300D2 Formaldehyde"
      icon: "mdi:biohazard"
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-05-25:
Do you get the issue if you don't use those external components?

wjohnson2250, 2025-05-26:
No.Sent from my iPad

ssieb, 2025-05-26:
Then it's a problem with the external components, not esphome.

wjohnson2250, 2025-05-26:
If I comment out external_components and all the dependent code, it all works ok. There is a dependency on 'esp_adc_cal', even though its not being used. Can't be right.

ssieb, 2025-05-26:
The esphome adc component uses `esp_adc_cal` and without external components it merely gives warnings about it being deprecated, but compiles and runs fine.  I don't know what interaction is involved with your external components, but that's where the issue is.

closed 2025-05-26 by ssieb

wjohnson2250, 2025-05-27:
I tried to locak clone of the repository, and accessing the fork of the repository direct. I even created a dummy esp_adc_cal module but to no avail. Below is the errors I keep getting:

INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/sm400d2.yaml...
✅ db_sound_meter.adf_pipeline platform was loaded
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Compiling app...
Processing sm300d2 (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50106.0 (5.1.6) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.10.2 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/build/sm300d2/.pioenvs/sm300d2/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'esp_adc_cal'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:302 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:309 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 11.74 seconds =========================

ssieb, 2025-05-27:
You could try using https://github.com/esphome/esphome/pull/7942

wjohnson2250, 2025-05-28:
Despite extensive testing, we've confirmed that the build error regarding esp_adc_cal is not caused by our custom or external components. The failure occurs solely due to how ESPHome's ESP-IDF integration handles deprecated components in Git-less environments like the Home Assistant add-on.

The esp_adc_cal component no longer exists in ESP-IDF v5+, yet ESPHome still triggers a dependency check when CONFIG_ADC_CALIBRATION_ENABLED=y is set.

Adding a dummy esp_adc_cal component (which forwards to esp_adc) did not resolve the issue, even when properly structured and listed in external_components.

The Git-related error (not a git repository) is a side effect of how CMake resolves components, not the root cause.

In contrast, the same configuration builds successfully when external_components are removed, further confirming that ESPHome's internal handling is too brittle in these conditions.

At this point, there's little more that can be done externally without changes to ESPHome or its handling of deprecated ESP-IDF APIs.