esphome/issues #7058: Cannot Compile a new download for ESP32

sjoram, 2025-05-25:
### The problem

I have just upgraded my HA instance to 2025.5.3 and ESPHome to 2025.5 and I now cannot compile a download for some devices. Well they seem to compile fine but when using the Manual download option I get the following error:

[image]

I thought it was my instance or my browser so I tried to do the same on my VMWare HAOS version and it gives the same problem. I've rebooted and flushed everything but to no avail. I then tried to recompile some other devices and lo and behold some of them work and others do not so it cannot be a problem with my specific environment.

The ones that present the proper download window all have no external components link and use pure ESPHome devices.
The following is an excerpt of the compile log for one of the unsuccessful downloads. All looks good but still the 404 error :(

Am I doing something wrong? As stated above I have an identical problem with ESPHOME on a docker as with the ESPHOME add on under HAOS. I have also logged in to HA from a completely different PC with the same problem......

```
INFO ESPHome 2025.4.0
INFO Reading configuration /config/esphome/jkbms-test.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing camper1-jk-bms (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
Compiling .pioenvs/camper1-jk-bms/src/main.cpp.o
Linking .pioenvs/camper1-jk-bms/firmware.elf
RAM:   [==        ]  16.3% (used 53440 bytes from 327680 bytes)
Flash: [========= ]  92.3% (used 1694621 bytes from 1835008 bytes)
Building .pioenvs/camper1-jk-bms/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/camper1-jk-bms/firmware.bin"], [".pioenvs/camper1-jk-bms/firmware.elf"])
SHA digest in image updated
Wrote 0x1af2a0 bytes to file /data/build/camper1-jk-bms/.pioenvs/camper1-jk-bms/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/camper1-jk-bms/firmware.bin"], [".pioenvs/camper1-jk-bms/firmware.elf"])
========================= [SUCCESS] Took 24.20 seconds =========================
INFO Successfully compiled program.

```

### Which version of ESPHome has the issue?

2025.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.3

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

sjoram, 2025-05-26:
So some more testing. I managed to revert to previous versions of HAOS and ESPHome. Reverting HAOS to 24.4.4 had no effect and neither did reverting to ESPHome 2025.3.

ESPHome 2025.2 however functions perfectly and I get offered the files to download with no 404 error.

So anybody know what the problem is? Is it in ESPHome or with my environment or does something need to be changed in all the external source repos to bring them up to date?

ssieb, 2025-05-26:
You didn't include the yaml, but I assume you're the person that asked on discord.  It ended up being caused by a substitution in the external components config.  This is fixed by https://github.com/esphome/esphome/pull/8911 and will be in the next point release.