esphome/issues #7077: Distorted sound on media player audio playback since 2025.5.0

MacrosorcH, 2025-05-28:
### The problem

Since ESPHome 2025.5.0, audio playback from Home Assistant to the media player of my Raspiaudio Muse Luxe is stuttering and glitching a lot, making it unusable.

In particular, playing MP3 files from either the Home Assistant's Media or from Music Assistant is suffering from the distortions. Web radio playback, on the contrary, seems to be unaffected.

ESPHome 2025.5.1 has the same issue. Rolling back to ESPHome 2025.4.2 fixes the issue.

### Which version of ESPHome has the issue?

2025.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.3

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### YAML Config

```yaml
substitutions:
  name: "altoparlante-studio"
  friendly_name: Altoparlante Studio

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
  password: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

web_server:

button:
  - platform: restart
    name: "Reboot"

text_sensor:
  - platform: wifi_info
    bssid:
      name: "WiFi BSSID"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  min_version: 2023.10.1
  on_boot:
    then:
      - light.turn_on:
          id: top_led
          effect: slow_pulse
          red: 100%
          green: 60%
          blue: 0%
      - media_player.volume_set:
          id: luxe_out
          volume: 30%

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

logger:

api:
  encryption:
    key: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

i2c:
  sda: GPIO18
  scl: GPIO23

improv_serial:

external_components:
  - source: github://RASPIAUDIO/esphomeLuxe@main
    components: [es8388]
    refresh: 0s

es8388:

globals:
  - id: wifi_connected
    type: bool
    initial_value: "false"
    restore_value: false

interval:
  - interval: 1s
    then:
      - if:
          condition:
            and:
              - lambda: "return !id(wifi_connected);"
              - wifi.connected:
          then:
            - globals.set:
                id: wifi_connected
                value: "true"
            - light.turn_on:
                id: top_led
                effect: pulse
                red: 0%
                green: 100%
                blue: 0%
            - delay: 2s
            - light.turn_on:
                id: top_led
                effect: none
                red: 0%
                green: 0%
                blue: 100%

i2s_audio:
  - i2s_lrclk_pin: GPIO25
    i2s_bclk_pin: GPIO5

media_player:
  - platform: i2s_audio
    name: None
    id: luxe_out
    dac_type: external
    i2s_dout_pin: GPIO26
    mode: stereo
    mute_pin:
      number: GPIO21
      inverted: true

microphone:
  - platform: i2s_audio
    id: luxe_microphone
    i2s_din_pin: GPIO35
    adc_type: external
    pdm: false

time:
  - platform: homeassistant
    id: homeassistant_time

sensor:
  - platform: adc
    pin: GPIO33
    name: Battery voltage
    id: battery_voltage
    device_class: voltage
    unit_of_measurement: "V"
    accuracy_decimals: 2
    state_class: measurement
    entity_category: diagnostic
    update_interval: 15s
    attenuation: auto
    filters:
      - multiply: 2
      - exponential_moving_average:
          alpha: 0.2
          send_every: 2
      - delta: 0.002
    on_value:
      then:
        - sensor.template.publish:
            id: battery_percent
            state: !lambda "return x;"

  - platform: template
    name: Battery
    id: battery_percent
    device_class: battery
    unit_of_measurement: "%"
    accuracy_decimals: 0
    state_class: measurement
    entity_category: diagnostic
    update_interval: 15s
    filters:
      - calibrate_polynomial:
          degree: 3
          datapoints:
            - 4.58 -> 100.0
            - 4.5 -> 97.1
            - 4.47 -> 94.2
            - 4.44 -> 88.4
            - 4.42 -> 82.7
            - 4.41 -> 76.9
            - 4.41 -> 71.1
            - 4.37 -> 65.3
            - 4.35 -> 59.5
            - 4.31 -> 53.8
            - 4.28 -> 48.0
            - 4.26 -> 42.2
            - 4.23 -> 36.4
            - 4.21 -> 30.6
            - 4.19 -> 24.9
            - 4.16 -> 19.1
            - 4.1 -> 13.3
            - 4.07 -> 10.4
            - 4.03 -> 7.5
            - 3.97 -> 4.6
            - 3.82 -> 1.7
            - 3.27 -> 0.0
      - lambda: return clamp(x, 0.0f, 100.0f);
  - platform: uptime
    type: timestamp
    name: Uptime
  - platform: wifi_signal
    name: "WiFi RSSI"
    update_interval: 60s
  - platform: internal_temperature
    name: "Device Temperature"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO19
      inverted: true
      mode:
        input: true
        pullup: true
    name: Volume Up
    on_click:
      - media_player.volume_up: 
          id: luxe_out
  - platform: gpio
    pin:
      number: GPIO32
      inverted: true
      mode:
        input: true
        pullup: true
    name: Volume Down
    on_click:
      - media_player.volume_down: 
          id: luxe_out
  - platform: gpio
    pin:
      number: GPIO12
      inverted: true
      mode:
        input: true
        pullup: true
    name: Action
    on_click:
      - media_player.toggle: 
          id: luxe_out

light:
  - platform: esp32_rmt_led_strip
    name: Led
    id: top_led
    pin: GPIO22
    chipset: SK6812
    num_leds: 1
    rgb_order: grb
    rmt_channel: 0
    default_transition_length: 0s
    gamma_correct: 2.8
    effects:
      - pulse:
          name: pulse
          transition_length: 250ms
          update_interval: 250ms
      - pulse:
          name: slow_pulse
          transition_length: 1s
          update_interval: 2s
```

### Anything in the logs that might be useful for us?

```txt

```

George1422, 2025-05-28:
While I don't have an answer for you on the distorted audio, and I'm not personally seeing it with the ESPHome 2025.1, my Arduino media players are still on 2025.4.3 (because I got them working without the echo, so not changing ATM), knock wood. I have had some debugging experience with these darn things. 

The Arduino media players seem to stream right from the audio source, where the esp-idf media players can stream from the ffmpeg proxy in esphome. 

So the question is: is the issue with esphome device or possibly the esphome ffmpeg proxy?

Setting the logger to level DEBUG gives you quite a bit of additional information not visible in the normal logging level. 

```
# Enable logging
logger:
  level: DEBUG
```

Now your ESP-IDF media player is setup differently than what I'm using, but the core bits are similar. Since you are forcing FLAC codec, you should see in the debug output the ffmpeg proxy sending / transcoding  all audio streams info FLAC. You might also test with MP3 and WAV forced formatted streams. With the ESP32D WROOM processor, it will only support WAV formatted audio streams because it has no PSRAM to use for codec decoding. 

Again I can't provide you an answer, only a way to get additional debugging information out of the esp32 logs. 

Also in your ESP-IDF config I "think" for your speaker the bits per channel should be 32 when you have stereo defined. Mono would be 16 bits. I'm not sure if this thinking is correct cause I'm still learning here, but with 16 bit stereo audio, that would be 8 bits for the left and 8 bits for the right, creating not so good audio??

```
bits_per_sample: 32bit
```
My configuration is similar to the one in this post (I might have even started with this config, but not sure but because he is using the same MPU as I): https://community.home-assistant.io/t/speaker-audio-media-player-in-esp32-s3-zero/881958/5

MacrosorcH, 2025-05-28:
@George1422 Thanks for your comment. I enabled DEBUG logging on the arduino-based config and played a few songs but there is nothing particular in the logs except things like these (note that the warnings show up exceptionally, only sometimes, but audio distortion/stuttering is constant):

```
21:39:13 | [D] | [media_player:081] | Media URL: http://<REDACTED>/media/local/NAS/Musica/Albums/Alessandro%20Mannarino/Bar%20della%20Rabbia/03%20svegliatevi%20italiani.mp3?authSig=<REDACTED>
21:39:13 | [D] | [esp32:117] | rtc: 0
21:39:13 | [W] | [component:257] | Component i2s_audio.media_player took a long time for an operation (559 ms).
21:39:13 | [W] | [component:258] | Components should block for at most 30 ms.
```

For the esp-idf config I am forcing FLAC because, as far as I understand, it's the least processor intensive. Am I wrong? I'll try to play with MP3 and WAV, as well as 32bit sampling, but my hopes to improve distortion and stuttering with these tweaks are low.

MacrosorcH, 2025-05-28:
Additional note: also the vanilla config available at https://github.com/esphome/media-players/blob/main/raspiaudio/raspiaudio-muse-luxe.yaml has the same distortion/stuttering issues.

MacrosorcH, 2025-05-28:
> is this mp3 served up by HA or some other system? 'NAS' might imply something else. 

I get distortion/stuttering both for files served by HA directly and from the NAS through HA. No difference.

> Using the ESP-IDF on a ESP32D WROOM with its limited memory (M5Stack Atom would also fall in this category) only WAV files work consistently in my experiences.

I will try WAV and let you know. Still, let me reiterate that, with the same config, I get distortion/stuttering on ESPHome 2025.5.x while 2025.4.2 works.

George1422, 2025-05-29:
This is just one more data point and may be misleading. I upgraded HA to 2025.5.1 over the weekend. Today I went into the computer room and could hear the fan running on the 7060. Looking at the CPU it was at 40%, which is abnormal. I couldn't find the process that was consuming the CPU. After a reboot it was back to 1%. It may be nothing, but I don't ever recall hearing the CPU fan on the previous version of HA. I'm just wondering if high CPU usage on HA could/would cause the stuttering you are reporting.

MacrosorcH, 2025-05-29:
> I'm just wondering if high CPU usage on HA could/would cause the stuttering you are reporting.

CPU usage on HA is below 10%, so no. If a CPU is peaking it must be the ESP32 of the Muse Luxe, but how can I monitor that one? Stuttering could indeed be a sign of hardware that cannot keep up with computation/memory load, but if this is the case it means ESPHome 2025.5.x is heavier than 2025.4.x on the esp board of the Muse Luxe.

Bascht74, 2025-06-01:
can you try the following?

add:

```
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components:
      - logger           # changed 5 days ago
```

I am using an nspanel that inits very long (50 seconds) and crashes with the new logger from 2025.5.x. 
(https://github.com/Blackymas/NSPanel_HA_Blueprint/issues/2543#issuecomment-2927463745)

With the old one the device started after some seconds. Maybe the changes to the logger affects the playback as well?

MacrosorcH, 2025-06-01:
@Bascht74 Thanks - I tried what you suggested and the situation improves noticeably, meaning that stuttering and distortion is clearly reduced but still present. So, the new logger from 2025.5.x. seems to be part of the problem but not all of it.

Any other idea to try?

Bascht74, 2025-06-01:
no, sorry! I don't use that.

George1422, 2025-06-01:
> Any other idea to try?

That is an interesting point that the logger is impacting audio quality. So if you remove the logger: tag altogether does the problem go away? FWIW if you still have the debug logging that will consume the most bandwidth. Removing the tag and then clean build the firmware will recreate your firmware without any logging code.

Also I think for clarity you should define (you already have) that you are using the arduino framework. 
What exact processor are you using (I know raspiaudio). 
What exact I2S audio device you are using. 

This could be used for correlation between different setups. 

Do you have another DIY media player that uses a esp32 wroom and max98357 I2S audio board? The idea is, maybe something with the I2S audio driver is not compatible with the current release of ESPHome (just spit balling at this point), but you are not using a "built in" driver for audio.

FWIW I think I saw the patch for the 67 audio repeat issue that started with 2025.4.0 was committed for the next build of esphome. I have no idea if that will change this issue or not. I know it does solve the repeating audio with the arduino framework.

MacrosorcH, 2025-06-02:
@George1422 I reduced the configuration to the bare minimum, removing logger entirely as well as most of other non-essential components (led, buttons, various sensors). The stripped-down config is pasted below. I don't have another DIY media player to try.

**Recap of setup:** Raspiaudio Muse Luxe (esp-wrover-kit board), without any modification. No idea what I2S audio device I am using, it's the one inside the Muse Luxe. ESPHome is compiled with arduino framework. I'm trying to focus on arduino because esp-idf creates even more audio glitches and suffers from an additional error on media pause which makes the media player unusable - I even tried forcing WAV instead of FLAC, no difference.

**HA setup:**
Hardware: RPi4 4GB (CPU use <10%, RAM use 40%, storage use 35%)  
Installation method: Home Assistant OS
Core: 2025.5.3
Supervisor: 2025.05.3
Operating System 15.2

**Result with the stripped-down configuration below:** 
- ESPHome **2025.5.1**: media player playback is still glitching and stuttering all the time, even if less than with logger and everything else. Removing logger in particular (or replacing it with logger from 2025.4.2) is what makes the most noticeable positive difference, but overall the problems for which I opened this issue remain.
- ESPHome **2025.4.2**: media player playback is fine.

I'm pretty sure that the patch for TTS repeat will not fix this problem, but glad to know it's coming with next build.

```
substitutions:
  name: "altoparlante-studio"
  friendly_name: Altoparlante Studio

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
  password: <REDACTED>

web_server:

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  min_version: 2023.10.1
  on_boot:
    then:
      - media_player.volume_set:
          id: luxe_out
          volume: 30%

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

api:
  encryption:
    key: <REDACTED>

i2c:
  sda: GPIO18
  scl: GPIO23

external_components:
  - source: github://RASPIAUDIO/esphomeLuxe@main
    components: [es8388]
    refresh: 0s

es8388:

i2s_audio:
  - i2s_lrclk_pin: GPIO25
    i2s_bclk_pin: GPIO5

media_player:
  - platform: i2s_audio
    name: None
    id: luxe_out
    dac_type: external
    i2s_dout_pin: GPIO26
    mode: stereo
    mute_pin:
      number: GPIO21
      inverted: true
```

George1422, 2025-06-03:
OK from my ~2 hours of testing I can't seem to duplicate your issue. I don't get distorted sound when playing music or tts speech. I do get that"Pfffffttt..." sound at the end of the audio speech, with the 67 second repeat last announcement issue. But I don't have distorted audio. I'm using a cheap 3" speaker, it sounds tinny but clear for both audio and speech. 

This is my exact configuration. 

```
esphome:
  name: altoparlante-studio
  name_add_mac_suffix: false
  min_version: 2023.10.1
  on_boot:
    then:
      - media_player.volume_set:
          id: luxe_out
          volume: 30%

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

# Enable Home Assistant API
api:
  encryption:
    key: "33AuBdjl1TyeEy2Cp53jylm6WNqzQhQf7cqwNAUCecI="

ota:
  - platform: esphome
    password: "335fbc0bfd010636d0c5cc906b506eac"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Altoparlante-Studio"
    password: "mI6C2xj8OV0P"

# Enable logging
logger:
  level: DEBUG

i2c:
  sda: GPIO18
  scl: GPIO23

external_components:
  - source: github://RASPIAUDIO/esphomeLuxe@main
    components: [es8388]
    refresh: 0s

es8388:

i2s_audio:
  - i2s_lrclk_pin: GPIO25
    i2s_bclk_pin: GPIO5

media_player:
  - platform: i2s_audio
    name: Media Player
    id: luxe_out
    dac_type: external
    i2s_dout_pin: GPIO26
    mode: stereo
    mute_pin:
      number: GPIO21
      inverted: true
    
```

[... 111 lines cut ...]
[19:08:55][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[19:08:55][D][esp32.preferences:114]: Saving 1 preferences to flash...
[19:08:55][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[19:10:01][D][media_player:074]: 'Media Player' - Setting
[19:10:01][D][media_player:084]:   Volume: 0.65
[19:10:03][D][media_player:074]: 'Media Player' - Setting
[19:10:03][D][media_player:081]:   Media URL: http://192.168.112.202:8123/media/house/anc_tone3.mp3?authSig=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiIzNzk4M2M3NmIwZDQ0ZDNmOWE4OTQ5NTliZTljYzRjYiIsInBhdGgiOiIvbWVkaWEvaG91c2UvYW5jX3RvbmUzLm1wMyIsInBhcmFtcyI6W10sImlhdCI6MTc0ODkwNTgwMywiZXhwIjoxNzQ4OTkyMjAzfQ.mr3I-PD3z9TqopvSIhbGkfLc7DKCpr4tt-oS7fkbBbM
[19:10:03][D][media_player:087]:  Announcement: yes
[19:10:03][W][component:257]: Component api took a long time for an operation (207 ms).
[19:10:03][W][component:258]: Components should block for at most 30 ms.
[19:10:06][D][media_player:074]: 'Media Player' - Setting
[19:10:06][D][media_player:081]:   Media URL: http://192.168.112.202:8123/api/tts_proxy/8RstDlxVgA7qxr8j8uIIzg.mp3
[19:10:06][D][media_player:087]:  Announcement: yes
[19:10:14][D][media_player:074]: 'Media Player' - Setting
[19:10:14][D][media_player:081]:   Media URL: http://www.101smoothjazz.com/101-smoothjazz.m3u
[19:10:14][W][component:257]: Component api took a long time for an operation (419 ms).
[19:10:14][W][component:258]: Components should block for at most 30 ms.
```

If you are getting distorted audio playing the smooth jazz channel, I might think eliminating the HA server is at fault.  
You also mentioned that you get distorted audio when playing a local mp3 not on the HA server. 

I realize you said the trigger was upgrading HA, but the truth table seems to point to the raspiaudio device (board) or your wifi network. Its possible that the HA upgrade placed an abnormal load on your wifi (??). 

We have to look what is in common between all of the conditions you tested. It would seem to be the board or wifi. Adding the logger places extra load on your wifi communications. Streaming audio packets arriving out of order or not arriving at all would create a distorted audio. Now understand I'm not saying the issue is your board or wifi, just the conditions point to one of the two. If you had a different audio player using a different esp32 chip and audio board we might be able to rule out the raspiaudio device.

MacrosorcH, 2025-06-03:
@George1422 Thanks for your help in debugging this issue. 

Playback of web radios always works fine, as I also wrote in the ticket description.

TTS also always plays fine (except for the unrelated repeating issue).

Distortion/glitches happen only with MP3 songs streamed from HA and/or Music Assistant. I have a Voice PE too and that one works perfectly fine.

Now, I made a few more experiments with ESPHome 2025.5.2 and Raspiaudio Muse Luxe, and I restricted the list of culprits to 2 components:
- **logger**
- **web_server**

Previously, I had removed only logger and the situation was noticeably better but not really fixed. Now I removed also web_server and the glitches are 99% gone.

What I can try to guess from this fact is that, in ESPHome 2025.5.x, both logger and web_server are heavier than they used to be in ESPHome 2025.4.x, where I could keep them all enabled without problems. Alternatively, other components are more resource-hungry and those two are somehow penalized. No proof of any of these theories.

To recap, the following stripped down config works reasonably fine on Raspiaudio Muse Luxe with ESPHome 2025.5.2. Now I will try to figure out how to tweak the full config.

```
esphome:
  name: "altoparlante-studio"
  name_add_mac_suffix: false
  min_version: 2023.10.1
  on_boot:
    then:
      - media_player.volume_set:
          id: luxe_out
          volume: 30%

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
  password: <REDACTED>

# commented as it causes audio glitches
#web_server:

# commented as it causes audio glitches
#logger:

api:
  encryption:
    key: <REDACTED>

i2c:
  sda: GPIO18
  scl: GPIO23

external_components:
  - source: github://RASPIAUDIO/esphomeLuxe@main
    components: [es8388]
    refresh: 0s

es8388:

i2s_audio:
  - i2s_lrclk_pin: GPIO25
    i2s_bclk_pin: GPIO5

media_player:
  - platform: i2s_audio
    name: None
    id: luxe_out
    dac_type: external
    i2s_dout_pin: GPIO26
    mode: stereo
    mute_pin:
      number: GPIO21
      inverted: true
```

MacrosorcH, 2025-06-14:
Closing this issue because I think the optimizations introduced in ESPHome 2025.6.0 do solve at least the part of the problem that is relatable to ESPHome. I hope the config above can help others with Raspiaudio Muse Luxe.

closed 2025-06-14 by MacrosorcH